thesofproject/sof #2467: [BUG][TGL]"ipc error for 0x60060000" when do pause/resume test with  multipipeline(random)

Liviali155, 2020-03-02:
Describe the bug
"ipc error for 0x60060000" when do pause/release test with  multipipeline(random)
After error occured aplay/arecord again,it can work normal

To Reproduce
1.sudo reboot 
2.plugin HDMI&DP
3.open 6 terminal to run below commands Separately
 ```
aplay -Dhw:0,0 -r48000 -c2 -fs16_le /wav/48California_Dreaming.wav -vv -i
aplay -Dhw:0,3 -r48000 -c2 -fs16_le /wav/48California_Dreaming.wav -vv -i
aplay -Dhw:0,4 -r48000 -c2 -fs16_le /wav/48California_Dreaming.wav -vv -i
arecord -Dhw:0,0 -r48000 -c2 -fs16_le s30.wav -vv -i
arecord -Dhw:0,6 -r48000 -c4 -fs16_le s30.wav -vv -i
arecord -Dhw:0,7 -r48000 -c4 -fs16_le s30.wav -vv -i
```
4.do pause/release Separately

Reproduction Rate
Tested 3times,reproduced 1time

Expected behavior
There is no I/O error when do pause/release test

Actual result
When do pause/release  about 15th of "arecord -Dhw:0,0 -r48000 -c2 -fs16_le s30.wav -vv -i",I/O error,dmesg show "ipc error for 0x60060000"
<details>
 <summary>error dmesg</summary>

````
[  850.240061] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  850.240120] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  850.671355] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 3
[  850.671369] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[  850.671879] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60060000 size 12
[  850.671892] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[  850.671900]  HDA Analog: ASoC: trigger FE cmd: 3 failed: -22
[  850.671910] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 4
[  850.672127] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
[  850.672132] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[  850.672225] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60070000 size 12
[  850.672229]  HDA Analog: ASoC: trigger FE cmd: 4 failed: -22
[  850.740057] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  850.740093] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  851.239767] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  851.239802] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  851.278635] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 3
[  851.278644] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[  851.278839] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60060000 size 12
[  851.279130] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20240000 successful
[  851.279136]  HDA Analog: ASoC: trigger FE cmd: 3 failed: -22
[  851.279142] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 0 dir 1 cmd 4
[  851.279771] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x2024001e successful
````
</details>

<details>
 <summary>sof_errortrace</summary>

````
CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA         1645726078.697917              nan intel/cavs/hda-dma.c:789 	hda_dma_link_check_xrun() error: xrun detected
    0      1         COMP 2.15    1645726091.093750        12.395833      src/audio/dai.c:634 	dai_report_xrun() error: overrun due to no data available
    0      1         COMP 2.15    1645726101.041667         9.947917      src/audio/dai.c:920 	comp_overrun() error: sink->free = 768, copy_bytes = 0
    0      1         PIPE         1645726174.687500        73.645836 src/audio/pipeline.c:773 	pipeline_copy() error: ret = -61, start->comp.id = 15, dir = 0
    0      1         PIPE 2.16    1645726187.239583        12.552083 src/audio/pipeline.c:940 	pipeline_task(): xrun recover failed! pipeline will be stopped!
    0      1         COMP 2.15    1646936159.166667   1209971.875000 rc/audio/component.c:160 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_PAUSE
    0      1         PIPE         1646936171.354167        12.187500 src/audio/pipeline.c:645 	pipeline_trigger() error: ret = -22, host->comp.id = 9, cmd = 2
    0      1          IPC         1646936183.645833        12.291667    src/ipc/handler.c:517 	ipc: comp 9 trigger 0x60000 failed -22
    0      1         COMP 2.15    1646937037.656250       854.010437 rc/audio/component.c:136 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_RELEASE
    0      1         PIPE         1646937049.739583        12.083333 src/audio/pipeline.c:645 	pipeline_trigger() error: ret = -22, host->comp.id = 9, cmd = 3
    0      1          IPC         1646937061.666667        11.927083    src/ipc/handler.c:517 	ipc: comp 9 trigger 0x70000 failed -22
    0      1         COMP 2.15    1648150158.750000   1213097.125000 rc/audio/component.c:160 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_PAUSE
    0      1         PIPE         1648150170.885417        12.135417 src/audio/pipeline.c:645 	pipeline_trigger() error: ret = -22, host->comp.id = 9, cmd = 2
    0      1          IPC         1648150183.072917        12.187500    src/ipc/handler.c:517 	ipc: comp 9 trigger 0x60000 failed -22
    0      1         COMP 2.15    1648152334.322917      2151.250000 rc/audio/component.c:136 	comp_set_state() error: wrong state = 1, COMP_TRIGGER_RELEASE
    0      1         PIPE         1648152346.406250        12.083333 src/audio/pipeline.c:645 	pipeline_trigger() error: ret = -22, host->comp.id = 9, cmd = 3
    0      1          IPC         1648152358.489583        12.083333    src/ipc/handler.c:517 	ipc: comp 9 trigger 0x70000 failed -22
    0      1         COMP 2.10    1649166807.447917   1014448.937500 rc/audio/component.c:169 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
    0      1         COMP 2.11    1649166832.968750        25.520834 rc/audio/component.c:169 	comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
````
</details>

Platform: TGL-U RVP with onboard codec ALC711 in HDA mode
Firmware:c3db256a https://github.com/thesofproject/sof/commits/master
Kernel: 30887e1 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/ sof-hda-generic-4ch.tplg same as firmware

[dmesg.0302.log](https://github.com/thesofproject/linux/files/4275436/dmesg.0302.log)
[sof_errortrace0302.log](https://github.com/thesofproject/linux/files/4275437/sof_errortrace0302.log)
[sof_trace0302.log](https://github.com/thesofproject/linux/files/4275438/sof_trace0302.log)

tlauda, 2020-03-03:
@Liviali155 Could you verify whether L1SEN is disabled?

Liviali155, 2020-03-03:
releated with https://github.com/thesofproject/sof/issues/2404

Liviali155, 2020-03-03:
> @Liviali155 Could you verify whether L1SEN is disabled?

How  to check it?

tlauda, 2020-03-03:
@ranj063 @plbossart Is L1SEN also disabled on TGL platforms?

plbossart, 2020-03-03:
> @ranj063 @plbossart Is L1SEN also disabled on TGL platforms?

Our default config is:

````
# CONFIG_SND_SOC_SOF_HDA_ALWAYS_ENABLE_DMI_L1 is not set
````

ranj063, 2020-03-03:
> @ranj063 @plbossart Is L1SEN also disabled on TGL platforms?

Yes, it is disabled for all of our HDA platforms but only when a capture stream is open.

Liviali155, 2020-03-04:
Found a step which can reproduce this issue every time,step as below:
1.sudo reboot
2.plugin HDMI&DP
3.open 4 terminal to run below script Separately at the same time
 ```
  ./sof-test-pause-release.sh -t pcm0p
  ./sof-test-pause-release.sh -t pcm3p
  ./sof-test-pause-release.sh -t pcm4p
  ./sof-test-pause-release.sh -t pcm0c
```

 "./sof-test-pause-release.sh -t pcm0c"  will I/O error,script as attachment
[sof-test-pause-release.txt](https://github.com/thesofproject/sof/files/4284874/sof-test-pause-release.txt)

tlauda, 2020-03-06:
@ClarexZhou @Liviali155 Could you check whether it reproduces with FW from this branch https://github.com/tlauda/sof/tree/topic/hda-link-xruns.

Liviali155, 2020-03-10:
> [uns](https://github.com/tlauda/sof/tree/topic/hda-link-xruns).

Tried with https://github.com/tlauda/sof/tree/topic/hda-link-xruns.,error not seen

Liviali155, 2020-03-10:
> @ClarexZhou @Liviali155 Could you check whether it reproduces with FW from this branch https://github.com/tlauda/sof/tree/topic/hda-link-xruns.

Tried with https://github.com/tlauda/sof/tree/topic/hda-link-xruns.,error not seen

mengdonglin, 2020-03-10:
@tlauda Has the fix been merged by master branch?

tlauda, 2020-03-10:
@mengdonglin Not yet.
@Liviali155 Error is not seen, but does the recording look correct? No glitches?

Liviali155, 2020-03-10:
> @mengdonglin Not yet.
> @Liviali155 Error is not seen, but does the recording look correct? No glitches?

I tested on https://github.com/tlauda/sof/tree/topic/hda-link-xruns. with below steps:
1.sudo reboot
2.plugin HDMI&DP
3.open 4 terminal to run below script Separately at the same time

 ./sof-test-pause-release.sh -t pcm0p
 ./sof-test-pause-release.sh -t pcm3p
 ./sof-test-pause-release.sh -t pcm4p
 ./sof-test-pause-release.sh -t pcm0c

error not seen ,other glitches means I need to test other function?

tlauda, 2020-03-10:
@Liviali155 I mean if the recorded file is OK? No quality problems or pops?

tlauda, 2020-03-10:
@Liviali155 @mengdonglin Fix in PR #2512.

Liviali155, 2020-03-10:
> @Liviali155 @mengdonglin Fix in PR #2512.

Tried with PR #2512
Error not seen and the recorded file is normal

tlauda, 2020-03-11:
Fixed with #2512.

closed 2020-03-11 by tlauda