thesofproject/sof #2469: I: [pulseaudio-13.99.1] alsa-util.c: Error opening PCM device hw:sofhdadsp,6: No such file or directory

lkraav, 2020-03-02:
Machine: Dell 7390 Ice Lake
Kernel: 5.5.7-gentoo+
Alsa-info: http://alsa-project.org/db/?f=e46cd6d581178ee8251c0d54b458cde23441ad11
Pulseaudio: 13.99.1
SOF-firmware: 1.4.2

Something is preventing Pulseaudio from completing profile initialization. Only error I'm seeing is in issue title. Full pulseaudio debug log below for reference. Any thoughts on what to try next?

```
leho@papaya pulse $ [-] pulseaudio -vvvv                                                                                                                                                                                                                                                                                                                                                                                                                                                                                        
I: [pulseaudio] main.c: setrlimit(RLIMIT_NICE, (31, 31)) failed: Operation not permitted                                                                                                                                                                                                                                                                                                                                                                                                                                        
I: [pulseaudio] main.c: setrlimit(RLIMIT_RTPRIO, (9, 9)) failed: Operation not permitted                                                                                                                                                                                                                                                                                                                                                                                                                                        
D: [pulseaudio] core-rtclock.c: Timer slack is set to 50 us.                                                                                                                                                                                                                                                                                                                                                                                                                                                                    
I: [pulseaudio] core-util.c: Failed to acquire high-priority scheduling: No such file or directory                                                                                                                                                                                                                                                                                                                                                                                                                              
I: [pulseaudio] main.c: This is PulseAudio 13.99.1                                                                                                                                                                                                                                                                                                                                                                                                                                                                              D: [pulseaudio] main.c: Compilation host: x86_64-pc-linux-gnu                                                                                                                                                                                                                                                                                                                                                                                                                                                                   
D: [pulseaudio] main.c: Compilation CFLAGS: -O2 -pipe -march=native -Wl,-O1 -Wl,--as-needed -Wall -W -Wextra -Wno-long-long -Wno-overlength-strings -Wunsafe-loop-optimizations -Wundef -Wformat=2 -Wlogical-op -Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wdeclaration-after-statement -Wfloat-equal -Wmissing-prototypes -Wredundant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wcast-align -Wstrict-aliasing -Wwrite-strings -Wno-unused-parameter -fno-common -fdiagnostics-show-option -fdiagnostics-color=auto                                                                                                                                                                          
D: [pulseaudio] main.c: Running on host: Linux x86_64 5.5.7-gentoo+ #14 SMP PREEMPT Mon Mar 2 13:19:30 EET 2020                                                                                                                                                                                                                                                                                                                                                                                                                 D: [pulseaudio] main.c: Found 8 CPUs.                                                                                                                                                                                                                     
I: [pulseaudio] main.c: Page size is 4096 bytes                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 D: [pulseaudio] main.c: Compiled with Valgrind support: yes                                                                
D: [pulseaudio] main.c: Running in valgrind mode: no                                                                                                                                                                                                                                                                                                                                                                                                                                                                            D: [pulseaudio] main.c: Running in VM: no                                                                                  
D: [pulseaudio] main.c: Running from build tree: no                                                                                                                                                                                                                                                                                                                                                                                                                                                                             D: [pulseaudio] main.c: Optimized build: yes                                                                                                                                                                                                              
D: [pulseaudio] main.c: FASTPATH defined, only fast path asserts disabled.                                                                                                                                                                                                                                                                                                                                                                                                                                                      I: [pulseaudio] main.c: Machine ID is 3ca321b6c89b0b95a0d5993d00000028.                                                                                                                                                                                                                                                                                                                                                                                                        [... cut ...] l=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.
I: [pulseaudio] module-udev-detect.c: Card /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0 (alsa_card.pci-0000_00_1f.3-platform-skl_hda_dsp_generic) failed to load module.                                                                    
I: [pulseaudio] module-udev-detect.c: Found 1 cards.
...
```

juimonen, 2020-03-02:
@lkraav I think the "6" is hard coded in alsa machine driver for dmic (or at least it was like that some time ago). You could check with arecord -l as the dmic should be listed there...

lkraav, 2020-03-02:
I added arecord output

```
leho@papaya / $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

Not sure what to make of it?

juimonen, 2020-03-02:
@lkraav hmm ok, to me it looks like you wouldn't have digital mics for some reason (so the driver at least seems to think so). @plbossart you have any idea? Is is that we don't have correct machine driver loaded?

lkraav, 2020-03-02:
Hmmm, I would assume Ice Lake laptops would have it across the board, right?

```
leho@papaya ~ $ [-] zgrep SOF /proc/config.gz 
...
CONFIG_SND_SOC_INTEL_SOF_RT5682_MACH=m
CONFIG_SND_SOC_SOF_TOPLEVEL=y
CONFIG_SND_SOC_SOF_PCI=m
CONFIG_SND_SOC_SOF_ACPI=m
CONFIG_SND_SOC_SOF=m
CONFIG_SND_SOC_SOF_PROBE_WORK_QUEUE=y
CONFIG_SND_SOC_SOF_INTEL_TOPLEVEL=y
CONFIG_SND_SOC_SOF_INTEL_ACPI=m
CONFIG_SND_SOC_SOF_INTEL_PCI=m
CONFIG_SND_SOC_SOF_INTEL_COMMON=m
# CONFIG_SND_SOC_SOF_BAYTRAIL_SUPPORT is not set
# CONFIG_SND_SOC_SOF_BROADWELL_SUPPORT is not set
# CONFIG_SND_SOC_SOF_MERRIFIELD_SUPPORT is not set
# CONFIG_SND_SOC_SOF_APOLLOLAKE_SUPPORT is not set
# CONFIG_SND_SOC_SOF_GEMINILAKE_SUPPORT is not set
# CONFIG_SND_SOC_SOF_CANNONLAKE_SUPPORT is not set
CONFIG_SND_SOC_SOF_COFFEELAKE_SUPPORT=y
CONFIG_SND_SOC_SOF_COFFEELAKE=m
CONFIG_SND_SOC_SOF_ICELAKE_SUPPORT=y
CONFIG_SND_SOC_SOF_ICELAKE=m
# CONFIG_SND_SOC_SOF_COMETLAKE_LP_SUPPORT is not set
# CONFIG_SND_SOC_SOF_COMETLAKE_H_SUPPORT is not set
# CONFIG_SND_SOC_SOF_TIGERLAKE_SUPPORT is not set
# CONFIG_SND_SOC_SOF_ELKHARTLAKE_SUPPORT is not set
# CONFIG_SND_SOC_SOF_JASPERLAKE_SUPPORT is not set
CONFIG_SND_SOC_SOF_HDA_COMMON=m
CONFIG_SND_SOC_SOF_HDA_LINK=y
CONFIG_SND_SOC_SOF_HDA_AUDIO_CODEC=y
# CONFIG_SND_SOC_SOF_HDA_ALWAYS_ENABLE_DMI_L1 is not set
CONFIG_SND_SOC_SOF_HDA_COMMON_HDMI_CODEC=y
CONFIG_SND_SOC_SOF_HDA_LINK_BASELINE=m
CONFIG_SND_SOC_SOF_HDA=m
CONFIG_SND_SOC_SOF_XTENSA=m
...
```

DMIC

```
$ [-] zgrep DMIC /proc/config.gz
CONFIG_SND_SOC_DMIC=m
```

juimonen, 2020-03-02:
@lkraav yes I meant the hw would be there but it is not _detected_ by the driver, or correct version is not loaded etc.. I'm currently not at the office, so can't check (and can't remember by heart) how the dmic should exactly look like. let me get back to you tomorrow or someone else with more knowledge will answer you before that... :)

plbossart, 2020-03-02:
@lkraav can you attach the full dmesg log (with dynamic debug enabled).
Not all devices have DMICs, we have to look at an NHLT property to find out.

lkraav, 2020-03-02:
So `dyndbg="+p"` was a bit overkill, not sure I can attach that anywhere :dancing_men: any suggestions for a smaller query scope that could work here?

Maybe a `grep dmic` of that boot does the trick?

PS I had `CONFIG_SPI=n`, not sure if related, but enabled it (saw Ice Lake in device description list).

```
$ [-] journalctl -b -1 | grep dmic
märts 02 21:24:55 papaya kernel: Failed to find symbol intel_nhlt_get_dmic_geo
märts 02 21:24:55 papaya kernel: Registering platform device 'dmic-codec'. Parent at 0000:00:1f.3
märts 02 21:24:55 papaya kernel: device: 'dmic-codec': device_add
märts 02 21:24:55 papaya kernel: kobject: 'dmic-codec' (000000001c6a934a): kobject_add_internal: parent: '0000:00:1f.3', set: 'devices'
märts 02 21:24:55 papaya kernel: bus: 'platform': add device dmic-codec
märts 02 21:24:55 papaya kernel: PM: Adding info for platform:dmic-codec
märts 02 21:24:55 papaya kernel: kobject: 'dmic-codec' (000000001c6a934a): kobject_uevent_env
märts 02 21:24:55 papaya kernel: kobject: 'dmic-codec' (000000001c6a934a): fill_kobj_path: path = '/devices/pci0000:00/0000:00:1f.3/dmic-codec'
märts 02 21:24:55 papaya kernel: devices_kset: Moving dmic-codec to end of list
märts 02 21:24:55 papaya kernel: PM: Moving platform:dmic-codec to end of list
märts 02 21:24:55 papaya kernel: snd_soc_dmic does not use snd_soc_core!
märts 02 21:24:55 papaya kernel: Allocating new usage for snd_soc_dmic.
märts 02 21:24:55 papaya kernel: kobject: 'snd_soc_dmic' (000000000ae763e3): kobject_add_internal: parent: 'module', set: 'module'
märts 02 21:24:55 papaya kernel: kobject: 'holders' (00000000a42f7ac8): kobject_add_internal: parent: 'snd_soc_dmic', set: '<NULL>'
märts 02 21:24:55 papaya kernel: kobject: 'notes' (000000007a39fdae): kobject_add_internal: parent: 'snd_soc_dmic', set: '<NULL>'
märts 02 21:24:55 papaya kernel: kobject: 'snd_soc_dmic' (000000000ae763e3): kobject_uevent_env
märts 02 21:24:55 papaya kernel: kobject: 'snd_soc_dmic' (000000000ae763e3): fill_kobj_path: path = '/module/snd_soc_dmic'
märts 02 21:24:55 papaya kernel: bus: 'platform': add driver dmic-codec
märts 02 21:24:55 papaya kernel: kobject: 'dmic-codec' (0000000074f9dea9): kobject_add_internal: parent: 'drivers', set: 'drivers'
märts 02 21:24:55 papaya kernel: bus: 'platform': driver_probe_device: matched device dmic-codec with driver dmic-codec
märts 02 21:24:55 papaya kernel: bus: 'platform': really_probe: probing driver dmic-codec with device dmic-codec
märts 02 21:24:55 papaya kernel: dmic-codec dmic-codec: no default pinctrl state
märts 02 21:24:55 papaya kernel: driver: 'dmic-codec': driver_bound: bound to device 'dmic-codec'
märts 02 21:24:55 papaya kernel: kobject: 'dmic-codec' (000000001c6a934a): kobject_uevent_env
märts 02 21:24:55 papaya kernel: kobject: 'dmic-codec' (000000001c6a934a): fill_kobj_path: path = '/devices/pci0000:00/0000:00:1f.3/dmic-codec'
märts 02 21:24:55 papaya kernel: bus: 'platform': really_probe: bound device dmic-codec to driver dmic-codec
märts 02 21:24:55 papaya kernel: kobject: 'drivers' (000000004e6e8ab8): kobject_add_internal: parent: 'snd_soc_dmic', set: '<NULL>'
märts 02 21:24:55 papaya kernel: kobject: 'dmic-codec' (0000000074f9dea9): kobject_uevent_env
märts 02 21:24:55 papaya kernel: kobject: 'dmic-codec' (0000000074f9dea9): fill_kobj_path: path = '/bus/platform/drivers/dmic-codec'
märts 02 21:24:56 papaya kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link dmic01
märts 02 21:24:56 papaya kernel: skl_hda_dsp_generic skl_hda_dsp_generic: info: override BE DAI link dmic16k
```

lkraav, 2020-03-02:
I find I also have to force `snd_intel_dspcfg.dsp-driver=3` for things to even enter SOF mode.

Auto-detection definitely failing here in some way.

plbossart, 2020-03-02:
If you can recompile the kernel, it's just simpler to add -DDEBUG in the parts you want, e.g.
https://github.com/thesofproject/linux/blob/65f9d39d7beba2832311cc3e09e824fd27f7334d/sound/soc/sof/Makefile#L3

add this for sound/soc/sof/Makefile and sound/soc/sof/intel/Makefile

You should not need any blacklists or options. I don't use them. It's better to remove all this and see why things are not detected.

Or you blacklist everything (snd-hda-intel, snd-sof-pci, snd-soc-skl) and start manually with dmesg -C; modprobe snd-sof-pci.

that way you only see what you need for SOF.

perexg, 2020-03-03:
It appears like UCM configuration problem. From the logs there is probably no digital microphone on this platform.

@lkraav : If you use the legacy HDA driver (snd_intel_dspcfg.dsp-driver=1), does the internal microphone work?

lkraav, 2020-03-03:
> If you use the legacy HDA driver (snd_intel_dspcfg.dsp-driver=1), does the internal microphone work?

@perexg yep works, been rolling with `snd_hda_intel` ever since acquiring this machine 1 month ago.

Can we decipher DMIC existence from ACPI data tables with certainty? Feels like the lowest burdain method.

mengdonglin, 2020-03-04:
@lkraav : can you share more info about your machine model?
Is it a Dell Latitude 7390 2-in-1 Laptop, with Intel Core i5-8350U, similiar to this one?
https://www.amazon.com/Dell-Latitude-7390-13-3inch-Touchscreen/dp/B07K2LMDGQ/ref=asc_df_B07K2LMDGQ/?tag=hyprod-20&linkCode=df0&hvadid=309818716690&hvpos=&hvnetw=g&hvrand=5748957374971564684&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9032154&hvtargid=pla-582799509773&psc=1

lkraav, 2020-03-04:
Thanks for pointing out my vague model description, I updated opening comment.

It's actually the XPS 13 7390 (gotta love DELL's overlapping enumeration schemes!) https://www.notebookcheck.net/Dell-XPS-13-7390-2-in-1.425189.0.html with Intel Core i7-1065G7 CPU

mengdonglin, 2020-03-10:
@lkraav thanks and we'll try to get this Dell laptop and check SOF compatibility on it.