thesofproject/sof #2472: [BUG]Multi media playback+analog pause and stop cause ipc error for 0x60070000 size 12

dengyangchao, 2020-03-03:
**Describe the bug**
Multi media playback+analog pause-release cause ipc error for 0x60070000 size 12.
When issue occur, sof can still work.

**To Reproduce**
1. Reboot system via "sudo reboot"
2. Open terminal 1 to aplay HW:0,0 "aplay -Dhw:0,0 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i"
2. Open terminal 2 to aplay HW:0,1 "aplay -Dhw:0,1 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i"
3. Pause terminal 1 and 2.
4. Use ctrl+c to stop terminal 1
5. Use ctrl+c to stop terminal 2

**Reproduction Rate**
5/5

**Expected behavior**
Both aplay can stop normally

**Impact**
Occur ipc error

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev   30887e1f
    * SOF: master  c3db256a
2) Name of the topology file
    * Topology: sof-byt-rt5682.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: BYT-MB with codec RT5682 in I2S mode

**Screenshots or console output**
[image]
[image]
[image]

**Dmesg **
```
[  139.526335] sof-audio-acpi 80860F28:01: PCM: stream 1 dir 0 DMA position 8960 DAI position 1281062
[  139.526520] sof-audio-acpi 80860F28:01: PCM: stream 1 dir 0 DMA position 8960 DAI position 1281062
[  139.526527] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 3
[  139.526533] sof-audio-acpi 80860F28:01: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[  139.527024] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[  139.620032] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[  139.620046] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  150.368186] sof-audio-acpi 80860F28:01: pcm: trigger stream 0 dir 0 cmd 4
[  150.368197] sof-audio-acpi 80860F28:01: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[  150.368317] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[  150.368328] sof-audio-acpi 80860F28:01: pcm: trigger stream 0 dir 0 cmd 0
[  150.368334] sof-audio-acpi 80860F28:01: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  150.368791] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  150.368798] sof-audio-acpi 80860F28:01: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  150.368887] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  150.368898] sof-audio-acpi 80860F28:01: pcm: free stream 0 dir 0
[  150.368921] sof-audio-acpi 80860F28:01: pcm: close stream 0 dir 0
[  150.622785] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[  150.622801] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  153.089089] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 4
[  153.089100] sof-audio-acpi 80860F28:01: ipc tx: 0x60070000: GLB_STREAM_MSG: TRIG_RELEASE
[  153.089253] sof-audio-acpi 80860F28:01: error: ipc error for 0x60070000 size 12
[  153.089263]  Media Playback 1: ASoC: trigger FE cmd: 4 failed: -22
[  153.089271] sof-audio-acpi 80860F28:01: pcm: trigger stream 1 dir 0 cmd 3
[  153.089277] sof-audio-acpi 80860F28:01: ipc tx: 0x60060000: GLB_STREAM_MSG: TRIG_PAUSE
[  153.089388] sof-audio-acpi 80860F28:01: error: ipc error for 0x60060000 size 12
[  153.089394]  Media Playback 1: ASoC: trigger FE cmd: 3 failed: -22
[  153.089405] sof-audio-acpi 80860F28:01: pcm: free stream 1 dir 0
[  153.089411] sof-audio-acpi 80860F28:01: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  153.089567] sof-audio-acpi 80860F28:01: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
[  153.089600] sof-audio-acpi 80860F28:01: pcm: close stream 1 dir 0
[  153.123428] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[  153.123449] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  153.623890] sof-audio-acpi 80860F28:01: ipc rx: 0x90020000: GLB_TRACE_MSG
[  153.623925] sof-audio-acpi 80860F28:01: ipc rx done: 0x90020000: GLB_TRACE_MSG
```
**SOF-error-trace**
```
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1         COMP 1.4     153052788.541667 153052784.000000 rc/audio/component.c:136 	comp_set_state() error: wrong state = 3, COMP_TRIGGER_RELEASE
    0      1         PIPE         153052798.385417         9.843750 src/audio/pipeline.c:645 	pipeline_trigger() error: ret = -22, host->comp.id = 19, cmd = 3
    0      1          IPC         153052810.052083        11.666667    src/ipc/handler.c:517 	ipc: comp 19 trigger 0x70000 failed -22
    0      1         COMP 1.4     153052965.468750       155.416672 rc/audio/component.c:160 	comp_set_state() error: wrong state = 3, COMP_TRIGGER_PAUSE
    0      1         PIPE         153052975.312500         9.843750 src/audio/pipeline.c:645 	pipeline_trigger() error: ret = -22, host->comp.id = 19, cmd = 2
    0      1          IPC         153052987.031250        11.718750    src/ipc/handler.c:517 	ipc: comp 19 trigger 0x60000 failed -22
    0      1         COMP 3.20    153053106.875000       119.843750 rc/audio/component.c:169 	comp_set_state() error: wrong state = 4, COMP_TRIGGER_RESET
    0      1         COMP 3.21    153053120.520833        13.645833 rc/audio/component.c:169 	comp_set_state() error: wrong state = 4, COMP_TRIGGER_RESET
```

[amixer.txt](https://github.com/thesofproject/sof/files/4278891/amixer.txt)
[dmesg.log](https://github.com/thesofproject/sof/files/4278892/dmesg.log)
[logger.log](https://github.com/thesofproject/sof/files/4278893/logger.log)
[sof-error-trace.log](https://github.com/thesofproject/sof/files/4278894/sof-error-trace.log)

dengyangchao, 2020-03-03:
APL UP2 NOCODEC
APL UP2 PCM512X
BYT MB NOCODEC
these 3 platform can also reproduce the issue

tlauda, 2020-03-03:
@dengyangchao Please verify whether #2477 fixes the issue.

dengyangchao, 2020-03-04:
@tlauda 
Tried on 
APL UP2 NOCODEC
APL UP2 PCM512X
BYT MB NOCODEC
BYT MB RT5682
After applied #2477 , issue can't be reproduced.

PR #2477 (2020-03-04): mixer: fix state transition
Mixer shouldn't go into COMP_STATE_PREPARE if there is at least
one source in COMP_STATE_PAUSED.

Fixes #2472.
closed 2020-03-04 by tlauda