thesofproject/sof #2473: [BUG][CML]Sound from aplay0,0(RT711) and aplay0,2(RT1308) auto speed up.

YvonneYang2, 2020-03-03:
**Describe the bug**
Sound from Aplay0,0(RT715) or Aplay0,2(RT1308) auto speed up.  options snd-sof-intel-hda-common sdw_clock_stop_quirks=0x1 and without option both can reproduce it. 

**To Reproduce**
1. Boot up with pulse audio on.
2. Run command "aplay -Dhw:0,2 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i"

**Reproduced rate**
100%
 
**Expected result**
Sound from Aplay0,0(RT715) or Aplay0,2(RT1308) is normal

**Actual result**
Sound from Aplay0,0(RT715) or Aplay0,2(RT1308) auto speed up. 

**Environment**
**Kernel**: https://github.com/thesofproject/linux/commits/topic/sof-dev commit:0ce0ab9
**https**://github.com/thesofproject/sof/commits/cml-010-hot-fix, commit: 0ed0dba
**tplg**: Same with FW branch + PR2358, sof-cml-rt711-rt1308-mono-rt715.tplg
**ucm**: https://github.com/thesofproject/alsa-ucm-conf/tree/master/ucm/sof-sdw-rt711-1308-715  commit: 39528c5 + https://github.com/thesofproject/alsa-ucm-conf/pull/24
**platform**: CML-U Laptop with ALC711+ ALC1308+ ALC715 in SDW mode

Logs:
[issue1.zip](https://github.com/thesofproject/linux/files/4273271/issue1.zip)

YvonneYang2, 2020-03-03:
Error seen with Kernel: sof-dev + FW: cml-010-drop-stable (commit: 5e5f69e) + PR2431

YvonneYang2, 2020-03-05:
Error not seen with sof-dev switch from 0x3 to 0xf in https://github.com/thesofproject/linux/blob/topic/sof-dev/drivers/soundwire/intel.c#L111 + cml-010-hot-fix

YvonneYang2, 2020-03-05:
Error not seen with sof-dev(0x3) commit: 24d9781 + cml-010-hotfix + f73332e

tlauda, 2020-03-17:
Fixed on branch https://github.com/thesofproject/sof/tree/cml-010-hot-fix2.

closed 2020-03-17 by tlauda