thesofproject/sof #2474: [BUG][CML]Pause/release at very fast rate will cause ipc error for 0x60060000 size 12, then DSP panic!

YvonneYang2, 2020-03-03:
**Describe the bug**
Pause/release at very fast rate on all pipline will cause ipc error for 0x60060000 size 12, then DSP panic! options snd-sof-intel-hda-common sdw_clock_stop_quirks=0x1 and without option both can reproduce it. 

**To Reproduce**
1. Boot up with pulse audio on.
2. Run command "play -Dhw:0,2 -r48000 -c2 -fs16_le wav/48California_Dreaming.wav -vv -i"
3. Do pause/release at very fast rate.(Maybe you can keep pressing the "space" key.)

**Reproduced rate**
80%
 
**Expected result**
Pause/release work fine

**Actual result**
Ipc error for 0x60060000 size 12, then DSP panic!

**Environment**	
**Kernel**: https://github.com/thesofproject/linux/commits/topic/sof-dev commit:0ce0ab9
**Kconfig**: The iso kernel kconfig merged into sof-default kconfig: [kernel_config.log](https://github.com/thesofproject/linux/files/4273306/kernel_config.log)
**https**: //github.com/thesofproject/sof/commits/cml-010-hot-fix, commit: 0ed0dba
**tplg**: Same with FW branch + PR2358, sof-cml-rt711-rt1308-mono-rt715.tplg
**ucm**: https://github.com/thesofproject/alsa-ucm-conf/tree/master/ucm/sof-sdw-rt711-1308-715  commit: 39528c5 + https://github.com/thesofproject/alsa-ucm-conf/pull/24
**platform**: CML-U Laptop with ALC711+ ALC1308+ ALC715 in SDW mode

Logs:
[issue2.zip](https://github.com/thesofproject/linux/files/4273290/issue2.zip)

YvonneYang2, 2020-03-03:
Error seen with Kernel: sof-dev + FW: cml-010-drop-stable (commit: 5e5f69e) + PR2431

YvonneYang2, 2020-03-05:
Error not seen with sof-dev switch from 0x3 to 0xf in https://github.com/thesofproject/linux/blob/topic/sof-dev/drivers/soundwire/intel.c#L111 + cml-010-hot-fix

plbossart, 2020-03-05:
> Error not seen with sof-dev switch from 0x3 to 0xf in https://github.com/thesofproject/linux/blob/topic/sof-dev/drivers/soundwire/intel.c#L111 + cml-010-hot-fix

@bardliao @RanderWang @slawblauciak I wonder if the firmware has the change needed for the MAT change? I remember very clearly there was a firmware dependency that was required to use this 0x3

YvonneYang2, 2020-03-05:
Error not seen with sof-dev(0x3) commit: 24d9781 + cml-010-hotfix + https://github.com/thesofproject/sof/commit/f73332eafedf5443f01bf26027fde2578e847ced#

bardliao, 2020-03-06:
> > Error not seen with sof-dev switch from 0x3 to 0xf in https://github.com/thesofproject/linux/blob/topic/sof-dev/drivers/soundwire/intel.c#L111 + cml-010-hot-fix
> 
> @bardliao @RanderWang @slawblauciak I wonder if the cml-010-drop-stable/ hot fix firmware has the change needed for the MAT change? I remember very clearly there was a firmware dependency that was required to use this 0x3

@plbossart Yes, you are right. That's a mismatch issue. We can either set DMAT to 0xf on kernel topic/sof-dev or set DMAT to 0x3 on FW cml-010-hotfix.

tlauda, 2020-03-17:
Fixed on branch https://github.com/thesofproject/sof/tree/cml-010-hot-fix2.

closed 2020-03-17 by tlauda