thesofproject/sof #2476: [BUG][CML]Multipipline stress test will cause I/O error, if try it again, will cause DSP panic!

YvonneYang2, 2020-03-03:
**Describe the bug**
Multipipline test will cause I/O error, if try it again, will cause DSP panic! options snd-sof-intel-hda-common sdw_clock_stop_quirks=0x1 and without option both can reproduce it. 

**To Reproduce**
1. Boot up with pulse audio on.
2. Run command "./multiplex_pipeline-test_update.sh pcm0p pcm2p pcm5p pcm1c pcm4c"
3. Check failed logs in folder named "logs"

**Reproduced rate**
100% within 5 iterations
 
**Expected result**
Multipipline stress test works fine

**Actual result**
I/O error occurred, if tried it again, will cause DSP panic

**Environment**	
**Kernel**: https://github.com/thesofproject/linux/commits/topic/sof-dev commit:0ce0ab9
**Kconfig**: The iso kernel kconfig merged into sof-default kconfig: [kernel_config.log](https://github.com/thesofproject/linux/files/4273306/kernel_config.log)
**https**: //github.com/thesofproject/sof/commits/cml-010-hot-fix, commit: 0ed0dba
**tplg**: Same with FW branch + PR2358, sof-cml-rt711-rt1308-mono-rt715.tplg
**ucm**: https://github.com/thesofproject/alsa-ucm-conf/tree/master/ucm/sof-sdw-rt711-1308-715  commit: 39528c5 + https://github.com/thesofproject/alsa-ucm-conf/pull/24
**platform**: CML-U Laptop with ALC711+ ALC1308+ ALC715 in SDW mode

Logs:
[issue4.zip](https://github.com/thesofproject/linux/files/4273392/issue4.zip)

YvonneYang2, 2020-03-03:
Error seen with Kernel: sof-dev + FW: cml-010-drop-stable (commit: 5e5f69e) + PR2431

YvonneYang2, 2020-03-05:
Error not seen with sof-dev switch from 0x3 to 0xf in https://github.com/thesofproject/linux/blob/topic/sof-dev/drivers/soundwire/intel.c#L111 + cml-010-hot-fix

YvonneYang2, 2020-03-05:
Error not seen with sof-dev(0x3) commit: 24d9781 + cml-010-hotfix + f73332e

tlauda, 2020-03-17:
Fixed on branch https://github.com/thesofproject/sof/tree/cml-010-hot-fix2.

closed 2020-03-17 by tlauda