thesofproject/sof #2478: [FEATURE] Allow componnets/buffers to XRUN

lgirdwood, 2020-03-03:
**Is your feature request related to a problem? Please describe.**
Currently if a component source buffer underruns or a component sink buffier overruns the pipeline line will enter an XRUN status and attempt recovery. This is desired in most pipelines but some topologies need to support use cases where we expect buffers to underrun or overrun.
```
Host ---> Proc----> Selector0 --> Buf0 ---- > DAI Playback
                      |
                      v
                     Buf1
                      |
                      v
Host <---------------Selector1 <----- Buf2 <----- Echo Ref DAI
```
In the example above we two host PCMs that can be independently started/stopped thereby causing buf1 to either underrun or overrun (and stop the pipelines). Buf1 should be permmitted to underrun or overrun without invoking pipeline XRUN logic and should over write oldest data (for overrun) and readback 0s (for underrun). 

**Describe the solution you'd like**
Add two new topology token for buffers.
```
TOKEN_BUFFER_UNDERRUN_NO_RECOVERY
TOKEN_BUFFER_OVERRUN_NO_RECOVERY
```
Firmware should process these tokens and if set for a buffer will not invoke pipeline XRUN status.
If buffer overflows, it should overwrite oldest data.
If buffer underruns, it should read back 0s.

lgirdwood, 2020-03-03:
Needed for https://github.com/thesofproject/sof/pull/2407 and https://github.com/thesofproject/sof/pull/2462

lgirdwood, 2020-03-03:
@akloniex @sebcarlucci fyi.

lgirdwood, 2020-03-04:
@bkokoszx can you have a chat with @akloniex - also needed for echo ref

akloniex, 2020-03-16:
@lgirdwood Regarding implementation details for the underrun case, I can think of 2 ways of approaching the subject:
1. Minimal changes, mostly focused around components' copy functions (check underrun-related flag in buffer, read as much as buffer has, fill in the zeros as needed)
2. Far-fetched changes regarding buffer read management, so that underrun would be transparent to the components, by utilizing behaviour similar to std fread()

Now, the first approach is much easier to develop and doesn't involve any big changes to the buffer API, but it comes with quite a heavy drawback - every component has to implement buffer underrun separately. The second approach is the exact opposite.

If you have any suggestions on how to approach this, how to improve on any of the above ideas, or if you have a different one of your own, then please point me in the right direction.

lbetlej, 2020-03-17:
@lgirdwood @akloniex @tlauda could you follow up on idea regarding buffer reader API ?

tlauda, 2020-03-17:
@lgirdwood Is this mechanism in some way related to the improved buffer locking API you've mentioned here https://github.com/thesofproject/sof/pull/2545#discussion_r392920305?

lgirdwood, 2020-03-18:
@tlauda yes, hooking into the buffer API is probably the best way forward. i.e.
```c
/* acquires lock for bytes to read - buffer will 0 bytes here if we underrun AND underrun flag is true */
buffer_rlock(buffer, bytes, flags);
/* acquires lock for bytes to write - buffer wont XRUN here if we overrun AND overrun flag is true */
buffer_wlock(buffer, bytes, flags);
```

akloniex, 2020-03-26:
@lgirdwood What should xrun-permitted buffers report when asked for avail/free data?
`(uint32_t)-1` seems 'most reasonable', but it doesn't feel comfortable, please clarify that for me.
If such buffer reported actual avail/free, client component would prevent xrunning in the first place, so such buffers need to alter their reports somehow.

lgirdwood, 2020-03-27:
@akloniex I think it's appropriate to say that these buffers will report they have buffer_size free and buffer_size avail.

lgirdwood, 2020-03-27:
@akloniex in fact we probably want to be more logical here with rules.

1) Buffer is full data - report buffer_size avail and period_size free
2) Buffer is empty - report period_size avail and buffer_size free
3) Buffer is not empty or full - report normally.

This way we should not block existing users.

paulstelian97, 2020-03-27:
How would the data itself be extracted from the buffers now? Until now it was simply by requesting a pointer for each sample from the buffer but now I feel like this approach may not work properly. Did I miss a PR which changes this logic?

lgirdwood, 2020-03-27:
It will us the same approach as the DMA trace buffer, i.e. oldest data is discarded

akloniex, 2020-03-27:
@paulstelian97 @lgirdwood wrote how this situation will be handled, few comments above:

> ```c
> /* acquires lock for bytes to read - buffer will 0 bytes here if we underrun AND underrun flag is true */
> buffer_rlock(buffer, bytes, flags);
> /* acquires lock for bytes to write - buffer wont XRUN here if we overrun AND overrun flag is true */
> buffer_wlock(buffer, bytes, flags);
> ```

paulstelian97, 2020-03-27:
So the buffer is modified accordingly.

cujomalainey, 2020-07-31:
This has been completed correct?

lgirdwood, 2020-08-04:
@akloniex @juimonen is this now completed ?

lgirdwood, 2020-08-18:
@bkokoszx @slawblauciak @ktrzcinx can we close this now ? I think all the parts are merged now ?

ktrzcinx, 2020-08-18:
I'have already use it, so I think it's ready

bkokoszx, 2020-08-18:
@lgirdwood 
I think we can close this via #2745

closed 2020-08-18 by cujomalainey