thesofproject/sof #2482: [BUG][APL][PCM512X] Xrun detected when do mutil-playback stress test

dengyangchao, 2020-03-04:
**Describe the bug**
Xrun detected when do mutil-playback stress test.

**To Reproduce**
1. Reboot system via "sudo reboot"
2. Cd /sof-test/test-case
3. Execute command "export TPLG=/lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg
4. Run stress test via command " ./multiple-pipeline-playback.sh -l 2000"

**Reproduction Rate**
1/519
Tried to run 2000 times and fail at 519th

**Expected behavior**
Mutil-playback stress test normal, and no error occur

**Impact**
Xrun detected and test fail

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev  65f9d39d
    * SOF: master    4f3f93d9
2) Name of the topology file
    * Topology: sof-apl-pcm512x.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: APL-UP2 with codec PCM512X in I2S mode

**Screenshots or console output**
[image]

**No dmesg error**

 *sof-logger-etrace*
```
  CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          DMA         5500324119.739584              nan intel/cavs/hda-dma.c:789 	hda_dma_link_check_xrun() error: xrun detected
    0      1         COMP 2.14    5500324123.854167         4.114583      src/audio/dai.c:639 	dai_report_xrun() error: underrun due to no data available
    0      1         COMP 2.14    5500324127.291667         3.437500      src/audio/dai.c:902 	comp_underrun() error: dev->comp.id = 14, source->avail = 32, copy_bytes = 0
    0      1         PIPE         5500324169.166667        41.875000 src/audio/pipeline.c:789 	pipeline_copy() error: ret = -61, start->comp.id = 14, dir = 1
    0      1         PIPE 2.15    5500324172.916667         3.750000 src/audio/pipeline.c:961 	pipeline_task(): xrun recover failed! pipeline will be stopped!
```

[dmesg.txt](https://github.com/thesofproject/sof/files/4286580/dmesg.txt)
[etrace.txt](https://github.com/thesofproject/sof/files/4286582/etrace.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/4286583/slogger.txt)

add test scripts here:
[sof-test-master.zip](https://github.com/thesofproject/sof/files/4286608/sof-test-master.zip)

lgirdwood, 2020-03-04:
@dengyangchao can you confirm the number of active streams that are being tested here. I can see four from the screenshot.

dengyangchao, 2020-03-05:
@lgirdwood Yes, the script use four streams for mutil-playback test, HW:0,0 HW:0,1  HW:0,2 and HW:0,3

tlauda, 2020-03-10:
@dengyangchao Please verify with #2512.

dengyangchao, 2020-03-11:
@tlauda Tried with #2512 , run 2000 times and issue can't reproduce

tlauda, 2020-03-11:
Fixed with #2512.

closed 2020-03-11 by tlauda