thesofproject/sof #2485: [BUG] cml-010-drop-stable patches missing from master.

lgirdwood, 2020-03-04:
The following patches are missing from master and need to be merged.
```git log --oneline --graph --decorate --abbrev-commit origin/master..origin/cml-010-drop-stable``` 
* 5e5f69eb (origin/cml-010-drop-stable) interrupt: dynamic clock change for waiti
* cc4c356d workaround: disable hifi3 pcm conversion

slawblauciak, 2020-03-09:
I don't think we want HIFI3 PCM conversion disabled by default on master.
@jajanusz I don't suppose the dynamic clock change is required either?

paulstelian97, 2020-03-09:
HIFI3 conversion may no longer need to be disabled as PR#2499 is now merged.

lgirdwood, 2020-03-09:
Updated, just the clock change now.

jajanusz, 2020-03-13:
5e5f69e Janusz Jankowski interrupt: dynamic clock change for waiti ->https://github.com/thesofproject/sof/pull/2326

closed 2020-03-13 by jajanusz