thesofproject/sof #2491: [BUG][APL][PCM512X]Ipc timed out for 0x60050000 size 12 when do capture test on HW:0,0

dengyangchao, 2020-03-06:
**Describe the bug**
Ipc timed out for 0x60050000 size 12 when do capture test on HW:0,0
Use "crtl+z" and "fg" to change foreground and background also can reproduce the issue.
When issue occur, sof can't back to work, wait a few minates system will get freeze.

**To Reproduce**
1. Reboot via "sudo reboot"
2. Use command "cd /sof-test/test-case" to use sof-test  test scripts.
3. Execute "export TPLG=/lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg"
4. Test capture via command " ./check-capture.sh -d 1 -l 100 -r 1"

**Reproduction Rate**
Tried 3 times
1/4    fail at 4th
1/1    fail at 1st
1/1    fail at 1st

**Expected behavior**
Capture test work well

**Impact**
Ipc tomed out occur

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev  ac3abc56
    * SOF: master   9ebb2881
2) Name of the topology file
    * Topology: sof-apl-pcm512x.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: APL-UP2 with codec pcm512x in I2S mode

Dmesg
```
cd-apl-up2-pcm512x-07 kernel: [   54.296146] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
cd-apl-up2-pcm512x-07 kernel: [   54.296150] sof-audio-pci 0000:00:0e.0: period_bytes:0x3fc0
cd-apl-up2-pcm512x-07 kernel: [   54.296151] sof-audio-pci 0000:00:0e.0: periods:4
cd-apl-up2-pcm512x-07 kernel: [   54.296160] sof-audio-pci 0000:00:0e.0: stream_tag 2
cd-apl-up2-pcm512x-07 kernel: [   54.296335] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
cd-apl-up2-pcm512x-07 kernel: [   54.297292] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
cd-apl-up2-pcm512x-07 kernel: [   54.297301] sof-audio-pci 0000:00:0e.0: pcm: stream dir 1, posn mailbox offset is 790984
cd-apl-up2-pcm512x-07 kernel: [   54.298024] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 1 cmd 1
cd-apl-up2-pcm512x-07 kernel: [   54.298582] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x2024001e successful
cd-apl-up2-pcm512x-07 kernel: [   54.298590] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
cd-apl-up2-pcm512x-07 kernel: [   54.298799] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
cd-apl-up2-pcm512x-07 kernel: [   54.806964] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 1 cmd 0
cd-apl-up2-pcm512x-07 kernel: [   54.806979] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
cd-apl-up2-pcm512x-07 kernel: [   55.310324] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60050000 size 12
cd-apl-up2-pcm512x-07 kernel: [   55.310337] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
cd-apl-up2-pcm512x-07 kernel: [   55.310346] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
cd-apl-up2-pcm512x-07 kernel: [   55.310395] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
cd-apl-up2-pcm512x-07 kernel: [   55.310407] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000003 rirb 00
cd-apl-up2-pcm512x-07 kernel: [   55.310412] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
cd-apl-up2-pcm512x-07 kernel: [   55.310421] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
cd-apl-up2-pcm512x-07 kernel: [   55.310427] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
cd-apl-up2-pcm512x-07 kernel: [   55.310446] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20240000 successful
cd-apl-up2-pcm512x-07 kernel: [   55.310455]  Port5: ASoC: trigger FE cmd: 0 failed: -110
```

[cmd.txt](https://github.com/thesofproject/sof/files/4296865/cmd.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/4296866/dmesg.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/4296867/slogger.txt)

dengyangchao, 2020-03-06:
Caused by master c60bcd0 , @tlauda please help to check

Liviali155, 2020-03-06:
related with https://github.com/thesofproject/sof/issues/2493,caused by the same Firmware commit

lgirdwood, 2020-03-06:
@singalsu fyi - the HiFi3 issue is back. Can we narrow this down to a particular conversion or does it apply to all types ?

tlauda, 2020-03-06:
@dengyangchao Please try with #2499.

dengyangchao, 2020-03-09:
@tlauda Tried with #2499 , issue still can be reproduced.  Fail at 136th.
There is no sof trace because #2507

[dmesg.log](https://github.com/thesofproject/sof/files/4304933/dmesg.log)

tlauda, 2020-03-09:
@dengyangchao What is the capture format and what is the dai format?

dengyangchao, 2020-03-09:
@tlauda 
Add info 
```
2020-03-09 07:01:31 UTC [COMMAND] arecord -Dhw:0,0 -r 48000 -c 2 -f S16_LE -d 1 /dev/null -v -q
Hardware PCM card 0 'sof-pcm512x' device 0 subdevice 0
Its setup is:
  stream       : CAPTURE
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16368
  period_size  : 4080
  period_time  : 85000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4080
  period_event : 0
  start_threshold  : 1
  stop_threshold   : 16368
  silence_threshold: 0
  silence_size : 0
  boundary     : 9214364837600034816
  appl_ptr     : 0
  hw_ptr       : 0
```
and the platform is APL-UP2 with codec PCM512X in I2S mode

dengyangchao, 2020-03-11:
After applied #2519 , tried 1000 times issue can't be reproduced.

PR #2519 (2020-03-11): pcm_converter: fix loops condition
Fixes loops condition in most of the PCM conversion functions.
Number of samples is unsigned, so current condition caused underflow.

Fixes #2491, #2493.
closed 2020-03-11 by tlauda