thesofproject/sof #2492: [BUG][TGL]"ASoC: trigger FE cmd: 1 failed: -110" display after boot up system.

sinahuang, 2020-03-06:
**Describe the bug**"ASoC: trigger FE cmd: 1 failed: -110" display after boot up system, and then system can freeze.

**To Reproduce**
1. Install FW and Kernel 
2.Boot up system.

**Reproduction Rate**
5/5

**Expected behavior**
System boot up successfully.

**Impact**
"ASoC: trigger FE cmd: 1 failed: -110" display after boot up system and system can be freezed.

**Test recipe**

Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev commit:ac3abc5
FW:  https://github.com/thesofproject/sof/commits/master commit:  0c7514e
Tplg: Same with FW branch, sof-tgl-rt711-i2s-rt1308.tplg 
Platform:TGL-U RVP with codec ALC711 in SDW mode and  RT1308 in I2S mode

sinahuang, 2020-03-06:
Bi-sect result:
Regression caused by FW commit: https://github.com/thesofproject/sof/commit/034c54a16a6536352dc85914517a49fd7ca8f0fd
@lbetlej   please check it.

emilchudzik, 2020-03-09:
We have still issue with proper setup TGL and SDW
- reworks SDW on ALC711  and I2S on 1308 - done
- BIOS settings: 

> Set Audio Link Mode to Advanced Link Config
> Enable: HDA link,  DMIC #0, SSP #2, SDW #1   (ALC1308 is on SSP #2, ALC711 is on SDW #1）
> Enter HD Audio DSP Features Configuration, Enable I2S and select I2S Codec: Realtek ALC1308 

- kernel: kernel: 24d9781/sof-dev
- FW: c35e1aa6/master or c60bcd0a/master
- acpi-scripts for TGL added: https://github.com/thesofproject/acpi-scripts/tree/master/TGL-RVP

there is still errors from SDW:
```
[    5.460342] intel-master sdw-master-0: intel_init: start
[    5.460438] intel-master sdw-master-0: intel_init: shim_init clock stop 0
...
[    5.465081] intel-master sdw-master-0: Slave Entry not found
[    5.465254] intel-master sdw-master-0: Slave Entry not found
...
[    5.467548] sdw_rt711_rt1308_rt715 sdw_rt711_rt1308_rt715: snd_soc_register_card failed -517

```

full dmesg after boot OS: 
[dmesg.txt](https://github.com/thesofproject/sof/files/4306127/dmesg.txt)

Who can debug it due to reproduce issue from topic ?

emilchudzik, 2020-03-09:
Thanks @RanderWang for bring up SDW on my machine.

We have no repro on the newest FW top:
FW: 0996f794a/master
kernel: 24d9781/sof-dev
tplg: Same with FW branch, sof-tgl-rt711-i2s-rt1308.tplg
platform: TGL-U RVP with codec ALC711 in SDW mode and RT1308 in I2S mode

I check also with disabled option in make menuconfig during FW building:
`Patform -> [ ] Enable low power ring oscillator as the default clock`

sinahuang, 2020-03-10:
Issue still can be reproduced on my side with master (commit:  0996f794)+sof-dev  (commit:ac3abc5)

lbetlej, 2020-03-10:
@sinahuang Could you check FW build with disabled option in make menuconfig during FW building:
Platform -> [ ] Enable low power ring oscillator as the default clock ? As @emilchudzik reported we have no repro.

sinahuang, 2020-03-10:
Verified it with disabled option in make menuconfig during FW building, issue can't be reproduced.

tlauda, 2020-03-10:
@sinahuang @lbetlej @emilchudzik Spurious DMIC overrun interrupt led to infinite IRQ handler triggering. With 30492d73f919bd925198dd234fb176f47900ee79 we are now clearing the bit. I think issue can be closed.

closed 2020-03-10 by tlauda