thesofproject/sof #2493: [BUG][APL][PCM512X]Ipc timed out for 0x60010000 size 108 when do muti-pipeline tress test

dengyangchao, 2020-03-06:
**Describe the bug**
Ipc timed out for 0x60010000 size 108 when do muti-pipeline tress test.
When issue occur, sof can't back to work, wait a few minates system will get freeze.

**To Reproduce**
1. Reboot via "sudo reboot"
2. Use command "cd /sof-test/test-case" to use sof-test test scripts.
3. Execute "export TPLG=/lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg"
4. Mutil-pipeline test via command " ./mutiple-pipeline-capture.sh -l 100"

**Reproduction Rate**
Tried 3 times
1/3    fail at 3rd
1/2    fail at 2nd
1/4     fail at 4th

**Expected behavior**
Mutil-pipeline test normal, and no error
**Impact**
Caused ipc timed out

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
              *Kernel: sof-dev ac3abc56
              *SOF: master 9ebb288
2) Name of the topology file
              *Topology: sof-apl-pcm512x.tplg
3) Name of the platform(s) on which the bug is observed.
               *Platform: APL-UP2 with codec pcm512x in I2S mode

Dmesg
```
cd-apl-up2-pcm512x-07 kernel: [   63.352023] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
cd-apl-up2-pcm512x-07 kernel: [   63.352030] sof-audio-pci 0000:00:0e.0: pcm: stream dir 1, posn mailbox offset is 790984
cd-apl-up2-pcm512x-07 kernel: [   63.352498] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 1 cmd 1
cd-apl-up2-pcm512x-07 kernel: [   63.353048] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x2024001e successful
cd-apl-up2-pcm512x-07 kernel: [   63.353054] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
cd-apl-up2-pcm512x-07 kernel: [   63.353480] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x60040000: GLB_STREAM_MSG: TRIG_START
cd-apl-up2-pcm512x-07 kernel: [   63.360771] sof-audio-pci 0000:00:0e.0: pcm: open stream 5 dir 1
cd-apl-up2-pcm512x-07 kernel: [   63.360776] sof-audio-pci 0000:00:0e.0: period min 192 max 16384 bytes
cd-apl-up2-pcm512x-07 kernel: [   63.360778] sof-audio-pci 0000:00:0e.0: period count 2 max 16
cd-apl-up2-pcm512x-07 kernel: [   63.360779] sof-audio-pci 0000:00:0e.0: buffer max 65536 bytes
cd-apl-up2-pcm512x-07 kernel: [   63.360930] sof-audio-pci 0000:00:0e.0: pcm: hw params stream 5 dir 1
cd-apl-up2-pcm512x-07 kernel: [   63.360934] sof-audio-pci 0000:00:0e.0: generating page table for 000000003239518b size 0xffc0 pages 16
cd-apl-up2-pcm512x-07 kernel: [   63.360941] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x340000 successful
cd-apl-up2-pcm512x-07 kernel: [   63.360956] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
cd-apl-up2-pcm512x-07 kernel: [   63.360959] sof-audio-pci 0000:00:0e.0: period_bytes:0x3fc0
cd-apl-up2-pcm512x-07 kernel: [   63.360961] sof-audio-pci 0000:00:0e.0: periods:4
cd-apl-up2-pcm512x-07 kernel: [   63.360968] sof-audio-pci 0000:00:0e.0: stream_tag 3
cd-apl-up2-pcm512x-07 kernel: [   63.360976] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
cd-apl-up2-pcm512x-07 kernel: [   63.858626] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 0 dir 1 cmd 0
cd-apl-up2-pcm512x-07 kernel: [   63.865824] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60010000 size 108
cd-apl-up2-pcm512x-07 kernel: [   63.865835] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
cd-apl-up2-pcm512x-07 kernel: [   63.865844] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
cd-apl-up2-pcm512x-07 kernel: [   63.865891] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
cd-apl-up2-pcm512x-07 kernel: [   63.865902] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000003 rirb 00
cd-apl-up2-pcm512x-07 kernel: [   63.865907] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
cd-apl-up2-pcm512x-07 kernel: [   63.865914] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
cd-apl-up2-pcm512x-07 kernel: [   63.865920] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
cd-apl-up2-pcm512x-07 kernel: [   63.865947] sof-audio-pci 0000:00:0e.0: error: hw params ipc failed for stream 3
cd-apl-up2-pcm512x-07 kernel: [   63.865954] sof-audio-pci 0000:00:0e.0: ASoC: 0000:00:0e.0 hw params failed: -110
cd-apl-up2-pcm512x-07 kernel: [   63.865965]  DMIC: ASoC: hw_params FE failed -110
```

[cmd.txt](https://github.com/thesofproject/sof/files/4297022/cmd.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/4297023/dmesg.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/4297025/slogger.txt)

add test scripts here:
[sof-test-master.zip](https://github.com/thesofproject/sof/files/4297029/sof-test-master.zip)

dengyangchao, 2020-03-06:
Caused by master c60bcd0 , @tlauda please help to check

tlauda, 2020-03-06:
@dengyangchao What is the host vs dai format of the last started stream? I'm trying to find out which pcm conversion method is faulty.

lgirdwood, 2020-03-06:
@singalsu fyi

tlauda, 2020-03-06:
@dengyangchao Please try with #2499.

dengyangchao, 2020-03-09:
@tlauda After applied #2499 issue still can be reproduced 

[dmesg.txt](https://github.com/thesofproject/sof/files/4305040/dmesg.txt)

tlauda, 2020-03-11:
@dengyangchao Please verify with #2519. It's already merged.

dengyangchao, 2020-03-12:
Tried 2000 times issue can't be reproduced, close

closed 2020-03-12 by dengyangchao