thesofproject/sof #2494: [BUG][GLK]"ipc timed out for 0x60040000 size 12" when run Check-playback: Stress - 100x

Liviali155, 2020-03-06:
**Describe the bug**
"ipc timed out for 0x60040000 size 12" when run Check-playback: Stress - 100x
After error occured,all the pipeline can not work,reboot can recovery

**To Reproduce**
1.cd sof-test(as attachment)
2.cd test-case
3. ./check-playback.sh -d 1 -l 100 -r 1

**Reproduction Rate**
3/3

**Expected behavior**
There is no I/O error when run playback stress

**Actual result**
I/O error,dmesg show "ipc timed out for 0x60040000 size 12"

[image]

<details>
 <summary>dmesg_error.log</summary>

````
kernel: [  437.071678] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 5 dir 0 cmd 1
kernel: [  437.072214] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x14001e successful
kernel: [  437.072218] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60040000: GLB_STREAM_MSG: TRIG_START
kernel: [  437.578315] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60040000 size 12
kernel: [  437.578320] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
kernel: [  437.578326] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
kernel: [  437.578368] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
kernel: [  437.578374] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
kernel: [  437.578375] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
kernel: [  437.578379] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
kernel: [  437.578380] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
kernel: [  437.578387]  HDMI1: ASoC: trigger FE cmd: 1 failed: -110
kernel: [  437.578394] sof-audio-pci 0000:00:0e.0: error: trace IO error
kernel: [  437.578397] sof-audio-pci 0000:00:0e.0: pcm: trigger stream 5 dir 0 cmd 0
kernel: [  437.578401] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
kernel: [  438.082314] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60050000 size 12
kernel: [  438.082319] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
kernel: [  438.082324] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
kernel: [  438.082365] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
kernel: [  438.082371] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
kernel: [  438.082372] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
kernel: [  438.082376] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
kernel: [  438.082378] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
kernel: [  438.082888] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20140000 successful
kernel: [  438.082892]  HDMI1: ASoC: trigger FE cmd: 0 failed: -110
kernel: [  438.690526] sof-audio-pci 0000:00:0e.0: pcm: free stream 5 dir 0
kernel: [  438.690533] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
kernel: [  439.194318] sof-audio-pci 0000:00:0e.0: error: ipc timed out for 0x60030000 size 12
kernel: [  439.194323] sof-audio-pci 0000:00:0e.0: info: preventing DSP entering D3 state to preserve context
kernel: [  439.194328] sof-audio-pci 0000:00:0e.0: status: fw entered - code 00000005
kernel: [  439.194370] sof-audio-pci 0000:00:0e.0: error: unexpected fault 0x00000000 trace 0x00004000
kernel: [  439.194376] sof-audio-pci 0000:00:0e.0: error: hda irq intsts 0x00000000 intlctl 0xc0000001 rirb 00
kernel: [  439.194377] sof-audio-pci 0000:00:0e.0: error: dsp irq ppsts 0x00000000 adspis 0x00000000
kernel: [  439.194381] sof-audio-pci 0000:00:0e.0: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
kernel: [  439.194382] sof-audio-pci 0000:00:0e.0: error: waking up any trace sleepers
kernel: [  439.194458] sof-audio-pci 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG	
    
````
</details>

No error softrace

GLK Chrome with onboard codec DA7219 in I2S mode
Firmware: 9ebb2881 https://github.com/thesofproject/sof/commits/master
Kernel: ac3abc56 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-glk-da7219.tplg same as firmware
[dmesg.txt](https://github.com/thesofproject/linux/files/4296962/dmesg.txt)
[sof_trace.log](https://github.com/thesofproject/linux/files/4296964/sof_trace.log)
[sof-test.zip](https://github.com/thesofproject/linux/files/4297007/sof-test.zip)

lgirdwood, 2020-03-06:
@Liviali155 I assume you mean you will bisect it ?

Liviali155, 2020-03-06:
> @Liviali155 I assume you mean you will bisect it ?

Caused by master c60bcd0 , @tlauda please help to check

Liviali155, 2020-03-06:
related with #2493 by the same Firmware commit

lgirdwood, 2020-03-06:
@singalsu fyi

tlauda, 2020-03-06:
@Liviali155 What is the PCM conversion here? What is the format for host and what is for dai?

Liviali155, 2020-03-06:
> @Liviali155 What is the PCM conversion here? What is the format for host and what is for dai?

Issue occured when aplay via HDMI(0,5) or (0,6)

```
2020-03-06 00:14:52 UTC [REMOTE_COMMAND] aplay -Dhw:0,6 -r 48000 -c 2 -f S16_LE -d 1 /dev/zero -v -q
Hardware PCM card 0 'sof-glkda7219max' device 6 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 16368
  period_size  : 4080
  period_time  : 85000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 4080
  period_event : 0
  start_threshold  : 16368
  stop_threshold   : 16368
  silence_threshold: 0
  silence_size : 0
  boundary     : 9214364837600034816
  appl_ptr     : 0
  hw_ptr       : 0
aplay: pcm_write:2011: write error: Input/output error
2020-03-06 00:14:55 UTC [REMOTE_ERR] aplay on PCM hw:0,6 failed at 57/100.
2020-03-06 00:14:57 UTC [REMOTE_INFO] Test Result: FAIL!
```
[image]

Liviali155, 2020-03-06:
Tried with https://github.com/thesofproject/sof/pull/2499 , issue can not  be reproduced

slawblauciak, 2020-03-09:
@Liviali155 PR has been merged, are we good to close this?

Liviali155, 2020-03-10:
PR #2499 merged,closed

closed 2020-03-10 by Liviali155