thesofproject/sof #2502: How do I install SOF under Ubuntu?

katsar0v, 2020-03-06:
This not a bug, but rather a feature request (maybe for docs). I have Dell Inspiron 7590 which has no microphone working when a fresh Ubuntu is installed (the 3.5mm jack does not work as well). ArchLinux -https://wiki.archlinux.org/index.php/Dell_Inspiron_15_(7590) says it works with the newest sof firmware, but how do I install it under Ubuntu?
I hope not every new Dell user has to clone this repo and clone it in order to get the mic working, but it would be much easier to install it via ppa or a .deb file. Thanks in advance, in case somebody decides to help me out here!

plbossart, 2020-03-06:
@katsar0v I assume you meant you have Ubuntu 19.10 with its v5.3 kernel, yes?

In that case you should only need firmware and topology files. If I am correct, would you be willing to test some of our install procedures, we are still in need of downstream beta testers.

Thanks!

katsar0v, 2020-03-06:
@plbossart I rolled back to Ubuntu 18.04 LTS but have `5.3.0-40-generic` (from `uname -r`)

I tried building from scratch - https://thesofproject.github.io/latest/getting_started/build-guide/build-from-scratch.html but it is really complicated and I would like something more simpler. 

As for beta testers - I can do this, as long as I don't have to re-install my whole Ubuntu on a daily basis, it is fine for me

I have also read that the kernels from 5.2+ also support SOF, but am not sure what that means. ([link](https://www.phoronix.com/scan.php?page=news_item&px=Intel-Sound-Open-Firmware-5.2))

plbossart, 2020-03-06:
ok, let's try to help, one step at a time.

Step1: make sure the SOF driver is selected:

with 5.3, you need to do this:

gedit /etc/modprobe.d/alsa-base.conf

add 
blacklist snd-hda-intel
blacklist snd-soc-skl

reboot and share the alsa-info.sh link

katsar0v, 2020-03-06:
Here you go: http://alsa-project.org/db/?f=6112e1ee9ab2014a99f3cc1c4a726575ad104943

plbossart, 2020-03-06:
ok, so the driver is already loaded and the card created.

IIRC with older solutions based on the 1.3 firmware we had a very low gain for the dmic, and it was barely usable. 
can you play with alsamixer and boot the gains for PGA10.0 and PGA11.0 by 20dB?

https://github.com/thesofproject/sof/issues/1675#issuecomment-519915444

katsar0v, 2020-03-06:
[image]

This is how my `alsamixer` works. When I visit a voice recorder online, it records the taps on my laptop

plbossart, 2020-03-06:
@katsar0v F4 to select capture and scroll right until you see PGA10.0 and PGA11.0, then boost them to the max

katsar0v, 2020-03-06:
[image]

this is what I could see,  the 4 columns on the right were on 50%

plbossart, 2020-03-06:
forget PulseAudio, can you test with 

arecord -Dhw:0,6 -c2 -r48000 -f S32_LE -d 10 mytest.wav

if this doesn't work try with -c4 and S24_LE, I don't recall the options

and input this into audacity

katsar0v, 2020-03-06:
> forget PulseAudio, can you test with
> 
> arecord -Dhw:0,6 -c2 -r48000 -f S32_LE -d 10 mytest.wav
> 
> if this doesn't work try with -c4 and S24_LE, I don't recall the options
> 
> and input this into audacity

Damn this worked! `mytest.wav` recorder my voice. Should I upload the file? I am not sure what to do with it in audacity

plbossart, 2020-03-06:
> > forget PulseAudio, can you test with
> > arecord -Dhw:0,6 -c2 -r48000 -f S32_LE -d 10 mytest.wav
> > if this doesn't work try with -c4 and S24_LE, I don't recall the options
> > and input this into audacity
> 
> Damn this worked! `mytest.wav` recorded my voice. Should I upload the file? I am not sure what to do with it in audacity

I meant look at the waveform, but if you can hear your voice then that's fine.

So what I think is happening is that your UCM file is wrong.

That should be in /usr/share/alsa/ucm/sof-hda something.

Can you attach the HiFi.conf it here?

katsar0v, 2020-03-06:
> I meant look at the waveform, but if you can hear your voice then that's fine.
>
> So what I think is happening is that your UCM file is wrong.
>
> That should be in /usr/share/alsa/ucm/sof-hda [...]

The quality from the `.wav` is really good, I don't think audacity is needed. Here is a screenshot as I could not find sof-hda something:

[image]

plbossart, 2020-03-06:
ok, so you don't have a UCM file, that explains why things are broken.

I will have to include @juimonen on this one. Jaska, do you know where we can find a UCM file with hdaudio + mic support that would work with the 5.3 kernel, 1.3 firmware and Ubuntu 18.04 LTS?

katsar0v, 2020-03-06:
@plbossart thanks so much for the help! There is hope for my mic and I could hear my voice (in a super good quality) 

Just guessing - https://github.com/plbossart/UCM/tree/master/skl_hda_card this has the same name as my card from here:

[image]

plbossart, 2020-03-06:
if you want to experiment before Monday, I took the last UCM file I could find

alsa-ucm-conf$ git checkout 3a2d80ecb74ee4cd02449873ece0407c7da0da4e

[sof-hda-dsp.tar.gz](https://github.com/thesofproject/sof/files/4300671/sof-hda-dsp.tar.gz)

copy this to /usr/share/alsa/ucm and reboot, you should see the dmics. There might be a dependency on alsa-lib though

plbossart, 2020-03-06:
> @plbossart thanks so much for the help! There is hope for my mic and I could hear my voice (in a super good quality)

Intel makes good products :-)

> Just guessing - https://github.com/plbossart/UCM/tree/master/skl_hda_card this has the same name as my card from here:

don't use this, I no longer maintain it.

katsar0v, 2020-03-06:
[image]

[image]

Nothing changed after reboot. Do you think I should install `alsa-lib` ?

plbossart, 2020-03-06:
@katsar0v we'd need to see the pulseaudio logs then.

The best would be to edit /etc/pulse/client.conf (or daemon.conf), then set autospawm = no

then reboot, open a terminal, and kill all pulse audio instances, if any

then run pulseaudio -vvv &> log.txt, and attach the log.txt. That should tell us what's wrong in the UCM file.

katsar0v, 2020-03-06:
[image]

Here you go:

[log.txt](https://github.com/thesofproject/sof/files/4300780/log.txt)

plbossart, 2020-03-06:
> I: [pulseaudio] (alsa-lib)utils.c: could not open configuration file /usr/share/alsa/ucm/DellInc.-Inspiron7590--0JKGD4/DellInc.-Inspiron7590--0JKGD4.conf
> I: [pulseaudio] (alsa-lib)parser.c: error: could not parse configuration for card DellInc.-Inspiron7590--0JKGD4

this is fine, not a real error since there's a fallback

> I: [pulseaudio] (alsa-lib)utils.c: could not open configuration file /usr/share/alsa/ucm/sof-skl_hda_card/sof-skl_hda_card.conf
> I: [pulseaudio] (alsa-lib)parser.c: error: could not parse configuration for card sof-skl_hda_card
> I: [pulseaudio] (alsa-lib)main.c: error: failed to import sof-skl_hda_card use case configuration -2
> I: [pulseaudio] alsa-ucm.c: UCM not available for card sof-skl_hda_card
> D: [pulseaudio] conf-parser.c: Parsing configuration file '/usr/share/pulseaudio/alsa-mixer/profile-sets/default.conf'

ok so this is likely because there's a name mismatch. I know there were changes in the _ and - used for the card name, so you probably need to match the name of the card with the name of the UCM file.

That's just silly I know :-)

Make sure all files match the card name

katsar0v, 2020-03-06:
This is what I see after changing the files:

[image]
[image]

Log.txt - https://file.io/v2XXbM

plbossart, 2020-03-06:
ok let's see on Monday what @juimonen thinks. Have a nice week-end.

juimonen, 2020-03-07:
@katsar0v, we should have some working ucm for 18 LTS ubuntu's here:
https://github.com/thesofproject/alsa-ucm-conf/tree/sof_ucm1

It should at least get the ucm loaded, if you could try that, let's see after that what happens :) 

Your Pulseaudio will use software volumes, which should work (I think). If you want hw volumes and possible mute leds blinking you need to update pulseaudio and sof topology, but that's another story then.

katsar0v, 2020-03-07:
@juimonen I can see the input now (and it reacts on my voice):

[image]

Recording voice works in firefox:

[image]

So I think we solved this! For some reason it does not work with Chrome, but I will inspect this