thesofproject/sof #2511: [BUG] [TGL] WOV capture fails for the second time,

RDharageswari, 2020-03-09:
**Describe the bug**
WOV capture fails for the second time.

**To Reproduce**
1)Start the WOV capture:
   arecord -Dhw:0,7 -M -N -c 2 -f S16_LE -r 16000 --buffer-size=64000 tmp.wav -vvv
2)Start tapping on the DMIC and the WOV capture starts.
3)Press Ctrl+C
4)Second time start the WOV capture.
localhost ~ #
 tmp.wav -vvv arecord -Dhw:0,7 -M -N -c 2 -f S16_LE -r 16000 --buffer-size=64000
Recording WAVE 'tmp.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
arecord: set_params:1403: Unable to install hw params:
ACCESS:  MMAP_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 16000
PERIOD_TIME: 999000
PERIOD_SIZE: 15984
PERIOD_BYTES: 63936
PERIODS: (4 5)
BUFFER_TIME: 3999000
BUFFER_SIZE: 63984
BUFFER_BYTES: 255936
TICK_TIME: 0

   This capture shows failure. From the failure logs, it looks like the first capture stream did not freed 
   completely.(The allocated stream1 is not freed) and hence the second time capture does not start.

**Reproduction Rate**
Always

Keyword:
after Ctrl+C of first WOV capture
[  376.026507] sof-audio-pci 0000:00:1f.3: Turning i915 HDAC power 0
[  376.026531] sof-audio-pci 0000:00:1f.3: error: transition from 0 to 4 not allowed
[  376.026532] sof-audio-pci 0000:00:1f.3: error: failed to power down DSP during suspend -22
[  376.026543] sof-audio-pci 0000:00:1f.3: can't suspend (snd_sof_runtime_suspend [snd_sof] returned -22)
PFA the kernel & firmware logs.

[first_wov_failure_fw_logs.txt](https://github.com/thesofproject/sof/files/4309445/first_wov_failure_fw_logs.txt)
[first_wov_failure_kernellogs.txt](https://github.com/thesofproject/sof/files/4309446/first_wov_failure_kernellogs.txt)
[second_capture_failure_kernel_logs.txt](https://github.com/thesofproject/sof/files/4309447/second_capture_failure_kernel_logs.txt)

RDharageswari, 2020-03-09:
[sof-tgl-max98357a-rt5682.m4.txt](https://github.com/thesofproject/sof/files/4309523/sof-tgl-max98357a-rt5682.m4.txt)

RDharageswari, 2020-03-10:
[WOV_Working.zip](https://github.com/thesofproject/sof/files/4314967/WOV_Working.zip)
Adding the topology patches

lbetlej, 2020-03-12:
@RDharageswari  Is the issue reproduced with arecord option 
-d, --duration=# Interrupt after # seconds ?

RDharageswari, 2020-03-12:
Yes i could reproduce the issue with 10 seconds also:
arecord -Dhw:0,7 -M -N -c 2 -f S16_LE -r 16000 --buffer-size=64000 tmp.wav -vvv -d 10

dmesg logs: first time:
[ 2866.862857] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[ 2866.862902] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
[ 2866.862910] sof-audio-pci 0000:00:1f.3: Turning i915 HDAC power 0
[ 2866.862953] sof-audio-pci 0000:00:1f.3: error: transition from 0 to 4 not allowed
[ 2866.862954] sof-audio-pci 0000:00:1f.3: error: failed to power down DSP during suspend -22
[ 2866.862963] sof-audio-pci 0000:00:1f.3: can't suspend (snd_sof_runtime_suspend [snd_sof] returned -22)
localhost ~ # arecord -Dhw:0,7 -M -N -c 2 -f S16_LE -r 16000 --buffer-size=64000 tmp.wav -vvv -d 10
second time capture fails:
localhost ~ # arecord -Dhw:0,7 -M -N -c 2 -f S16_LE -r 16000 --buffer-size=64000 tmp.wav -vvv -d 10
Recording WAVE 'tmp.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
arecord: set_params:1403: Unable to install hw params:
ACCESS:  MMAP_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 16000
PERIOD_TIME: 999000
PERIOD_SIZE: 15984
PERIOD_BYTES: 63936
PERIODS: (4 5)
BUFFER_TIME: 3999000
BUFFER_SIZE: 63984
BUFFER_BYTES: 255936
TICK_TIME: 0

[ 2880.905186] sof-audio-pci 0000:00:1f.3: pcm: open stream 7 dir 1
[ 2880.905189] sof-audio-pci 0000:00:1f.3: period min 192 max 256000 bytes
[ 2880.905190] sof-audio-pci 0000:00:1f.3: period count 2 max 160
[ 2880.905191] sof-audio-pci 0000:00:1f.3: buffer max 1280000 bytes
[ 2880.905417] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 7 dir 1
[ 2880.905421] sof-audio-pci 0000:00:1f.3: generating page table for 000000001e00a2d0 size 0x3e7c0 pages 63
[ 2880.905436] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[ 2880.906741] sof-audio-pci 0000:00:1f.3: error: stream reset failed
[ 2880.906743] sof-audio-pci 0000:00:1f.3: error: hdac prepare failed: ffffff92
[ 2880.906744] sof-audio-pci 0000:00:1f.3: error: platform hw params failed
[ 2880.906747] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
[ 2880.906749]  DMIC16k: ASoC: hw_params FE failed -110
[ 2880.906752] sof-audio-pci

wwittbrx, 2020-03-13:
I can't reproduce this issue on TGL RVP with slim driver tests.

Running on
https://github.com/mrajwa/sof/commits/tgl_dev_2
746cb8b965d362a99b10d140fa869da80dc9cefa

Are we running on the same FW? Also, could you provide us with the kernel source?

RDharageswari, 2020-03-13:
I am currently using the OTC repo with few changes from broonie tree:
31f37b4d5033 ASoC: SOF: Intel: hda-loader: clear the IPC ack bit after FW_PURGE done
2337f615bf53 ASoC: SOF: ipc: check ipc return value before data copy
907e9be8f1ac ASoC: SOF: pcm: skip DMA buffer pre-allocation
Firmware used was:
TGL FW memory layout changing
@Rajwa, Marcin have a simple test branch for the reference KWD HDA generic tplg in https://github.com/mrajwa/sof/tree/tgl_dev (fw and ldc shared in attachment).

RDharageswari, 2020-03-13:
@xiulipan Can you please help with the kernel?

mrajwa, 2020-03-15:
@RDharageswari, kernel has a problem with triggering stream. This is unknown kernel so I can't tell why is it failing, please provide link to the source.

keyonjie, 2020-03-16:
Hi @xiulipan can we try if we can reproduce this issue with Ubuntu on RVP?

xiulipan, 2020-03-16:
@keyonjie @RDharageswari @mrajwa confirmed with master(47420dc92c5e48fec50161e8b157a66b31cc9193) + https://github.com/thesofproject/sof/pull/2405
Kernel https://github.com/thesofproject/linux/commit/0f9070542fa2feb1e5486c6160c44f65b672aef8
kconfig https://github.com/thesofproject/kconfig/blob/master/kconfig-sof-default.sh
```
$ sudo sof-logger -l sof-tgl.ldc
         TIMESTAMP              DELTA  C# COMPONENT          LOCATION                       CONTENT
[2387521387.031250] (             nan) c0 MEM                         src/lib/alloc.c:280   ERROR error: 171 blocks needed for allocation but only 170 blocks are remaining
[2387523072.760417] (     1685.729126) c0 SA                          src/lib/agent.c:56    ERROR validate(), ll drift detected, delta = 51166
[2387538287.812500] (    15215.051758) c0 kpb          11.53          src/audio/kpb.c:391   ERROR kpb_params(): kpb has been already configured.
[2429327494.218750] ( 41789208.000000) c0 SA                          src/lib/agent.c:56    ERROR validate(), ll drift detected, delta = 59315
[2429327996.770833] (      502.552094) c0 kpb                         src/audio/kpb.c:443   ERROR kpb_prepare() error: wrong host params.
[2429328033.593750] (       36.822918) c0 PIPE                   src/audio/pipeline.c:527   ERROR pipeline_prepare() error: ret = -22, dev->comp.id = 52
[2429328074.114583] (       40.520832) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 11 comp 52 prepare failed -22
```
dmesg
```
[ 1214.673646] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[ 1214.681850] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[ 1214.681857] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[ 1214.681860] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[ 1214.681865]  DMIC16k: ASoC: hw_params FE failed -22
[ 1214.681872] sof-audio-pci 0000:00:1f.3: pcm: free stream 7 dir 1
[ 1214.682121] sof-audio-pci 0000:00:1f.3: pcm: close stream 7 dir 1
[ 1215.270366] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[ 1215.270408] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG

```
Full DMA trace log is attached.
[kwd.txt](https://github.com/thesofproject/sof/files/4337302/kwd.txt)

mrajwa, 2020-03-16:
@xiulipan, please show arecord command you start with.

xiulipan, 2020-03-16:
@mrajwa 
CMD is `arecord -Dhw:0,7 -M -N -c 2 -f S32_LE -r 16000 tmp.wav -vvv`

Test with some debug log code 

```diff
diff --git a/src/audio/kpb.c b/src/audio/kpb.c
index 94c2edb3d4a5..3f2c0bb24324 100644
--- a/src/audio/kpb.c
+++ b/src/audio/kpb.c
@@ -1487,14 +1487,18 @@ static inline bool validate_host_params(struct comp_dev *dev,
        size_t pipeline_period_size = (dev->pipeline->ipc_pipe.period / 1000)
                                        * bytes_per_ms;

+       comp_cl_err(&comp_kpb, "validate_host_params() period_size %lu buffer_size %lu min size %lu", host_period_size,host_buffer_size, HOST_BUFFER_MIN_SIZE(kpb->buffer_size));
+
        if (!host_period_size || !host_buffer_size) {
                /* Wrong host params */
+               comp_cl_err(&comp_kpb, "validate_host_params() error: No host params.");
                return false;
        } else if (HOST_BUFFER_MIN_SIZE(kpb->buffer_size) >
                   host_buffer_size) {
                /* Host buffer size is too small - history data
                 * may get overwritten.
                 */
+               comp_cl_err(&comp_kpb, "validate_host_params() error: Host buffer size is too small %lu.", HOST_BUFFER_MIN_SIZE(kpb->buffer_size));
                return false;
        } else if (kpb->sync_draining_mode) {
                /* Sync draining allowed. Check if we can perform draining
@@ -1507,8 +1511,10 @@ static inline bool validate_host_params(struct comp_dev *dev,
                 * of buffered data.
                 */
                if ((host_period_size / KPB_DRAIN_NUM_OF_PPL_PERIODS_AT_ONCE) <
-                   pipeline_period_size)
-                       return false;
+                   pipeline_period_size) {
+                                       comp_cl_err(&comp_kpb, "validate_host_params() error: KPB_DRAIN_NUM_OF_PPL_PERIODS_AT_ONCE.");
+                                       return false;
+                               }
        }

        return true;

```
Got new error logs

```
$ sudo sof-logger -l sof-tgl.ldc
         TIMESTAMP              DELTA  C# COMPONENT          LOCATION                       CONTENT
[4007141969.895833] (             nan) c0 kpb                         src/audio/kpb.c:1490  ERROR validate_host_params() period_size 15936 buffer_size 63936 min size 0
[4007153612.968750] (    11643.073242) c0 MEM                         src/lib/alloc.c:280   ERROR error: 171 blocks needed for allocation but only 170 blocks are remaining
[4007155308.802083] (     1695.833374) c0 SA                          src/lib/agent.c:56    ERROR validate(), ll drift detected, delta = 58017
[4007177248.020833] (    21939.218750) c0 kpb          11.53          src/audio/kpb.c:391   ERROR kpb_params(): kpb has been already configured.
[4026960852.239583] ( 19783604.000000) c0 kpb                         src/audio/kpb.c:1490  ERROR validate_host_params() period_size 15936 buffer_size 63936 min size 537600
[4026960896.458333] (       44.218750) c0 kpb                         src/audio/kpb.c:1501  ERROR validate_host_params() error: Host buffer size is too small 537600.
[4026960932.031250] (       35.572918) c0 kpb                         src/audio/kpb.c:443   ERROR kpb_prepare() error: wrong host params.
[4026960967.708333] (       35.677082) c0 PIPE                   src/audio/pipeline.c:527   ERROR pipeline_prepare() error: ret = -22, dev->comp.id = 52
[4026961007.760417] (       40.052082) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 11 comp 52 prepare failed -22

```
It seems the `kpb->buffer_size` is not init before parpare?

xiulipan, 2020-03-16:
@mrajwa @RDharageswari @keyonjie Also test with runtime-pm.
This issue only happen with no-pm-suspend/resume between two record.
@wwittbrx I think for slim driver test, every test is with fresh loaded FW?

wwittbrx, 2020-03-16:
@xiulipan We have more iteration options. There are normal iterations, which reload FW every time as you said, we also have topology iterations, which free all components and start again with topology building without reloading, and playback iterations, which only free the streams and restarts them. I tested this with all iteration kinds.

Please try this command with buffer size specified:
arecord -M -N -D hw:0,7 -f s24_le -c 2 -r 16000 --buffer-size=68000 record.wav -vvv -d 5
This should fix the wrong host params error.

mrajwa, 2020-03-16:
@xiulipan, @RDharageswari there is a bug on my testing branch which result in wrong validation of host params on the very first call only. But if this is the case it means you did not specified proper host buffer size. Please use the command @wwittbrx shown above.

PS: I have also fixed the missing check on FW side. The patch is already available on https://github.com/mrajwa/sof/tree/tgl_dev_2

RDharageswari, 2020-03-16:
arecord -M -N -D hw:0,7 -f s24_le -c 2 -r 16000 --buffer-size=68000 record.wav -vvv -d 5
Even with the command i got the error:

[  125.321329] sof-audio-pci 0000:00:1f.3: Turning i915 HDAC power 0
[  125.321345] sof-audio-pci 0000:00:1f.3: error: transition from 0 to 4 not allowed
[  125.321346] sof-audio-pci 0000:00:1f.3: error: failed to power down DSP during suspend -22
[  125.321354] sof-audio-pci 0000:00:1f.3: can't suspend (snd_sof_runtime_suspend [snd_sof] returned -22)
[  127.268554] sof-audio-pci 0000:00:1f.3: pcm: open stream 7 dir 1
[  127.268558] sof-audio-pci 0000:00:1f.3: period min 192 max 256000 bytes
[  127.268559] sof-audio-pci 0000:00:1f.3: period count 2 max 160
[  127.268560] sof-audio-pci 0000:00:1f.3: buffer max 1280000 bytes
[  127.268838] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 7 dir 1
[  127.268842] sof-audio-pci 0000:00:1f.3: generating page table for 00000000170a3b84 size 0x84cc0 pages 133
[  127.268864] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  127.270191] sof-audio-pci 0000:00:1f.3: error: stream reset failed
[  127.270193] sof-audio-pci 0000:00:1f.3: error: hdac prepare failed: ffffff92
[  127.270194] sof-audio-pci 0000:00:1f.3: error: platform hw params failed
[  127.270196] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
[  127.270199]  DMIC16k: ASoC: hw_params FE failed -110
[  127.270203] sof-audio-pci 0000:00:1f.3: pcm: free stream 7 dir 1
[  127.270421] sof-audio-pci 0000:00:1f.3: pcm: close stream 7 dir 1
localhost ~ #

xiulipan, 2020-03-17:
@wwittbrx @mrajwa @RDharageswari Test with tgl_dev_2 branch patches.
It works with `--buffer-size=68000`
@mrajwa Could you try to send a PR for your fixes on tgl_dev_2?

mrajwa, 2020-03-17:
@xiulipan, I think you meant "could you upstream your fix from tgl_dev_2" right?

mrajwa, 2020-03-17:
@RDharageswari, have you tested with latest https://github.com/mrajwa/sof/tree/tgl_dev_2?
I don't follow what FW and kernel do you use....

RDharageswari, 2020-03-18:
Hi @mrajwa 
I did check with the attached firmware on my kernel and still see issues.
First 2 /3 iterations go well and the moment i hit the failure..subsequent capture fails.
I am using the FW built out of https://github.com/mrajwa/sof/tree/tgl_dev_2
For the kernel i am trying to re-test with v5.4.Will update once i test.

xiulipan, 2020-03-18:
@mrajwa @RDharageswari I run more times on my DUT, did not see issue like your said that failed on 3/3.

Below is my 5/5 tests
```
$ sudo sof-logger -l sof-tgl.ldc
         TIMESTAMP              DELTA  C# COMPONENT          LOCATION                       CONTENT
[136439847616.093750] (             nan) c0 kpb                         src/audio/kpb.c:1492  ERROR validate_host_params() period_size 135936 buffer_size 543936 min size 0
[136439855744.635422] (     8128.541504) c0 MEM                         src/lib/alloc.c:280   ERROR error: 171 blocks needed for allocation but only 170 blocks are remaining
[136439855823.489594] (       78.854164) c0 SA                          src/lib/agent.c:56    ERROR validate(), ll drift detected, delta = 40395
[136439872406.250000] (    16582.759766) c0 kpb          11.53          src/audio/kpb.c:392   ERROR kpb_params(): kpb has been already configured.
[136444600150.781250] (  4727744.500000) c0 SA                          src/lib/agent.c:56    ERROR validate(), ll drift detected, delta = 46752
[136444600653.385422] (      502.604156) c0 kpb                         src/audio/kpb.c:1492  ERROR validate_host_params() period_size 135936 buffer_size 543936 min size 537600
[136444615473.333344] (    14819.948242) c0 kpb          11.53          src/audio/kpb.c:392   ERROR kpb_params(): kpb has been already configured.
[136444619830.208344] (     4356.875000) c0 kpb          11.53          src/audio/kpb.c:652   ERROR kpb_copy() error: nothing to copy sink->free 5120 source->avail 0
[136446709969.218750] (  2090139.000000) c0 kpb                         src/audio/kpb.c:1492  ERROR validate_host_params() period_size 135936 buffer_size 543936 min size 537600
[136446724901.979172] (    14932.760742) c0 kpb          11.53          src/audio/kpb.c:392   ERROR kpb_params(): kpb has been already configured.
[136450551972.708344] (  3827070.750000) c0 kpb                         src/audio/kpb.c:1492  ERROR validate_host_params() period_size 135936 buffer_size 543936 min size 537600
[136450566884.687500] (    14911.979492) c0 kpb          11.53          src/audio/kpb.c:392   ERROR kpb_params(): kpb has been already configured.
[136452514863.489594] (  1947978.750000) c0 SA                          src/lib/agent.c:56    ERROR validate(), ll drift detected, delta = 60441
[136452515366.770844] (      503.281250) c0 kpb                         src/audio/kpb.c:1492  ERROR validate_host_params() period_size 135936 buffer_size 543936 min size 537600
[136452530451.250000] (    15084.479492) c0 kpb          11.53          src/audio/kpb.c:392   ERROR kpb_params(): kpb has been already configured.
```

mengdonglin, 2020-03-18:
@mrajwa Can you please share a PR for master branch to support KWD on TGL?

mrajwa, 2020-03-18:
@mengdonglin, sure however we still don't know what is the reason of the issue @RDharageswari is describing.
@RDharageswari, on the beginning you said the second capture fails, now you are saying "First 2 /3 iterations go well..." so something changed recently?
The issue seems to be on kernel side not being able to change states. 
I am waiting for your tests results with different kernel.

ranj063, 2020-03-18:
@RDharageswari looking at the original logs: It looks like after you start the WoV stream, the DSP enters D0I3 but never exits back to D0I0 when you close the stream. And this is why you see that the transition from D0I3 -> D3 ( 0 to 4 in the logs) is not allowed.

**[  348.303273] sof-audio-pci 0000:00:1f.3: New DSP state 0 substate 1** --> DSP is in D0I3
[  365.964915] sof-audio-pci 0000:00:1f.3: ipc rx: 0x50050034
[  365.964918] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x50050034
[  373.523923] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 7 dir 1 cmd 0
[  373.523930] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000
[  373.524110] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000
[  373.524160] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x240000 successful
[  373.524167] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000
[  373.525255] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000
[  373.525336] sof-audio-pci 0000:00:1f.3: pcm: free stream 7 dir 1
[  373.525437] sof-audio-pci 0000:00:1f.3: pcm: close stream 7 dir 1
[  373.525577] sof-audio-pci 0000:00:1f.3: received event 4 for widget DETECT12.0
[  373.525581] sof-audio-pci 0000:00:1f.3: received event 8 for widget DETECT12.0
[  373.525587] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000
[  373.525727] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60050000
[  373.525736] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60030000
[  373.526774] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x60030000
[  376.026120] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x20140000 successful
[  376.026154] sof-audio-pci 0000:00:1f.3: ipc tx: 0x40010000
[  376.026337] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x40010000
[  376.026432] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x1010f0f successful
[  376.026437] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0xf0f successful
[  376.026440] sof-audio-pci 0000:00:1f.3: DSP core(s) enabled? 0 : core_mask 1
[  376.026500] sof-audio-pci 0000:00:1f.3: Debug PCIR: 00000010 at  00000044
[  376.026507] sof-audio-pci 0000:00:1f.3: Turning i915 HDAC power 0
**[  376.026531] sof-audio-pci 0000:00:1f.3: error: transition from 0 to 4 not allowed** --> DSP tries to enter D3 and fails
[  376.026532] sof-audio-pci 0000:00:1f.3: error: failed to power down DSP during suspend -22
[  376.026543] sof-audio-pci 0000:00:1f.3: can't suspend (snd_sof_runtime_suspend [snd_sof] returned -22)

ranj063, 2020-03-19:
problem root caused to a missing patch in the chrome backport:
"ASoC: SOF: Intel: CNL: add set_power_state() ops" https://github.com/thesofproject/linux/commit/73b51957122d

@sathya-nujella FYI

RDharageswari, 2020-03-19:
Since we work remotely i was just able to do
arecord
Ctrl+C
multiple times and verify. I couldn't still validate the end to end stack.
As of now using the board from India and will verify it tonight with actual functionality and S0ix then update

ranj063, 2020-03-20:
@RDharageswari can you please close this issue once you've verified with the added kernel patch?

RDharageswari, 2020-03-20:
@ranj063 , I couldnt verify the functionality lsat night due to limited board availability. Will get the board only on Monday.

xiulipan, 2020-03-23:
I have create sof-tgl-max98373-rt5682.tplg with KWD in https://github.com/thesofproject/sof/pull/2405

@RDharageswari @emilchudzik could you have a try on the Chrome DUT?
I am using the latest master FW 9da1fce2478df1f5b4993228410ecadae35c22bd
and sof-dev kernel 9b5b6c2a3aef3ed119f89cd5cdd0726153b71bd4

RDharageswari, 2020-03-24:
i verified the Wov functionality with the latest master FW 9da1fce
and the missed patch "ASoC: SOF: Intel: CNL: add set_power_state() ops" thesofproject/linux@73b5195
functionality works fine.

closed 2020-03-24 by tlauda