thesofproject/sof #2520: @paulstelian97 I think we should not hardcode parameters but try to read them from sai->params. Let me have a look and come up with a proposal. Hopefully today or tomorrow.

dbaluta, 2020-03-10:
@paulstelian97 I think we should not hardcode parameters but try to read them from sai->params. Let me have a look and come up with a proposal. Hopefully today or tomorrow.

_Originally posted by @dbaluta in https://github.com/thesofproject/sof/pull/2500_

dbaluta, 2020-03-10:
@IulianOlaru249 @constantinraducanu you can have a look at this if you have some time.

ymdatta, 2020-03-10:
@dbaluta: Yes, i will take a look at this. Can you assign this issue to me? Thanks.

costi2, 2020-03-10:
@dbaluta I will look on this too.

dbaluta, 2020-03-11:
@ymdatta @costi2 @IulianOlaru249 great o know about your interest. 

If you have time you can also try to see if you can tackle one of unassigned issues from here. Check with the reporter of the issue for more details.

https://github.com/thesofproject/sof/issues?q=is%3Aissue+is%3Aopen+FEATURE

closed 2025-07-05 by github-actions[bot]