thesofproject/sof #2529: [BUG] Distortion on the second pipeline connected to demux

bardliao, 2020-03-11:
**Describe the bug**
Distortion can be seen on the second amp output with https://github.com/thesofproject/sof/pull/2526 applied

**To Reproduce**
1. run aplay
2. listen to the output from both amps

**Reproduction Rate**
100%

**Expected behavior**
Clear output from both amps

**Impact**
annoyance

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: topic/sof-dev branch https://github.com/thesofproject/linux/commit/d05959d5021cefbbd841773ee25f6c7387e6bfd9 + https://github.com/thesofproject/linux/pull/1863
    * SOF: master branch https://github.com/thesofproject/linux/commit/d05959d5021cefbbd841773ee25f6c7387e6bfd9 + https://github.com/thesofproject/sof/pull/2526
2) Name of the topology file
    * Topology: sof-cml-rt711-rt1308-rt715.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: CML laptop with 2 rt1308s

**Screenshots or console output**
[sof-logger-distortion.txt](https://github.com/thesofproject/sof/files/4318348/sof-logger-distortion.txt)

bardliao, 2020-03-11:
@slawblauciak I know the sof-log log is not sufficient for the audio quality issue. But can you check if you can find anything wrong with the sof-log log? In the meantime, I will try to do loopback to capture the audio data for you.

bardliao, 2020-03-12:
@slawblauciak I did a loopback test today, and I can get the same distortion as what I hear.  An interesting thing is that the sound is very clear when I test it with a 1K tone.
Below are the recording file captured by loopback. The first picture is 1k tone and the second one is a voice says left channel and right channel repeatedly.
[image]

[image]

I also upload the captured wav here

[1k.zip](https://github.com/thesofproject/sof/files/4323031/1k.zip)
[lr.zip](https://github.com/thesofproject/sof/files/4323032/lr.zip)

Could you look into it?

slawblauciak, 2020-03-12:
I'm on it. Can you also reproduce the issue by generating some other tone? Like 1.5K? Or 2K?

bardliao, 2020-03-12:
@slawblauciak That is the result of testing with your testing file.
[image]
[sine.zip](https://github.com/thesofproject/sof/files/4323403/sine.zip)

bardliao, 2020-03-13:
@slawblauciak The distortion is gone if I revert https://github.com/thesofproject/sof/pull/2541. I also tried adding a PGA to the second DAI, and the distortion happens. So, what I observed is the distortion will happen if a PGA on the second pipeline. In other words, the second pipeline can only work with pass through pipeline.

bardliao, 2020-03-13:
I also don't observe distortion if I play a 32bits wav no matter if a PGA on the second pipeline or not. @RanderWang Do you have any comments?

slawblauciak, 2020-03-13:
@bardliao looking at the #2541: I don't really understand why it was like that to begin with. Why apply volume to only one of the demux outputs. It is an interesting finding tho, it's very strange.

bardliao, 2020-03-13:
> @bardliao looking at the #2541: I don't really understand why it was like that to begin with. Why apply volume to only one of the demux outputs. It is an interesting finding tho, it's very strange.

@slawblauciak The reason of #2541 is that I want to keep the same volume level for two pipelines. @RanderWang said he think the issue probably due to bad format conversion. ie, convert 16/24 bits to 32 bits  see https://github.com/thesofproject/sof/blob/master/src/audio/dai.c#L769.

slawblauciak, 2020-03-13:
I think there's a problem with stream param propagation, also why you had to do that hack with the 2 channels. So, I think it might all be related. I'm currently investigating that.

jajanusz, 2020-03-18:
@bardliao Can you share all the formats that fail?
Also please share the topology that you used for your 'loopback' case?

bardliao, 2020-03-18:
> @bardliao Can you share all the formats that fail?
> Also please share the topology that you used for your 'loopback' case.

@jajanusz I think the issue could be related to [dconfig->frame_fmt = SOF_IPC_FRAME_S32_LE;](https://github.com/thesofproject/sof/blob/master/src/audio/dai.c#L772). Could you check it first?
I only tested S16_LE and S32_LE for now. And S32_LE works (no noise). I can test more if you need it.
I don't keep my topology of the loopback test. I modified on both kernel topology for the loopback test. The idea is to connect 2nd pipeline to rt711 codec and trigger capture and playback at the same time. Then use the same sdw area for playback and capture. So we can capture the playback data.
BTW, please also see https://github.com/thesofproject/sof/issues/2529#issuecomment-598556621

jajanusz, 2020-03-18:
I've seen that comment, but what you want me to do with it? It looks like HW requirement so it has to stay 32bit, if it doesn't work for non-32bit formats then volume has bad conversion logic or topology doesn't make sense.

>@jajanusz I think the issue could be related to dconfig->frame_fmt = SOF_IPC_FRAME_S32_LE;. Could you check it first?

I'd like to but we don't have repro, so it's hard to  check. That's why I ask for all these things, so we can get repro on our side ;)

jajanusz, 2020-03-18:
@bardliao Please add logs that show which method conversion was chosen in dai_params, so we'll know if demux is propagating correct formats.
After this: https://github.com/thesofproject/sof/blob/b20191b362a6d14a44eead247ae677edd90ff6dd/src/audio/dai.c#L279

bardliao, 2020-03-18:
> I've seen that comment, but what you want me to do with it? It looks like HW requirement so it has to stay 32bit, if it doesn't work for non-32bit formats then ~volume has bad conversion logic~ (volume doesn't have conv, dai has) or topology doesn't make sense.
> 
> > @jajanusz I think the issue could be related to dconfig->frame_fmt = SOF_IPC_FRAME_S32_LE;. Could you check it first?
> 
> I'd like to but we don't have repro, so it's hard to check. That's why I ask for all these things, so we can get repro on our side ;)

@jajanusz Please below is the environment to repro it.
Platform:  Dell CML-H laptop
kernel: https://github.com/bardliao/linux/commits/synchronize-playback 
FW&Tplg: https://github.com/bardliao/sof/commits/demux-test

bardliao, 2020-03-18:
> @bardliao Please add logs that show which method conversion was chosen in dai_params, so we'll know if demux is propagating correct formats.
> After this:
> 
> https://github.com/thesofproject/sof/blob/b20191b362a6d14a44eead247ae677edd90ff6dd/src/audio/dai.c#L279

dd->frame_fmt is always 2, and local_fmt depends on the testing wav format. ie. It is 0 for S16_LE and 2 for S32_LE. But I don't know how to check if we use functions defined in pcm_converter_generic.c or pcm_converter_hifi3.c.

jajanusz, 2020-03-18:
Can you change 0x02 in demux blob (it takes format from there instead from pipeline)?
```
CONTROLBYTES_PRIV(DEMUX_priv,
`       bytes "0x53,0x4f,0x46,0x00,0x00,0x00,0x00,0x00,'
`       0x28,0x00,0x00,0x00,0x00,0x60,0x00,0x03,'
`       0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`       0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`       >>0x02<< CHANGE THIS,0x00,0x02,0x00,0x02,0x00,0x00,0x00,'
```
There should be 0x00 for 16 bits.
So it'll look more like:
```
CONTROLBYTES_PRIV(DEMUX_priv,
`       bytes "0x53,0x4f,0x46,0x00,0x00,0x00,0x00,0x00,'
`       0x28,0x00,0x00,0x00,0x00,0x60,0x00,0x03,'
`       0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`       0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,'
`       0x00,0x00,0x02,0x00,0x02,0x00,0x00,0x00,'
```
Let me know if that helps.

bardliao, 2020-03-19:
> Can you change 0x02 in demux blob (it takes format from there instead from pipeline)?
>
> ```
> CONTROLBYTES_PRIV(DEMUX_priv,
> `       bytes "0x53,0x4f,0x46,0x00,0x00,0x00,0x00,0x00,'
> ` [...]

@jajanusz With this change, S16_LE file can play but S32_LE is noisy.

jajanusz, 2020-03-19:
> @jajanusz With this change, S16_LE file can play but S32_LE is noisy.

That's expected behaviour, you have to match format there with your format.

emilchudzik, 2020-03-19:
@bardliao I have CML-H as you:
Platform: Dell CML-H laptop

I use your kernel and FW and topology.
kernel: https://github.com/bardliao/linux/commits/synchronize-playback
FW&Tplg: https://github.com/bardliao/sof/commits/demux-test

But I have no repro with it.

Then I try with your patch: https://github.com/thesofproject/sof/issues/2533 and still no repro.

I use alsamixer to volume up controls.
Then I use command to play:
sudo aplay -vv -Dhw:1,2 -f dat - c2 doremi-48k-s16le-stereo.wav
or
sudo aplay -vv -Dhw:1,2 doremi-48k-s16le-stereo.wav

I have always clear sound on both speakers, without noise.

Did you modify topology? I got this one: sof-cml-rt711-rt1308-rt715.tplg from your branch

emilchudzik, 2020-03-19:
Could you share how you made loopback? I see in comments that you record to file output, so you have to modify toplogy to made loopback, right?

bardliao, 2020-03-19:
> That's expected behaviour, you have to match format there with your format.

But, the problem is we don't know what format will user play. Besides, there is no issue on the 1st pipeline connected to demux. The issue happens only on the 2nd pipeline.

bardliao, 2020-03-19:
> @bardliao I have CML-H as you:
> Platform: Dell CML-H laptop
>
> I use your kernel and FW and topology.
> kernel: https://github.com/bardliao/linux/commits/synchronize-playback
> FW&Tplg: [...]

Sorry, it should be #2541. I applied it to my repo, too
https://github.com/bardliao/sof/commits/demux-test

> I use alsamixer to volume up controls.
> Then I use command to play:
> sudo aplay -vv -Dhw:1,2 -f dat - c2 doremi-48k-s16le-stereo.wav
> or
> sudo aplay -vv -Dhw:1,2 doremi-48k-s16le-stereo.wav
>
> I have [...]

No, I didn't modify topology manually and yes, sof-cml-rt711-rt1308-rt715.tplg is the right one.

bardliao, 2020-03-19:
> Could you share how you made loopback? I see in comments that you record to file output, so you have to modify toplogy to made loopback, right?

Ok, I will do it.

emilchudzik, 2020-03-19:
> Sorry, it should be #2541. I applied it to my repo, too
> https://github.com/bardliao/sof/commits/demux-test

My mistake, I used 2541, not 2533 (error i pasting here)

> No, I didn't modify topology manually and yes, sof-cml-rt711-rt1308-rt715.tplg is the right one.

bardliao, 2020-03-19:
@emilchudzik The loopback environment is:
kernel: https://github.com/bardliao/linux/commits/demux-loopback-test
sof: https://github.com/bardliao/sof/commits/demux-looback-test
Please run below script before testing.
[rt711_mixer.txt](https://github.com/thesofproject/sof/files/4354383/rt711_mixer.txt)
[rt1308_mixer.txt](https://github.com/thesofproject/sof/files/4354384/rt1308_mixer.txt)
Open two terminal. 
run arecord and aplay at the same time (start arecord before aplay)
arecord -Dhw:0,1 -fDAT loop.wav -vv
aplay -Dhw:0,2 test.wav -vv

Check loop.wav.
We do loopback from headset port. You can hear the noise from headset and only left speaker will output. And aplay -Dhw:0,0 will not work.

tlauda, 2020-03-20:
Fixed by #2589.

closed 2020-03-20 by tlauda