thesofproject/sof #2531: [TGL][BUG] "DMIC: ASoC: trigger FE cmd: 1 failed: -110" occured when do multi-pipeline stress test.

sinahuang, 2020-03-12:
**Describe the bug**
"DMIC: ASoC: trigger FE cmd: 1 failed: -110" occured when do multi-pipeline stress test.

**To Reproduce**
1.sudo reboot
2.Run "./multiplex_pipeline-test.sh pcm0p pcm2p pcm6p pcm1c pcm3c".

scripts:

[multiplex_pipeline-test.txt](https://github.com/thesofproject/linux/files/4318309/multiplex_pipeline-test.txt)

**Reproduce rate**
Test for 300 times, failed at 17th time.

**Expected result**
Multi-pipeline should work normally.

**Actual result**
"DMIC: ASoC: trigger FE cmd: 1 failed: -110" occured when do multi-pipeline stress test.

```
[  459.053474] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  459.053476] sof-audio-pci 0000:00:1f.3: error: waking up any trace sleepers
[  459.053480]  DMIC: ASoC: trigger FE cmd: 1 failed: -110
[  459.053486] sof-audio-pci 0000:00:1f.3: pcm: trigger stream 3 dir 1 cmd 0
[  459.053491] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60050000: GLB_STREAM_MSG: TRIG_STOP
[  459.557722] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[  459.557727] sof-audio-pci 0000:00:1f.3: info: preventing DSP entering D3 state to preserve context
[  459.557736] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  459.557848] sof-audio-pci 0000:00:1f.3: error: can't enter idle
[  459.557849] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[  459.557851] sof-audio-pci 0000:00:1f.3: error: panic at src/lib/agent.c:51
```
**Test recipe**

Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev commit:d05959
FW:  https://github.com/thesofproject/sof/commits/master commit:  8e5ab9c
Tplg: Same with FW branch, sof-tgl-rt711-i2s-rt1308.tplg 
Platform:TGL-U RVP with codec ALC711 in SDW mode and  RT1308 in I2S mode

[multipipeline_test_27_dmesg.log](https://github.com/thesofproject/linux/files/4318284/multipipeline_test_27_dmesg.log)

sinahuang, 2020-03-12:
This issue happened after blocked issue #2492 fixed with FW commit:https://github.com/thesofproject/sof/commit/30492d73f919bd925198dd234fb176f47900ee79, from my bi-sect result might be caused by https://github.com/thesofproject/sof/commit/30492d73f919bd925198dd234fb176f47900ee79 or https://github.com/thesofproject/sof/commit/c596217ca5c405c5cf65a3b91267123e59645e29
@tlauda Please help to check it.

sinahuang, 2020-03-16:
Tested with master(commit:8e5ab9c2)+sof-dev(commit: 0f90705), issue can be reproduced. 
Test for 300 times, failed at 37th time.
[multipipeline_test_37_logger_error.log](https://github.com/thesofproject/sof/files/4336247/multipipeline_test_37_logger_error.log)
[multipipeline_test_37_all_logger.log](https://github.com/thesofproject/sof/files/4336245/multipipeline_test_37_all_logger.log)
[multipipeline_test_37_dmesg.log](https://github.com/thesofproject/sof/files/4336246/multipipeline_test_37_dmesg.log)

tlauda, 2020-03-16:
@sinahuang It looks like DMIC clocking problem. Are you sure your HW is correctly configured/reworked? We don't see such behavior on our TGL RVPs.
@emilchudzik Could you help with our rework list and BIOS settings?
@singalsu Any ideas?

sinahuang, 2020-03-17:
My HW configure/reworks as below:
1. Reworked SDW on ALC711 and I2S on 1308 on TGL-RVP .

2. BIOS settings:
Set Audio Link Mode to Advanced Link Config
Enable: HDA link, DMIC #0, DMIC #1 , SSP #2, SDW #1 (ALC1308 is on SSP #2, ALC711 is on SDW #1）
Enter HD Audio DSP Features Configuration, Enable I2S and select I2S Codec: Realtek ALC1308

singalsu, 2020-03-17:
@tlauda Those RORs should not happen. Apparently some slowness with DMA capture apparently does it. The other source for ROR is wrong configuration of decimation but I doubt it as happened since it worked until 37th time. I wonder if in addition to clearing the status bit should the FIFO be emptied by setting FINIT for OUTCTLRx? That would flush all the samples in FIFO and create discontinuity into audio.

tlauda, 2020-03-17:
@singalsu I think it won't help in the long run. It looks like there is something fundamentally wrong with the DMIC clocking.
@sinahuang Do you have any other TGL RVP? Maybe there is something wrong with the HW on this one, since we don't observe such issue on our boards.

sinahuang, 2020-03-18:
Used the other TGL RVP with  codec ALC711 in SDW mode and RT1308 in I2S mode, issue still can be reproduced.

Test for 300 times, failed at 28th time.
[multipipeline_test_28_dmesg.log](https://github.com/thesofproject/sof/files/4347007/multipipeline_test_28_dmesg.log)

tlauda, 2020-03-18:
@sinahuang If it only reproduces with multiple pipelines, then please verify with higher clock. Please build FW with disabled low power clock: **Platform -> [ ] Enable low power ring oscillator as the default clock**.

sinahuang, 2020-03-18:
Checked on FW with disabled low power clock: Platform -> [ ] Enable low power ring oscillator as the default clock, Issue still can be reproduced.

tlauda, 2020-03-18:
@sinahuang Could you check again? I've rebuilt the master with this option disabled and the test has passed 300 times on your platform.

sinahuang, 2020-03-19:
Check it again with master (commit: f1fa3be9) +sof-dev(commit:45f6424) after disabled low power clock: Platform -> [ ] Enable low power ring oscillator as the default clock , issue still can be reproduced.

Test for 300 times, failed at 29th time.

[multipipeline_test_29_dmesg.log](https://github.com/thesofproject/sof/files/4352982/multipipeline_test_29_dmesg.log)

tlauda, 2020-03-19:
@sinahuang Run 300 iterations again and test passed with low power clock disabled. Can you make sure you are using the right firmware?

sinahuang, 2020-03-19:
Checked this issue on latest FW (commit:f1fa3be9) with disabled low power clock: Platform -> [ ] Enable low power ring oscillator as the default clock,test for 300 times, issue can't be reproduced.

tlauda, 2020-03-19:
@mengdonglin Issue caused by overload of DSP. Similar to #2386 and #2280.

keyonjie, 2020-03-24:
I hit similar issue during DSM playback and echo reference capture running simultaneously, I think we should disable LPRO by default until we can handle it dynamically, thoughts? @mmaka1 @tlauda

slawblauciak, 2020-04-07:
Hi @sinahuang, can you please retest it now?
The high performance clock is now default on TGL, as a result this issue should no longer occur.

sinahuang, 2020-04-07:
Issue can't be reproduced with sof-dev commit:052e45d and master commit: b4e8b1e on TGL-U RVP with codec ALC711 in SDW mode and  RT1308 in I2S mode. So closed it.

closed 2020-04-07 by sinahuang