thesofproject/sof #2532: [BUG][BSW]“DSP panic” when test "Check-playback: Stress - 100x"

Liviali155, 2020-03-12:
**Describe the bug**
“DSP panic” when test "Check-playback: Stress - 100x",error occured when aplay via (0,0),after error occured,all the pipeline can not work,reboot can recovery

```ubuntu@jf-bsw-cyn-max98090-4:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofchtmax98090 [sof-chtmax98090], device 0: Low Latency (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofchtmax98090 [sof-chtmax98090], device 1: Media Playback 1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

**To Reproduce**
1.sudo reboot
2.cd sof-test(as attachment)
3.cd test-case
4. export TPLG=/lib/firmware/intel/sof-tplg/sof-cht-max98090.tplg
5.check-playback.sh -d 1 -l 100 -r 1

**Reproduction Rate**
2/3
1.failed at 29th of 100
2.failed at 80th of 100

**Expected behavior**
There is no DSP panic  when aplay/arecord

**Actual result**
DSP panic,all the pipeline can not work

[image]
[image]

<details>
 <summary>dmesg_error.log</summary>

````
kernel: [ 1120.372246] sof-audio-acpi 808622A8:00: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
kernel: [ 1120.372394] sof-audio-acpi 808622A8:00: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
kernel: [ 1120.372857] sof-audio-acpi 808622A8:00: pcm: free stream 0 dir 0
kernel: [ 1120.373305] sof-audio-acpi 808622A8:00: pcm: close stream 0 dir 0
kernel: [ 1120.678215] sof-audio-acpi 808622A8:00: ipc rx: 0x90020000: GLB_TRACE_MSG
kernel: [ 1120.678243] sof-audio-acpi 808622A8:00: ipc rx done: 0x90020000: GLB_TRACE_MSG
kernel: [ 1121.897015] sof-audio-acpi 808622A8:00: error : DSP panic!
kernel: [ 1121.897048] sof-audio-acpi 808622A8:00: panic: dsp_oops_offset 1329152 offset 1329152
kernel: [ 1121.897144] sof-audio-acpi 808622A8:00: error: can't enter idle
kernel: [ 1121.897161] sof-audio-acpi 808622A8:00: error: trace point 00000000
kernel: [ 1121.897176] sof-audio-acpi 808622A8:00: error: panic at src/lib/agent.c:51
kernel: [ 1121.897192] sof-audio-acpi 808622A8:00: error: DSP Firmware Oops
kernel: [ 1121.897209] sof-audio-acpi 808622A8:00: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060125 SAR     0x00000000
kernel: [ 1121.897225] sof-audio-acpi 808622A8:00: EPC1     0x00000000 EPC2     0x00000000 EPC3     0x00000000 EPC4    0xff2cfd65
kernel: [ 1121.897241] sof-audio-acpi 808622A8:00: EPC5     0xff2cfd65 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
kernel: [ 1121.897257] sof-audio-acpi 808622A8:00: EPS2     0x00000000 EPS3     0x00000000 EPS4     0x00060520 EPS5    0x00060520
kernel: [ 1121.897272] sof-audio-acpi 808622A8:00: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x000000a2
kernel: [ 1121.897285] sof-audio-acpi 808622A8:00: stack dump from 0xff307120
kernel: [ 1121.897307] sof-audio-acpi 808622A8:00: 0xff307120: 000000f8 00000000 ff307168 00000000
kernel: [ 1121.897325] sof-audio-acpi 808622A8:00: 0xff307124: 00000000 00000000 00000033 fa000000
kernel: [ 1121.897344] sof-audio-acpi 808622A8:00: 0xff307128: c8ff7c00 fefeb02b f8407828 ffff8e13
kernel: [ 1121.897361] sof-audio-acpi 808622A8:00: 0xff30712c: f8407828 ffff8e13 00000000 00000000
kernel: [ 1121.897379] sof-audio-acpi 808622A8:00: 0xff307130: b7231c00 ffff8e13 a3cc0fb0 ffffffff
kernel: [ 1121.897397] sof-audio-acpi 808622A8:00: 0xff307134: 00000000 00000000 00000000 00000000
kernel: [ 1121.897415] sof-audio-acpi 808622A8:00: 0xff307138: b7231c00 ffff8e13 a3cc1140 ffffffff
kernel: [ 1121.897432] sof-audio-acpi 808622A8:00: 0xff30713c: 00000000 00000000 00000000 00000000
kernel: [ 1121.897452] sof-audio-acpi 808622A8:00: error: ipc host -> DSP: pending no complete no raw 0x0
kernel: [ 1121.897466] sof-audio-acpi 808622A8:00: error: mask host: pending yes complete no raw 0x2
kernel: [ 1121.897483] sof-audio-acpi 808622A8:00: error: ipc DSP -> host: pending no complete no raw 0x800008000dead009
kernel: [ 1121.897496] sof-audio-acpi 808622A8:00: error: mask DSP: pending no complete no raw 0x0
kernel: [ 1121.897512] sof-audio-acpi 808622A8:00: error: waking up any trace sleepers
kernel: [ 1126.440138] sof-audio-acpi 808622A8:00: pcm: open stream 0 dir 0
````
</details>

Platform: BSW with onboard codec MAX98090 in I2S mode
Firmware: 9ebb2881 https://github.com/thesofproject/sof/commits/master
Kernel: ac3abc56 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-cht-max98090.tplg same as firmware

[sof-test.zip](https://github.com/thesofproject/sof/files/4322567/sof-test.zip)

[dmesg.txt](https://github.com/thesofproject/sof/files/4322565/dmesg.txt)

Liviali155, 2020-03-12:
Error not seen on below paltforms:
GLK Chrome with onboard codec DA7219 in I2S mode
CML laptop with onboard codec ALC3204 in HDA mode

plbossart, 2020-03-12:
@Liviali155 if you see an error on Cherrytrail you should also see it on Baytrail/Minnowboard?

Liviali155, 2020-03-12:
> @Liviali155 if you see an error on Cherrytrail you should also see it on Baytrail/Minnowboard?

@dengyangchao please help to check on Baytrail/Minnowboard

Liviali155, 2020-03-12:
Bisect result:
Changed Firmware to 0214 (4d17d0de),error not seen
Changed Firmware to 0217 (b206b7a8I),issue can be reproduced

lgirdwood, 2020-03-12:
@Liviali155 can you bisect again, the commits above are unrelated to playback on CHT. One is for GDB and the other for ASRC. Are you sure  b206b7a is at fault ?

dengyangchao, 2020-03-13:
> > @Liviali155 if you see an error on Cherrytrail you should also see it on Baytrail/Minnowboard?
> 
> @dengyangchao please help to check on Baytrail/Minnowboard

@Liviali155 issue can't reproduce on BYT Minnowboard with codec RT5682 in I2S mode

Liviali155, 2020-03-13:
Regression caused by Firmware commit:https://github.com/thesofproject/sof/commit/99df3ae2bfd6ace400694c873060aca2707b74d0

Liviali155, 2020-03-13:
@tlauda  please help to check it

tlauda, 2020-03-13:
@lgirdwood Is there any difference in DRAM size between BYT and CHT? Only way that commit could cause this is trying to use memory, which isn't available.

tlauda, 2020-03-13:
@Liviali155 The FW binary you're using is built with GCC or XCC? I've built current master with GCC and test passed multiple times, but I haven't tried with XCC as I don't have config.

lgirdwood, 2020-03-13:
No, both the same memory.. We could revert stack size change for BYT/CHT given they have simpler features compared to APL etc.

Liviali155, 2020-03-16:
> @Liviali155 The FW binary you're using is built with GCC or XCC? I've built current master with GCC and test passed multiple times, but I haven't tried with XCC as I don't have config.

I used XCC

Liviali155, 2020-03-17:
on BDW with codec RT286 in WSB mode also has this issue when run "check-playback-100times"

[image]

<details>
 <summary>dmesg_error.log</summary>

````
kernel: [ 1090.537635] sof-audio-acpi INT3438:00: ipc rx: 0x600a0001: GLB_STREAM_MSG: POSITION
kernel: [ 1090.537655] sof-audio-acpi INT3438:00: posn : host 0x8070 dai 0xed60 wall 0x307e43d
kernel: [ 1090.537660] sof-audio-acpi INT3438:00: ipc rx done: 0x600a0001: GLB_STREAM_MSG: POSITION
kernel: [ 1090.537673] sof-audio-acpi INT3438:00: PCM: stream 0 dir 0 DMA position 8220 DAI position 15192
kernel: [ 1090.537708] sof-audio-acpi INT3438:00: PCM: stream 0 dir 0 DMA position 8220 DAI position 15192
kernel: [ 1090.570461] sof-audio-acpi INT3438:00: error : DSP panic!
kernel: [ 1090.570466] sof-audio-acpi INT3438:00: panic: dsp_oops_offset 649216 offset 649216
kernel: [ 1090.570531] sof-audio-acpi INT3438:00: error: can't enter idle
kernel: [ 1090.570533] sof-audio-acpi INT3438:00: error: trace point 00000800
kernel: [ 1090.570534] sof-audio-acpi INT3438:00: error: panic at src/lib/agent.c:51
kernel: [ 1090.570536] sof-audio-acpi INT3438:00: error: DSP Firmware Oops
kernel: [ 1090.570538] sof-audio-acpi INT3438:00: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060125 SAR     0x00000000
kernel: [ 1090.570540] sof-audio-acpi INT3438:00: EPC1     0x00000000 EPC2     0x00000000 EPC3     0x00017dbd EPC4    0x00000000
kernel: [ 1090.570541] sof-audio-acpi INT3438:00: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
kernel: [ 1090.570543] sof-audio-acpi INT3438:00: EPS2     0x00000000 EPS3     0x00060520 EPS4     0x00000000 EPS5    0x00000000
kernel: [ 1090.570545] sof-audio-acpi INT3438:00: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00002440
kernel: [ 1090.570546] sof-audio-acpi INT3438:00: stack dump from 0x0040a600
kernel: [ 1090.570549] sof-audio-acpi INT3438:00: 0x0040a600: 00409710 00000000 0040a620 00000028
kernel: [ 1090.570551] sof-audio-acpi INT3438:00: 0x0040a604: 00000000 00000000 00000033 00014b14
kernel: [ 1090.570553] sof-audio-acpi INT3438:00: 0x0040a608: 58c0b900 9ec64d5f 8192d028 ffff9fff
kernel: [ 1090.570555] sof-audio-acpi INT3438:00: 0x0040a60c: 8192d028 ffff9fff 00000000 00000000
kernel: [ 1090.570556] sof-audio-acpi INT3438:00: 0x0040a610: 81384c80 ffff9fff 9d6c0960 ffffffff
kernel: [ 1090.570558] sof-audio-acpi INT3438:00: 0x0040a614: 00000000 00000000 00000000 00000000 
````
</details>

[dmesg.txt](https://github.com/thesofproject/sof/files/4343278/dmesg.txt)

tlauda, 2020-03-17:
@Liviali155 Please verify with #2563.

closed 2020-03-18 by tlauda

Liviali155, 2020-03-18:
> @Liviali155 Please verify with #2563.

Tried with PR  #2563 on BSW,error not seen

PR #2563 (2020-03-18): task: reduce default stack size for some platforms
Reduces default EDF task's stack size for some platforms,
which have less memory and aren't used for more serious
processing.

Fixes #2532.