thesofproject/sof #2533: [BUG] DEMUX comp not work normally

xiulipan, 2020-03-12:
**Describe the bug**
The DEMUX should copy the data from source to sink without any error, but now with DEMXU comp, the copy out data is broken.

**To Reproduce**
Use demux tplg sof-apl-demux-pcm512x.tplg (need #2524 )  to replace the sof-apl-pcm512x.tplg on UP2 PCM512X

```
aplay -Dhw:0,0 -c 2 -f s32_le -r 48000 32le-2ch-sin-100-200hz.wav -vvv
arecord -Dhw:0,4 -f dat 2ch_0312_04.wav -vvv
```

**Reproduction Rate**
100%

**Expected behavior**
The record wav should be same as input wav

**Impact**
This will be a stopper for echo reference and smart amplifier 

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev d05959d5
    * SOF: master 8e5ab9c2ea (need #2524 )  
2) Name of the topology file
    * Topology: sof-apl-demux-pcm512x.tplg rename to sof-apl-pcm512x.tplg 
3) Name of the platform(s) on which the bug is observed.
    * Platform: APL

**Screenshots or console output**
[image]

xiulipan, 2020-03-12:
Also test with https://github.com/thesofproject/sof/pull/2525 on TGL
wav files attached
2ch_0312_04.wav test on APL with DEMUX and 32le-2ch-sin-100-200hz.wav
4ch_0312_01.wav test on TGL with DEMUX and 32le-8ch-sin-100-800hz.wav
8ch_0312_02.wav test on TGL with DEMUX and 32le-8ch-sin-100-800hz.wav
8ch_0312_03.wav test on TGL with nocodec loopback and 32le-8ch-sin-100-800hz.wav
[demux_test.zip](https://github.com/thesofproject/sof/files/4322887/demux_test.zip)

@ClarexZhou Could you try to use the sof-apl-demux-pcm512x.tplg on an UP2 to check if the output from codec also have noise?

dengyangchao, 2020-03-12:
@xiulipan  Applied #2524 and rename sof-apl-demux-pcm512x.tplg  to  sof-apl-pcm512x.tplg   on platform APL-UP2-PCM512X. Then tried below scenarios:
1. Aplay via HW:0,0(headset) is no output
2. Aplay via HW:0,1(HDMI) can heard the output and no noise.  
Did i do something wrong here?
[amixer.txt](https://github.com/thesofproject/sof/files/4323211/amixer.txt)

xiulipan, 2020-03-12:
@dengyangchao Interesting. I only care about the 0,0 headset codec output. Can you check amixer setting with normal UP2 PCM512X.

lgirdwood, 2020-03-12:
I would like to see demux/mux fixed to remove the binary blobs. this is related to poor understaning and usage complexity. @juimonen fyi.

juimonen, 2020-03-12:
@lgirdwood ok, it is now very clear what you want, thanks. I've been little bit hesitant to work on the demux on fw side (as it is not done by me), but I'll see what I can do. I'll make some RFC PR suggestions about the structs and topology.

lgirdwood, 2020-03-12:
@juimonen thanks - if we look at the config structures used by the binary kcontrol (Ive added comments inline)
```c
struct mux_stream_data {
	uint32_t pipeline_id;
	uint8_t num_channels;
	uint8_t mask[PLATFORM_MAX_CHANNELS];

	uint8_t reserved[(20 - PLATFORM_MAX_CHANNELS - 1) % 4]; // padding to ensure proper alignment of following instances
};

struct sof_mux_config {
	uint16_t frame_format;   /* not sure why needed - come from hw_params */
	uint16_t num_channels; /* diito, from hw_params */
	uint16_t num_streams; /* unclear if this is source or sink or total */

	uint16_t reserved; // padding to ensure proper alignment

	struct mux_stream_data streams[];
};
```
They need documented for sure alongside the mux code that sets the data.

We either need to represent a mux (N streams -> 1 stream) or a demux (1 stream -> N streams).

1) For every MUX and DEMUX output channel we create an enum kcontrol that contains all possible input stream channel choices. i.e. pipeline1 left, pipeline 1 right, pipeline 2 left, pipeline 2 right and so on
2) The enum will only permit 1:1 mapping between. i.e. we cannot mix channels
3) Channel duplication is allowed i.e. we can route input pipeline1 left to more than 1 output channel.
4) topology should only create the minimum enums required based on actual channels/streams and not pre-defined max channels and streams.
5) MUX will create a small number of large enums whilst DEMUX will create a larger number of small enums.

xiulipan, 2020-03-13:
@lgirdwood Do you mean that we need a refine for MUX/DEMUX? I agree that the binary blob is very hard to understand and config. But how long would this take?

@juimonen Do you have any idea why we got this issue with following binary blob?
https://github.com/thesofproject/sof/blob/366aa67822a578e134765cf59def4e429d4e08c1/tools/topology/m4/demux_coef_default.m4#L1-L11
It should work with the sof-apl-demux-pcm512x.tplg but I think we may got some issue with the total copy function.

juimonen, 2020-03-13:
@xiulipan demux has never worked perfectly. The first version we did was only working with branched demux pipe directly connected to scheduler (pcm) i.e. you could not add any components to branched pipeline. This was because our fw scheduling didn't support this kind of pipe.

So I only recently tried to change the interface to use tokens and faced the same situation as you, so idisp id's we're wrong and the whole topology didn't load. I did not test it further, I just wanted to get to tokens (so loading). if you want to play with the blob, check https://github.com/thesofproject/sof/pull/2536, there are examples for demux with -h option.

I'll try the demux myself, but I'm little bit in "slow mode" now as I'm working from home.

lgirdwood, 2020-03-13:
> @lgirdwood Do you mean that we need a refine for MUX/DEMUX? I agree that the binary blob is very hard to understand and config. But how long would this take?

Yes, we only refine the topology so that we create enum kcontrols instead of this undecipherable blob. The ABI between FW and kernel remains the same.The FW will still support the blob method, but should also be able to use the enums. 
1) Kernel needs some helper functions to store a local blob for each mux/demux component in topology. It then allows the enum kcontrol to change the internal blobs before updating FW.
2) FW - may not need any update
3) Topology needs new macros for constructing enum text arrays and attaching to demux/mux components.

tlauda, 2020-03-16:
@xiulipan What is the final statement? I'm not sure whether it's worth looking into or maybe you're configuration is wrong?

xiulipan, 2020-03-17:
@tlauda Still testing, not sure if it is some configure issue or comp issue.

akloniex, 2020-03-19:
@xiulipan any status update? do you need some help with demux configuration?

xiulipan, 2020-03-19:
@akloniex I am not sure if my configure is wrong, but I can make it work with 2-ch. But it seems not work well with 4-ch. Can you give some advice?

tlauda, 2020-03-19:
@xiulipan @akloniex It's probably because mux/demux for some reason uses hardcoded format from blob instead from params. It's the same issue as with bit depth in #2529.

akloniex, 2020-03-19:
@tlauda Mux/Demux uses params that were passed to it during params propagation, and overwrites what was in the blob in it's configuration, inside mux_params() function call.
The problem might occur if demux has 2 downstream pipelines with different number of channels configured, which was not doable in the past, when components held params, but currently, after moving params to buffers, such configuration might be valid. Currently params from 1st pipeline will be used for processing both.

tlauda, 2020-03-19:
@akloniex Actually you're right. Params are updated, but processing function is not. You've just rootcaused the issue.

akloniex, 2020-03-19:
@tlauda You're right. In the past that was done in (de)mux_prepare function, but must've been lost to history.
I'm currently doing some minor cleanup around mux, so I might tackle this issue in the meantime.

tlauda, 2020-03-20:
@xiulipan Let us know whether #2589 fixes the issue.

xiulipan, 2020-03-20:
@tlauda @akloniex It seem the 4ch => 2ch or 8ch => 4ch DEMUX still not working well
@emilchudzik could you confirm do you check DEMUX like this?

I am using https://github.com/thesofproject/sof/pull/2525
Please also help to check if my control bytes have any issue in #2525 
4ch => 2ch sof-tgl-nocodec-mux-4ch.tplg
[image]

8ch => 2ch sof-tgl-nocodec-mux-8ch.tplg
[image]

And another issue is when I try to start the PCM1C when PCM0C is enabled, I will got following error

```
$ sudo ./sof-logger -l sof-tgl.ldc
         TIMESTAMP              DELTA  C# COMPONENT          LOCATION                       CONTENT
[5961957032.708334] (             nan) c0 mux          2.8      src/audio/component.c:186   ERROR comp_set_state() error: wrong state = 5, COMP_TRIGGER_PREPARE
[5961957096.197917] (       63.489582) c0 PIPE                   src/audio/pipeline.c:551   ERROR pipeline_prepare() error: ret = -22, dev->comp.id = 14
[5961957135.833334] (       39.635418) c0 IPC                       src/ipc/handler.c:311   ERROR ipc: pipe 3 comp 14 prepare failed -22
[5961957359.635417] (      223.802078) c0 mux          2.8      src/audio/component.c:176   ERROR comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
[5961957485.572917] (      125.937500) c0 dai          2.12     src/audio/component.c:176   ERROR comp_set_state() error: wrong state = 5, COMP_TRIGGER_RESET
[5965396270.572917] (  3438785.000000) c0 mux          2.8      src/audio/component.c:153   ERROR comp_set_state() error: wrong state = 1, COMP_TRIGGER_STOP
[5965396312.343750] (       41.770832) c0 PIPE                   src/audio/pipeline.c:691   ERROR pipeline_trigger() error: ret = -22, host->comp.id = 6, cmd = 0
[5965396354.166667] (       41.822918) c0 IPC                       src/ipc/handler.c:468   ERROR ipc: comp 6 trigger 0x50000 failed -22
```

juimonen, 2020-03-20:
@xiulipan check this out: https://github.com/thesofproject/sof/pull/2594
would probably help you a lot in configuration...
also this might help:
https://github.com/thesofproject/sof-docs/pull/203

xiulipan, 2020-03-20:
@juimonen I checked my byte configure, and I think I have the right configuration. The result looks wired, so I need to double confirm about the configuration so that we can start to reproduce the issue.

tlauda, 2020-03-20:
@xiulipan Your configuration looks OK. Could you make sure that your SSP is configured to 4/8 channels? And also please try with FW built with disabled low power clock. I've noticed that DEMUX is very load heavy. Platform -> [ ] Enable low power ring oscillator as the default clock.

keyonjie, 2020-03-25:
I spent some time to understand how mux/demux and its configuration blob work, I think basically they can work, though the channel mapping/mask is a little bit convoluted.

What I concern most is that we have designed a too *powerful* component -- which can not only support channel multiplexer, but also channel mixer features. The good point is that it is very flexible for use, but the bad point is that checking all 8 channel * 8 mask bit and data accumulation for each sample copying is actually time consuming and with low efficiency, I believe that's why we are hitting DSP overload issues when using mux/demux with LPRO enabled (especially when we are supporting 4 or 8 channels).

So I would vote the point @lgirdwood mentioned above that "The enum will only permit 1:1 mapping between. i.e. we cannot mix channels".

tlauda, 2020-03-25:
@keyonjie Good point. I have minor patch in the backlog, which instead of iterating over all source channels, it only scans channels mask. From what I've seen it cuts down MCPS by 30% for 8 channels, but still we are talking about ~70 MCPS for two sink buffers.

@singalsu Do you know if there is a smart way for doing this using SIMD instructions? We basically  just need to take mask of channels from the input stream and put into output stream. If we decide to keep mixing, then also saturation is needed.

singalsu, 2020-03-25:
The SIMD multiply or multiply-add instruction can be set to work with slots of data with multiplier weights set to fractional 0 or 1.0 values to let the slots pass. The control decoding should set up this kind of weights vector in advance for copy() operation to consume along data. Such way the demux processing should be fast. The weight can be 16 bit so then the multiply should be able to handle the 16/32 input data with max parallellism.

juimonen, 2020-03-25:
Generally I would not vote for degrading the functionality. If we are mix/muxing 8 channels, it _will_ take mips. If we are taking mips, but not doing anything useful, it should be fixed.

lgirdwood, 2020-03-25:
@keyonjie we should NOT be mixing any channels here. We should only be muxing or demuxing them. The enum kcontrols will fully support any mapping in both modes.

@juimonen @singalsu @tlauda The mapping functionality means we need to do expensive operations during init only and de-compose the copy() to a more complex memcpy()

MUX: we have multiple sources and one sink. e.g.
```c

/* we create an array of these for each source to sink channel mapping during init */
struct copy_data_mappping {
    void *sink_ptr; /* ptr in sink buffer for this mapping */
    void *source_ptr; /* pointer in source data buffer for this mapping*/
    int sink_step;  /* increment added to sink pointer for next dest ptr */
    int source_step /* increment added to source pointer for next source sample */
}; 

/* the mappings are then all updated with current base pointers at the start of copy() */

/* remainder of copy then becomes  */
for (i = 0; i < num_mappings; i++) {
        map[i]->sink_ptr = map[i]->source_ptr;
        map[i]->sink_ptr  += map[i]->sink_step;
        map[i]->source_ptr  += map[i]->source_step;
}
```
and the same for DEMUX. IIRC, HiFi may even have SIMD to help with the scatter / gather loop above.

keyonjie, 2020-03-25:
> Generally I would not vote for degrading the functionality. If we are mix/muxing 8 channels, it _will_ take mips. If we are taking mips, but not doing anything useful, it should be fixed.

I think we have no real usage of the channel mixing function from mux/demux component until now, so it is not too late to degrade it if it really help on performance.

keyonjie, 2020-03-25:
> @keyonjie we should NOT be mixing any channels here. We should only be muxing or demuxing them. The enum kcontrols will fully support any mapping in both modes.
> 
> @juimonen @singalsu @tlauda The mapping functionality means we need to do expensive operations during init only and de-compose the copy() to a more complex memcpy() only
> 
> e.g. MUX: we have multiple sources and one sink.
> 
> ```c
> /* we create an array of these for each source to sink channel mapping during init */
> struct copy_data_mappping {
>     void *sink_ptr; /* ptr in sink buffer for this mapping */
>     void *source_ptr; /* pointer in source data buffer for this mapping*/
>     int sink_step;  /* increment added to sink pointer for next dest ptr */
>     int source_step /* increment added to source pointer for next source sample */
> }; 
> 
> /* the mappings are then all updated with current base pointers at the start of copy() */
> 
> /* remainder of copy then becomes  */
> for (i = 0; i < num_mappings; i++) {
>         map[i]->sink_ptr = map[i]->source_ptr;
>         map[i]->sink_ptr  += map[i]->sink_step;
>         map[i]->source_ptr  += map[i]->source_step;
> }
> ```
> 
> and the same for DEMUX. IIRC, HiFi may even have SIMD to help with the scatter / gather loop above.

This is good example, ack.

keyonjie, 2020-03-27:
@xiulipan @bkokoszx @akloniex I have root caused the noise issue here, will submit a PR tonight.

lgirdwood, 2020-03-27:
@keyonjie Will your PR include the performance fix ?

keyonjie, 2020-03-28:
@lgirdwood not yet, the performance fix is a relative big change(corresponding tplg/blob and ABI changes needed) so better to do it after the recent releases.

Here comes the PR: https://github.com/thesofproject/sof/pull/2653, please help review, it is verified  in DSM+Echo Ref scenario.

juimonen, 2020-04-01:
Hi, I tried the sof-apl-demux-pcm512 topology and with some older firmware it worked somehow (distorted sound), now when I updated to latest firmware, recording is totally silent... @kv2019i suspects we don't even have playback (with audible sound). Little bit difficult to remotely debug this...
@tlauda @akloniex can you try this in up2?

juimonen, 2020-04-01:
and yes sorry I had my up2 dusting for a while, so no idea what he old firmware was. I was updating everything to latest and greatest.

kv2019i, 2020-04-01:
@juimonen Ack, demux topology doesn't work with today's FW master. I tried going back to 1.4.2 and there it still works as expected. So something has broke.

lgirdwood, 2020-04-01:
See #2687 lets simplify.

keyonjie, 2020-04-03:
> @juimonen Ack, demux topology doesn't work with today's FW master. I tried going back to 1.4.2 and there it still works as expected. So something has broke.

@juimonen @kv2019i can you try the latest master as @akloniex 's #2600 was merged yesterday?

YvonneYang2, 2020-04-03:
Tested it with latest sof-dev (commit:fce7f93) + master (commit: d7ad0c13), issue can still be reproduced on APL-UP2-PCM512X.
tplg: same as fw branch: rename sof-apl-demux-pcm512x.tplg to sof-apl-pcm512x.tplg

Reproduce steps:
```
aplay -Dhw:0,0 -c 2 -f s32_le -r 48000 32le-2ch-sin-100-200hz.wav -vvv
arecord -Dhw:0,4 -f dat 2ch_0312_04.wav -vvv
```
Screenshot:
[image]

[wav_files_and_demsg_sof_trace.zip](https://github.com/thesofproject/sof/files/4426450/wav_files_and_demsg_sof_trace.zip)

slawblauciak, 2020-04-28:
@akloniex mind taking a look at it?

juimonen, 2020-04-29:
latest firmware af38b20bc84f79c8 and latest topology sof-apl-demux-pcm512x.tplg I can get proper sound from echoref pipe!

terminal 1:
aplay test_play.wav

Terminal 2: 
 arecord -Dhw:0,4 -f S16_LE -c 2 -r 48000 -d 10 test.wav

So maybe someone else can also verify?

juimonen, 2020-04-30:
@akloniex any idea why we didn't use the "caps" in ipc_buffer for underrun/overrun flags and defined new flags?

struct sof_ipc_buffer {
        struct sof_ipc_comp comp;
        uint32_t size;          /**< buffer size in bytes */
        uint32_t caps;          /**< SOF_MEM_CAPS_ */
        uint32_t flags;         /**< SOF_BUF_ flags defined above */

I think it is also 32bit bit value which is using only couple of bits for memory related flags...

xiulipan, 2020-05-20:
Move to new #2959 with new test and new issue.

closed 2020-05-20 by xiulipan