thesofproject/sof #2534: [BUG] Volume tokens in topologies may conflict with other pipelines

singalsu, 2020-03-12:
**Describe the bug**
We have a common mistake in SOF topology files with volume tokens. Different pipelines share the same identifier names for tokens "containers": capture_pga_tokens, capture_pga_conf, playback_pga_tokens, playback_pga_conf. Therefore editing volume ramp length (SOF_TKN_VOLUME_RAMP_STEP_MS) in one pipeline macro can have no impact or side effect to other volume pipelines. We might want to have different volume ramp length especially in capture for headset codec and DMIC.

There's need to clean up every topology by appending e.g. pipeline ID into these token container names in the pipeline macros.

**To Reproduce**
Steps to reproduce the behavior: (e.g. list commands or actions used to reproduce the bug)

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

lgirdwood, 2020-03-12:
these components should be per component instance and not global for all of that same type.

fredoh9, 2020-04-21:
After #2660, mow Volume tokens have  PIPELIN_ID.
i.e pga_tokens_1,  pga_conf_1

closed 2020-04-21 by fredoh9