thesofproject/sof #2538: RT5682 recording is sometime noisy

sathyap-chrome, 2020-03-13:
On 2 of CML customer designs, sometimes the recording is having noise.

Steps to reproduce:
arecord -f cd -d 10 (Fail Rate: 3/10 times)

Use Chrome OS image latest : it is using SOF v0.8
Attached is 1 of failed recording.

NOTE: this issue is not triaged - just logging many are not able to access the customer report.
[Fail_3.zip](https://github.com/thesofproject/linux/files/3936167/Fail_3.zip)

aiChaoSONG, 2020-03-13:
@plbossart
a) No need to pause/release, just run arecord several times, noise will appear
c) I just found change bclk from 2.4MHz to 4.8 MHz, this issue disappeared. So we suspect that this issue should be related to SSP clock. 4.8MHz is not the final solution, let's leave this to FW team to see if there is any SSP register not properly configurated.

sathyap-chrome, 2020-03-13:
@plbossart The sof-rt5682.m4 is what is used for all CML using RT5682 (the rt1011 variant reuses this for RT5682 config. only speaker config is additional)

So i see only difference, GLK with MCLK 19.2 and CML with 24.
considering the same, i want to give a try if we can set MCLK to 19.2 on CML ( for test purpose using M/N support) and see if issue is still seen.

I've reached out to vendor too on codec behavior on the experiment from @aiChaoSONG

sathyap-chrome, 2020-03-13:
@aiChaoSONG can you try if below change works on CML ?

I'm trying to see if we can use MCLK as 19.2 ( hope M/N can take care of this ? )

-       SSP_CONFIG(I2S, SSP_CLOCK(mclk, SSP_MCLK_RATE, codec_mclk_in),
+       SSP_CONFIG(I2S, SSP_CLOCK(mclk, 19200000, codec_mclk_in),

aiChaoSONG, 2020-03-13:
@sathyap-chrome Changing mclk to 19200000 doesn't help with this issue, I got `tplg component load failed` and `failed to instantiate card`

sathyap-chrome, 2020-03-13:
Thankd for checking @aiChaoSONG

plbossart, 2020-03-13:
> @plbossart
> a) No need to pause/release, just run arecord several times, noise will appear
> c) I just found change bclk from 2.4MHz to 4.8 MHz, this issue disappeared. So we suspect that this issue [...]

the only difference is a divider value.
we'd need to verify if indeed the MCLK is taken as the source clock for the codec PLL, if the behavior changes depending on the value of the BCLK, it means that either the BCLK is taken as a reference, or that the ratio between MCLK/BCLK should not be larger than a specific factor.

sathyap-chrome, 2020-03-17:
Update from codec vendor : based on our experiments, codec vendor make certain modifiation in driver to support 24M CLK and not use internal PLL and sees positive results. working on a clean patch for the same. will update here in couple of days

sathyap-chrome, 2020-03-17:
https://patchwork.kernel.org/patch/11441955/ is the fix from codec vendor for the issue. 
Thanks to all for help in debug.
 Closing the issue now.

closed 2020-03-17 by tlauda