thesofproject/sof #2543: [BUG] unsupported widget type asrc when run build-tools.sh

aiChaoSONG, 2020-03-13:
The SND_SOC_TPLG_DAPM_ASRC is delete from upstream alsa-lib(`src/topology/damp.c: line24~line48`), so topologies with ASRC component will be failed to build. we got :`widget 'ASRC1.0': unsupported widget type asrc` error when run build-tools.sh.

aiChaoSONG, 2020-03-13:
@lgirdwood @plbossart What are we going to do with this?

lgirdwood, 2020-03-13:
@aiChaoSONG can you bisect to see when this was removed and who did it ?

tlauda, 2020-03-13:
@lgirdwood It's already in the bug: 
> alsa-lib commit which makes the change: alsa-project/alsa-lib@aa1bac2

plbossart, 2020-03-13:
@aiChaoSONG Sorry I don't get what you meant by 'delete from upstream'. It was never added?

@singalsu you've added those topologies with ASRC, did you face any issues with alsa-lib?

singalsu, 2020-03-13:
I haven't got this issue. I've run tools build with various recent Ubuntu Linuxes. There has no more been need to use alsa tools git version to create SOF topologies.

Why is ASRC being removed from ALSA?

tlauda, 2020-03-13:
> I haven't got this issue. I've run tools build with various recent Ubuntu Linuxes. There has no more been need to use alsa tools git version to create SOF topologies.
> 
> Why is ASRC being removed from ALSA?

It looks like some kind of accidental removal?
Maybe @perexg could elaborate.

perexg, 2020-03-13:
Yep, it's accidental removal. Fixed in alsa-lib's master. Thanks for the note.

closed 2020-03-13 by lgirdwood

marc-hb, 2020-03-25:
FWIW reproduced with out-of-the-box alsa-lib 1.2.2-1.fc31

marc-hb, 2020-05-13:
While no release include the v1.2.2-6-gcaf77a93cef5 fix yet, it has been cherry-picked in the `alsa-git.patch` of alsa-utils version 1.2.2 **-2** .fc31 https://koji.fedoraproject.org/koji/rpminfo?rpmID=21111550

On Ubuntu 20.04, alsa-utils 1.2.2-1ubuntu1 does not have this fix.

marc-hb, 2020-06-22:
Here's a quick, dirty and lazy workaround for this situation: 

```shell
# Remove asrc from ALG_MODE_TESTS list
sed -i -e '/^ALG_MODE_TESTS=.*asrc/ s/asrc//' tools/test/topology/tplg-build.sh 

# Comment out all lines with 'asrc'
sed -i -e '/asrc/ s/.*/# &/' tools/topology/CMakeLists.txt

git commit -m -n 'HACK: disable ASRC1.0' tools/test/topology/tplg-build.sh  tools/topology/CMakeLists.txt
```
See PR `scripts: build-tools.sh: Make tools individually buildable` #3064 for more background

marc-hb, 2020-10-01:
Fix still not cherry-picked by Ubuntu 20.04 in `libatopology2 1.2.2-2.1ubuntu2`