thesofproject/sof #2546: [BUG][APL][WHL] SOF_IPC_PANIC_ASSERT in KD with SSP

jdrzaszcz, 2020-03-13:
**Describe the bug**
Running keyword detection feature with full topology,  using SSP loopback to get injection signal.
When reloading the topology without reloading FW in 17 Iterations is an exception Timeout error in the playback stream and than Panic.

*.*Topology**
```
                pipe_plb
               +------------------+
        +----+ | +---+  +-------+ |
        |Host|-->|Buf|->|SSP Dai|----------------------------------+
        +----+ | +---+  +-------+ |                                |
               +----------------- +                                |
                                                                   |
                   +-------------------------+                     |
        +--------+ | +---+   +-----+   +---+ |                     |
        |KD Dummy|<->|Buf|<->|ChSel|<->|Buf|<->+                   |
        +--------+ | +---+   +-----+   +---+ | |  pipe_cap         |
                   +-------------------------+ |                   |
                                               |                   |
                             +-----------------+                   |
        +--------------------|-----------------------------------+ |
        |+----+  +------+  +---+  +---+  +---+  +---+  +-------+ | |
        ||Host|<-|Buffer|<-|Kpb|<-|Buf|<-|Vol|<-|Buf|<-|SSP Dai|---+
        |+----+  +------+  +---+  +---+  +---+  +---+  +-------+ |
        +--------------------------------------------------------+
```
**To Reproduce**
Run test_02_kd_with_ssp.py 
with --topology_iterations=100
Repo:
https://gitlab.devtools.intel.com/cavs_scripts-py/cavs_scripts-py

**Reproduction Rate**
100 %

**Expected behavior**
Run all topology iterations without timeout and panic.

**Environment**
SOF master 15736c57667e541e619ec82618c322b7977f5fbd

**Logs**
[2020_03_13_09_54_43.zip](https://github.com/thesofproject/sof/files/4328965/2020_03_13_09_54_43.zip)
[2020_02_28_10_11_17.zip](https://github.com/thesofproject/sof/files/4328966/2020_02_28_10_11_17.zip)
[2020_03_13_10_35_59.zip](https://github.com/thesofproject/sof/files/4328967/2020_03_13_10_35_59.zip)

tlauda, 2020-03-16:
@jdrzaszcz #2557 should fix the issue. Another problem is the fact, that test framework increases pipeline id every topology iteration. Probably the same one should be reused.

jdrzaszcz, 2020-03-17:
@tlauda yes pr 2557 fix this issue. Btw, thanks for the hint.

PR #2557 (2020-03-18): pipeline: implement proper position offset retrieval
Implements proper position offset retrieval in order to
update current stream position in mailbox. Previous one
was using pipeline_id in calculations, which has been
failing for bigger values. Lack of error handling has
also been a problem. This new solution uses predefined
list of available offsets and picks the first free from
the list. This way it will fail only if we exceed the
maximum number of simultaneously supported offsets.

Fixes #2546.
closed 2020-03-18 by tlauda