thesofproject/sof #2547: [FEATURE] All codec or DMIC capture pipelines should contain high-pass IIR or DC block

singalsu, 2020-03-13:
**Is your feature request related to a problem? Please describe.**
The DMIC as direct attach, DMIC via codec, and codec headset input pipelines should include high-pass filtering function. The high-pass or DC blocking  feature in direct attach DMIC and several codecs seems to be insufficient to suppress the pop noise in capture beginning. The DC pulse is often much stronger than typical speech level. The default volume capture pipeline and the start volume ramp is not sufficient to prevent some strong DC to happen in in first hundreds of milliseconds of audio capture. 

**Describe the solution you'd like**
Instantiate the IIR EQ configured with high-pass response or the work-in-progress DC blocker component  to pipelines.

**Describe alternatives you've considered**
No alternatives. The volume ramp is already quite long and it is efficient in canceling the pop sound only for first tens on milliseconds even with with 200 - 500 ms linear ramp time. The volume ramp is also important to to preserve since the IIR and DC blocker take some time to settle their DC level.

**Additional context**
Add any other context or screenshots about the feature request here.

lgirdwood, 2020-03-13:
@singalsu would it make sense to only run the IIR/DC block st capture start then ramp it down -> off ? i.e. to save resources

singalsu, 2020-03-13:
No, the offset remains there all the time in worst case and it will vary in time due to temperature etc. If the component would automatically switch to bypass after e.g. 1s the offset would appear again.

The xtensa optimize IIR is very efficient. Also we can optimize the new DC blocker.

lgirdwood, 2020-03-13:
@singalsu ok great, btw we will need to look at quantifying the worst case load in cycles for each platform based on frames, channels formats etc.  This is easier for fixed cost processing like volume, but you may need to come up with ways to quantify for ASRC, SRC etc. Just something ti think about, needed for EDF scheduler updates.

singalsu, 2024-10-15:
This work was completed by merge of #9331.

closed 2024-10-15 by singalsu