thesofproject/sof #2551: [BUG] SOF tools build fails

singalsu, 2020-03-16:
**Describe the bug**

```
singalsu@singalsu-MOBL:~/work/current/sof/sof/tools/build_tools$ make
[  0%] Building C object probes/CMakeFiles/sof-probes.dir/probes_main.c.o
/home/singalsu/work/current/sof/sof/tools/probes/probes_main.c: In function ‘parse_data’:
/home/singalsu/work/current/sof/sof/tools/probes/probes_main.c:222:5: error: converting a packed ‘struct probe_data_packet’ pointer (alignment 1) to a ‘uint32_t’ {aka ‘unsigned int’} pointer (alignment 4) may result in an unaligned pointer value [-Werror=address-of-packed-member]
  222 |     w_ptr = (uint32_t *)packet;
      |     ^~~~~
In file included from /home/singalsu/work/current/sof/sof/tools/probes/probes_main.c:16:
/home/singalsu/work/current/sof/sof/tools/probes/../../src/include/ipc/probe.h:88:8: note: defined here
   88 | struct probe_data_packet {
      |        ^~~~~~~~~~~~~~~~~
/home/singalsu/work/current/sof/sof/tools/probes/probes_main.c:254:26: error: taking address of packed member of ‘struct probe_data_packet’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
  254 |      w_ptr = (uint32_t *)&packet->data;
      |                          ^~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [probes/CMakeFiles/sof-probes.dir/build.make:63: probes/CMakeFiles/sof-probes.dir/probes_main.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:91: probes/CMakeFiles/sof-probes.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```**To Reproduce**
Run scripts/build-tools.sh

**Reproduction Rate**
How often does the issue happen ? i.e. 1/10 (once in ten attempts), 1/1000 or all the time.
Does the reproduction rate vary with any other configuration or user action, if so please describe and show the new reproduction rate.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {SHA}
    * SOF: {SHA}
2) Name of the topology file
    * Topology: {FILE}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {PLATFORM}

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

paulstelian97, 2020-03-16:
I feel like this is a deja-vu. Does it happen on master or on which branch?

singalsu, 2020-03-16:
On git master.

singalsu, 2020-03-16:
Topologies can be build with this quick edit

```
diff --git a/tools/CMakeLists.txt b/tools/CMakeLists.txt
index 6d08afa746cd..d0e59314c001 100644
--- a/tools/CMakeLists.txt
+++ b/tools/CMakeLists.txt
@@ -14,7 +14,7 @@ project(SOF_TOOLS C)
 
 set(SOF_ROOT_SOURCE_DIRECTORY "${PROJECT_SOURCE_DIR}/..")
 
-add_subdirectory(probes)
+#add_subdirectory(probes)
 add_subdirectory(logger)
 add_subdirectory(ctl)
 add_subdirectory(topology)

```

paulstelian97, 2020-03-16:
@abonislawski Can you take a look?

abonislawski, 2020-03-16:
Looks like something new (pointer (alignment 4) may result in an unaligned pointer value) in GCC.
No warnings on GCC 7.3.0

lgirdwood, 2020-03-16:
@abonislawski Please update your GCC. Your version is now rather ancient.

plbossart, 2020-03-16:
Guys, can you please use Ubuntu 19.10 for development? I am also hit by this issue. Thanks

plbossart, 2020-03-16:
> Guys, can you please use Ubuntu 19.10 for development? I am also hit by this issue. Thanks

CI as well btw...

plbossart, 2020-03-16:
compile fix

````diff
diff --git a/tools/probes/probes_main.c b/tools/probes/probes_main.c
index f267fd3d..d9471169 100644
--- a/tools/probes/probes_main.c
+++ b/tools/probes/probes_main.c
@@ -31,7 +31,7 @@
 #define APP_NAME "sof-probes"
 
 #define PACKET_MAX_SIZE        4096    /**< Size limit for probe data packet */
-#define DATA_READ_LIMIT 1024   /**< Data limit for file read */
+#define DATA_READ_LIMIT 4096   /**< Data limit for file read */
 #define FILES_LIMIT    32      /**< Maximum num of probe output files */
 #define FILE_PATH_LIMIT 128    /**< Path limit for probe output files */
 
@@ -181,10 +181,10 @@ void parse_data(char *file_in)
        FILE *fd_in;
        struct wave_files files[FILES_LIMIT];
        struct probe_data_packet *packet;
-       uint32_t data[DATA_READ_LIMIT];
+       uint8_t data[DATA_READ_LIMIT];
        uint32_t total_data_to_copy = 0;
        uint32_t data_to_copy = 0;
-       uint32_t *w_ptr;
+       uint8_t *w_ptr;
        int i, j, file;
 
        enum p_state state = READY;
@@ -199,12 +199,13 @@ void parse_data(char *file_in)
        }
 
        packet = malloc(PACKET_MAX_SIZE);
-       memset(&data, 0, sizeof(uint32_t) * DATA_READ_LIMIT);
+       // FIXME: test allocation for Pete's sake.
+       memset(&data, 0, sizeof(uint8_t) * DATA_READ_LIMIT);
        memset(&files, 0, sizeof(struct wave_files) * FILES_LIMIT);
 
        /* data read loop to process DATA_READ_LIMIT bytes at each iteration */
        do {
-               i = fread(&data, sizeof(uint32_t), DATA_READ_LIMIT, fd_in);
+               i = fread(&data, sizeof(uint8_t), DATA_READ_LIMIT, fd_in);
                /* processing all loaded bytes */
                for (j = 0; j < i; j++) {
                        /* SYNC received */
@@ -218,8 +219,8 @@ void parse_data(char *file_in)
                                memset(packet, 0, PACKET_MAX_SIZE);
                                /* request to copy full data packet */
                                total_data_to_copy = sizeof(struct probe_data_packet) /
-                                       sizeof(uint32_t);
-                               w_ptr = (uint32_t *)packet;
+                                       sizeof(uint8_t);
+                               w_ptr = (uint8_t *)packet;
                                state = SYNC;
                        }
                        /* data copying section */
@@ -233,7 +234,7 @@ void parse_data(char *file_in)
                                        data_to_copy = total_data_to_copy;
                                        total_data_to_copy = 0;
                                }
-                               memcpy(w_ptr, data + j, data_to_copy * sizeof(uint32_t));
+                               memcpy(w_ptr, data + j, data_to_copy * sizeof(uint8_t));
````

[probe_fix.diff.txt](https://github.com/thesofproject/sof/files/4338905/probe_fix.diff.txt)

abonislawski, 2020-03-16:
Fixed in #2559

closed 2020-03-18 by lgirdwood

marc-hb, 2020-03-28:
Submitted different fix: `Revert __builtin_assume_aligned(4) probe packet, replace with aligned(4)` #2655

PR #2559 (2020-03-18): probes-app: build fix
Fixes #2551