thesofproject/sof #2553: Friendlier README

dandv, 2020-03-16:
I found this repo via the [ArchLinux page on getting sound to work](https://wiki.archlinux.org/index.php/Lenovo_ThinkPad_X1_Carbon_(Gen_7)#Volume_Controls) on this annoying downgrade of the ThinkPad X1C (sound and microphone worked fine on 6th gen machines, but not on 7th gen; how is this progress? Anyway). I'm a Linux user, but not a developer. I'd like like to have sound volume control, and a functional microphone.

Trying to build SOF, I saw that the [build instructions ](https://github.com/thesofproject/sof#build-instructions) were technical (even cryptic) right off the bat. I'm not familiar with CPU architectures, but figured out that's what "Apollolake" and "Cannonlake" must be, because they sound like something I had read about the CPU in the X1, "Whiskeylake". Turns out [they are](https://en.wikipedia.org/wiki/List_of_Intel_CPU_microarchitectures). But what about Baytrail and Cherrytrail? At this point, I have no idea what those are, because the aren't included in that list of microarchitectures linked from two sentences ago.

I understand this project is for developers. Can we also help casual users build the firmware?

Mainly, how should the user determine the microarchitecture they should use for the build process? I see Whiskey Lake in my Settings -> About, but "whiskey" doesn't appear anywhere in the README. What architecture should I choose? Could the build tool autodetect the architecture?

lgirdwood, 2020-03-16:
@deb-intel is this something you can take care of ?

plbossart, 2020-03-16:
@dandv if you build the firmware yourself, it will not run on the ThinkPad devices which are locked by a production key - the authentication will fail.
You can only compile+run the firmware on 'open' platforms, such as the Up2, Up Extreme boards and Chromebooks.

dandv, 2020-03-16:
@plbossart: I'm confused... That [ArchLinux page](https://wiki.archlinux.org/index.php/Lenovo_ThinkPad_X1_Carbon_(Gen_7)#Audio) advises ThinkPad X1C owners to install the SOF. Are some of [the releases](https://github.com/thesofproject/sof/releases) binaries compatible with the X1C 7th gen? Which extension?

plbossart, 2020-03-16:
@dandv they should have said 'install the binaries from https://github.com/thesofproject/sof-bin/tree/stable-v1.4.2'
The go.sh script should install everything and then the kernel will pick what's necessary for your platform (firmware binary and topology).

You will still need to make sure you have a UCM file installed from alsa-ucm-conf.

deb-intel, 2020-03-16:
> @deb-intel is this something you can take care of ?

@lgirdwood Let me see what I can do. I might ask you questions a little later. . . . .

lgirdwood, 2020-03-16:
@deb-intel thanks, it's more about answering questions that @dandv has raised regarding getting and deploying the firmware for casual users. i.e. using the sof-bin repo as @plbossart has described. Also some text on the Intel code names, probably we cant point to wikipedia ? My understanding is that initial SOF support should be 20.04 releases of RH/Ubuntu so users can obviously get the FW as part of their distro.

deb-intel, 2020-03-16:
@lgirdwood Do you think a developer should go through the README page again and update it? I don't think it has been technically reviewed in a while. I can definitely correct the confusion on the Intel code names and make grammatical content updates, but I think a real user/installer should review/update this page. Thoughts?

dandv, 2020-03-17:
I'm a real user and would be happy to test-drive the binaries, *if* there's a stable release.

Unfortunately I can't risk downtime due to unstable releases, or other things than sound breaking.

First off, the `go.sh` script is not implemented yet - https://github.com/thesofproject/sof-bin/blob/master/go.sh.

Second, how did they ArchLinux folks who wrote that wiki page manage to install everything? Maybe one of them could chip in with working instructions?

lgirdwood, 2020-03-17:
@deb-intel I can go through some of the details and send a PR, if you can do the codenames and clean it up for grammar.
@dandv thanks, and no need to risk downtime. I think the 20.04 distro releases onwards should have all the ingredients in place for users who want to test updated firmwares and build their own topologies. I guess the arch folks had the correct kernel patches and used the sof-bin repo (i.e. manually copied the binaries from sof-bin to /lib).

deb-intel, 2020-03-17:
@lgirdwood Deal! Thanks!

dandv, 2020-03-17:
@lgirdwood: I'm running 18.04 LTS, and updating to 20 would risk some downtime (or take some significant downtime to avoid the risk; I usually update to the next LTS when upgrading my laptop).

Any way I can test a stable build on Ubuntu 18.04?

lgirdwood, 2020-03-17:
@dandv sorry, I'm not really that knowledgeable about ubuntu packaging, I don't know if @tanuk plans to make deb packages for 18.04 ? This would also mean back porting kernel patches to 18.04 which is probably not trivial effort.

tanuk, 2020-03-18:
@lgirdwood I'm not involved with Ubuntu, are you mixing me with someone else?

lgirdwood, 2020-03-18:
@tanuk apologies if I am, are you still maintaining debian audio packages ?

tanuk, 2020-03-20:
I have never maintained any packages in Debian. I maintain audio packages in OpenEmbedded.

lgirdwood, 2020-03-20:
Thanks @tanuk, must be my age!  I do remember you working on Pulseaudio and audio userspace when you were at Intel. I think @dbaluta is taking care of packaging SOF for OE atm.

kv2019i, 2025-12-17:
No activity in 5 years, closing. Please reopen if still relevant.

closed 2025-12-17 by kv2019i