thesofproject/sof #2560: [BUG][JSL][KD] DSP panic happens when running KD pipleline

yongzhi1, 2020-03-16:
**Describe the bug**
Run KD pipeline with arecord returns:

[  223.798829] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  223.912254] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  223.952422] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[  223.957961] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  223.964977] sof-audio-pci 0000:00:1f.3: error: runtime exception
[  223.971032] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[  223.977351] sof-audio-pci 0000:00:1f.3: error: panic at :0
[  223.982886] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops

**To Reproduce**
arecord -Dhw:0,8 -M -N -c 2 -f S24_LE -r 16000 --buffer-size=68000 /tmp/tmp.wav -vvv -d 5

**Reproduction Rate**
All the time.

**Environment**
DSP FW uses SOF v0.1 release.

Topology is built with the following patch on top of jsl-001-drop-stable
https://github.com/thesofproject/sof/pull/2490

yongzhi1, 2020-03-16:
[dmesg-err-log.txt](https://github.com/thesofproject/sof/files/4340762/dmesg-err-log.txt)

tlauda, 2020-03-17:
@yongzhi1 Please try with #2562.

yongzhi1, 2020-03-17:
Hi, @tlauda, to try #2562 , can you share jsl_defconfig for xcc build? Thanks!!

yongzhi1, 2020-03-18:
I just tried daily/20200317/sof-df6c637e/fw/xcc/sof-jsl.ri.
md5sum sof-jsl.ri 
e61f34aa388c32a769023c0ebe1aafd8  Downloads/sof-jsl.ri

And here is the error:

[  126.921642] sof-audio-pci 0000:00:1f.3: error : DSP panic!
[  126.927875] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  126.935761] sof-audio-pci 0000:00:1f.3: error: runtime exception
[  126.942583] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[  126.949686] sof-audio-pci 0000:00:1f.3: error: panic at :0
[  126.955875] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[  126.962659] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[  126.976996] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000004 PS       0x00060925 SAR     0x00000000
[  126.989026] sof-audio-pci 0000:00:1f.3: EPC1     0xbe019906 EPC2     0xbe032dde EPC3     0x00000000 EPC4    0x00000000
[  127.001081] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  127.013127] sof-audio-pci 0000:00:1f.3: EPS2     0x00060f20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[  127.025152] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[  127.037182] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe070360
[  127.044208] sof-audio-pci 0000:00:1f.3: 0xbe070360: be019906 00000000 00000000 00000000
[  127.053241] sof-audio-pci 0000:00:1f.3: 0xbe070364: 00000000 00000000 be061500 00000000
[  127.062282] sof-audio-pci 0000:00:1f.3: 0xbe070368: 00000000 00000000 00000000 00000000
[  127.071286] sof-audio-pci 0000:00:1f.3: 0xbe07036c: c04fd1d4 ffffffff 53485028 ffff927b
[  127.080288] sof-audio-pci 0000:00:1f.3: 0xbe070370: 53485028 ffff927b 0dead000 00000000
[  127.089311] sof-audio-pci 0000:00:1f.3: 0xbe070374: afd09218 ffffffff 448abe78 ffffa2f3
[  127.098349] sof-audio-pci 0000:00:1f.3: 0xbe070378: 6f5add00 ffff927b afd09218 ffffffff
[  127.107360] sof-audio-pci 0000:00:1f.3: 0xbe07037c: d405ba00 fa68ba06 4045fbd0 ffffa2f3
[  127.419860] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60010000 size 108
[  127.428614] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  127.437256] sof-audio-pci 0000:00:1f.3: error: runtime exception
[  127.444449] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[  127.452557] sof-audio-pci 0000:00:1f.3: error: panic at :0
[  127.459349] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[  127.466553] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[  127.480921] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000004 PS       0x00060925 SAR     0x00000000
[  127.493009] sof-audio-pci 0000:00:1f.3: EPC1     0xbe019906 EPC2     0xbe032dde EPC3     0x00000000 EPC4    0x00000000
[  127.505017] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[  127.517025] sof-audio-pci 0000:00:1f.3: EPS2     0x00060f20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[  127.529094] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[  127.541098] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe070360
[  127.548058] sof-audio-pci 0000:00:1f.3: 0xbe070360: be019906 00000000 00000000 00000000
[  127.557106] sof-audio-pci 0000:00:1f.3: 0xbe070364: 00000000 00000000 be061500 00000000
[  127.566104] sof-audio-pci 0000:00:1f.3: 0xbe070368: 00000000 00000000 00000000 00000000
[  127.575102] sof-audio-pci 0000:00:1f.3: 0xbe07036c: 53485028 ffff927b 456071a0 ffff927b
[  127.584165] sof-audio-pci 0000:00:1f.3: 0xbe070370: 40d4f8c8 ffffa2f3 c04f3dce ffffffff
[  127.593157] sof-audio-pci 0000:00:1f.3: 0xbe070374: afcf2331 ffffffff 40d4f888 ffffa2f3
[  127.602156] sof-audio-pci 0000:00:1f.3: 0xbe070378: 00000014 00000000 40d4f910 ffffa2f3
[  127.611159] sof-audio-pci 0000:00:1f.3: 0xbe07037c: 45607128 ffff927b 46423400 ffff927b
[  127.620238] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[  127.630411] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  127.639615] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  127.650607] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 1
[  127.658913] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -110
[  127.667326]  DMIC16k: ASoC: hw_params FE failed -110
[  130.659886] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[  130.668531] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[  130.676382] sof-audio-pci 0000:00:1f.3: error: runtime exception
[  130.683153] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[  130.690202] sof-audio-pci 0000:00:1f.3: error: panic at :0
[  130.696387] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[  130.703150] sof-audio-pci 0000:00:1f.3: error: Exception Cause: LoadStorePIFDataErrorCause, Synchronous PIF data error during LoadStore access
[  130.717495] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000000d EXCVADDR 0x00000004 PS       0x00060925 SAR     0x00000000
[  130.729494] sof-audio-pci 0000:00:1f.3: EPC1     0xbe019906 EPC2     0xbe032dde EPC3     0x00000000 EPC4    0x00000000
[... cut ...]
[  130.838540] sof-audio-pci 0000:00:1f.3: 0xbe070378: 0000000c 00000000 4017fc40 ffffa2f3
[  130.847536] sof-audio-pci 0000:00:1f.3: 0xbe07037c: 45607128 ffff927b 00000001 00000000
[  130.856557] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[  130.866732] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[  130.875942] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[  130.886880] sof-audio-pci 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend

tlauda, 2020-03-18:
@yongzhi1 Fix is not yet on master. You need to build it manually with #2562 on top of it.

yongzhi1, 2020-03-18:
Thanks @tlauda , with that now I have:
localhost ~ # sof-logger -l /home/sof-jsl.ldc   
 CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1          KPB 8.41       506767.760417    506767.750000      src/audio/kpb.c:596 	kpb_copy() error: nothing to copy sink->free 0 source->avail 1296
    0      1          KPB 8.41       546768.072917     40000.312500      src/audio/kpb.c:596 	kpb_copy() error: nothing to copy sink->free 0 source->avail 2576
    0      1          KPB 8.41       586769.218750     40001.144531      src/audio/kpb.c:596 	kpb_copy() error: nothing to copy sink->free 0 source->avail 3856
    0      1          KPB 8.41       626767.656250     39998.437500      src/audio/kpb.c:596 	kpb_copy() error: nothing to copy sink->free 0 source->avail 5136
    0      1          KPB 8.41       666767.604167     39999.949219      src/audio/kpb.c:596 	kpb_copy() error: nothing to copy sink->free 0 source->avail 6416
    0      1          KPB 8.41       706765.572917     39997.968750      src/audio/kpb.c:596 	kpb_copy() error: nothing to copy sink->free 0 source->avail 7680
    0      1          KPB 8.41       746656.927083     39891.355469      src/audio/kpb.c:596 	kpb_copy() error: nothing to copy sink->free 0 source->avail 7680
    0      1          KPB 8.41       786656.718750     39999.792969      src/audio/kpb.c:596 	kpb_copy() error: nothing to copy sink->free 0 source->avail 7680
    0      1          DMA            826654.427083     39997.707031 src/drivers/dw/dma.c:1072	dw_dma_get_data_size() error: xrun detected
    0      1          DAI 8.44       826661.562500         7.135417      src/audio/dai.c:622 	dai_report_xrun() error: overrun due to no data available
    0      1         COMP            826667.552083         5.989583      src/audio/dai.c:694 	comp_overrun() error: dev->comp.id = 44, sink->free = 0, copy_bytes = 0
    0      1         PIPE            826732.552083        65.000000 src/audio/pipeline.c:866 	pipeline_copy() error: ret = -61, start->comp.id = 44, dir = 0
    0      1         PIPE 8.45       826741.145833         8.593750 src/audio/pipeline.c:1037	pipeline_task(): xrun recover failed! pipeline will be stopped!

My FW is built on top of jsl-001-drop-stable with the two commits in #2562.

tlauda, 2020-03-18:
@yongzhi1 Could you check whether the same issue happens on master + #2562? It might be either some missing hotfix or some kind of bug in topology.

yongzhi1, 2020-03-18:
Hi, @tlauda ,

Applied #2562 on top of master branch at:
6749a36c3c33 ipc: buffer: Properly free buffer using buffer_free instead of rfree.
6e43ab8ecd9a ipc: buffer: Remove obsolete rfree on NULL pointer.

The logger complained version
localhost ~ # sof-logger -l /home/sof-jsl.ldc 
Error: abi version in /home/sof-jsl.ldc file does not coincide with abi version used by logger.
logger ABI Version is 4:2:0
ldc_file ABI Version is 3:14:0

sof-logger -l /home/sof-jsl.ldc 
Error: abi version in /home/sof-jsl.ldc file does not coincide with abi version used by logger.
logger ABI Version is 4:2:0
ldc_file ABI Version is 3:14:0

Force version checking ignored:
 
         TIMESTAMP              DELTA  C# COMPONENT          LOCATION                       CONTENT
[1678733532.083333] (             nan) c0 dmic-dai     2.0   ....../intel/cavs/dmic.c:1532  ERROR dmic_irq_handler() error: full fifo B or PDM overrun
[1678733657.187500] (      125.104164) c0 dmic-dai     2.0   ....../intel/cavs/dmic.c:1532  ERROR dmic_irq_handler() error: full fifo B or PDM overrun
[1678733782.343750] (      125.156250) c0 dmic-dai     2.0   ....../intel/cavs/dmic.c:1532  ERROR dmic_irq_handler() error: full fifo B or PDM overrun
[1678733907.604167] (      125.260414) c0 dmic-dai     2.0   ....../intel/cavs/dmic.c:1532  ERROR dmic_irq_handler() error: full fifo B or PDM overrun
[1678734032.552083] (      124.947914) c0 dmic-dai     2.0   ....../intel/cavs/dmic.c:1532  ERROR dmic_irq_handler() error: full fifo B or PDM overrun
[1678734157.447917] (      124.895836) c0 dmic-dai     2.0   ....../intel/cavs/dmic.c:1532  ERROR dmic_irq_handler() error: full fifo B or PDM overrun
[1678734282.239583] (      124.791664) c0 dmic-dai     2.0   ....../intel/cavs/dmic.c:1532  ERROR dmic_irq_handler() error: full fifo B or PDM overrun
[1678734407.343750] (      125.104164) c0 dmic-dai     2.0   ....../intel/cavs/dmic.c:1532  ERROR dmic_irq_handler() error: full fifo B or PDM overrun

tlauda, 2020-03-18:
@yongzhi1 Looks like problem with DMIC.

yongzhi1, 2020-03-18:
Thanks @tlauda, will find a setup with working DMIC and share the results.

tlauda, 2020-03-20:
@yongzhi1 Are you sure the topology is correct?
@keyonjie Could help verify if #2490 is OK?
KWD is working with our slim driver on JSL.

yongzhi1, 2020-03-20:
Hi @tlauda , @keyonjie, I double checked the topology, but did not catch anything unusual: 

[image]

mrajwa, 2020-04-01:
@yongzhi1 please provide FW & Kernel commit ID + the topology file you are using.

PS: The topology graph you provided above is not very readable and likely not correct

yongzhi1, 2020-04-03:
Hi, @mrajwa,

Here is the kernel I am using:
https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-5.4 

Just rebased topology  #2490 and rebuild FW at 
d00c8d599bae kpb: perform additional pointer sanity checks.

With the follow cmd, the log showed failure at validate_host_params() for HOST_BUFFER_MIN_SIZE(hb_size_req) > host_buffer_size:

# arecord -Dhw:0,6 -M -N -c 2 -f S24_LE -r 16000 --buffer-size=64000 /tmp/tmp.wav -vvv -d 5
Recording WAVE '/tmp/tmp.wav' : Signed 24 bit Little Endian, Rate 16000 Hz, Stereo
arecord: set_params:1403: Unable to install hw params:
ACCESS:  MMAP_INTERLEAVED
FORMAT:  S24_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 64
CHANNELS: 2
RATE: 16000
PERIOD_TIME: 999000
PERIOD_SIZE: 15984
PERIOD_BYTES: 127872
PERIODS: (2 3)
BUFFER_TIME: 1999500
BUFFER_SIZE: 31992
BUFFER_BYTES: 255936
TICK_TIME: 0
[err-log-buffer-size-001.txt](https://github.com/thesofproject/sof/files/4424381/err-log-buffer-size-001.txt)

Thanks for help!!

mrajwa, 2020-04-03:
@yongzhi1, you are missing a kernel patch from @keyonjie I believe. Clearly you are asking for host buffer size 64000 frames but getting about half of it. @keyonjie, can you point @yongzhi1 to the patch which fixes the problem with setting host buffer size? I've checked his topology and it looks OK in terms of min/max settings.

mrajwa, 2020-04-03:
@yongzhi1, can you please try to add this kernel patch https://github.com/thesofproject/linux/pull/1717/commits/bf82c24a59b83f0d116c8f969887ad78471f595d

keyonjie, 2020-04-03:
> Hi @tlauda , @keyonjie, I double checked the topology, but did not catch anything unusual:
> 
> [image]

Sorry, worked on TGL and didn't notice the tagging to me.

@yongzhi1 there is issue with your tplg, the KWD pipelines are mixed. let me review your .m4 file and give some comments.

keyonjie, 2020-04-03:
@yongzhi1 looks your new version works now?

[image]

yongzhi1, 2020-04-04:
Thanks @mrajwa and @keyonjie,

Cherry-picked the 2 patches, with that:

1) The same error still happens for buffer-size=64000
arecord -Dhw:0,6 -M -N -c 2 -f S16_LE -r 16000 --buffer-size=64000 /tmp/tmp.wav -vvv -d 5
Recording WAVE '/tmp/tmp.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
arecord: set_params:1403: Unable to install hw params:
ACCESS:  MMAP_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 16000
PERIOD_TIME: 1000000
PERIOD_SIZE: 16000
PERIOD_BYTES: 64000
PERIODS: 4
BUFFER_TIME: 4000000
BUFFER_SIZE: 64000
BUFFER_BYTES: 256000
TICK_TIME: 0

On FW side:

[296167053.125000] (        7.552083) c0 kpb          8.42           src/audio/kpb.c:1526  ERROR ZZZ HOST_BUFFER_MIN_SIZE = 268800 host_buffer_size = 256000

2) The host buffer size related error is not reproducible with  --buffer-size=67200:
arecord -Dhw:0,6 -M -N -c 2 -f S24_LE -r 16000 --buffer-size=67200 /tmp/tmp.wav -vvv -d 5

However, observed other errors:

[391686736.666667] (     2252.968750) c0 MEM                         src/lib/alloc.c:283   ERROR 171 blocks needed for allocation but only 170 blocks are remaining
[391842651.770833] (        8.333333) c0 kpb          8.42           src/audio/kpb.c:671   ERROR kpb_copy(): nothing to copy sink->free 0 source->avail 2592
[391882650.833333] (        8.020833) c0 kpb          8.42           src/audio/kpb.c:671   ERROR kpb_copy(): nothing to copy sink->free 0 source->avail 5120
[391922479.635417] (        8.333333) c0 kpb          8.42           src/audio/kpb.c:671   ERROR kpb_copy(): nothing to copy sink->free 0 source->avail 5120
[391962475.625000] (        8.020833) c0 kpb          8.42           src/audio/kpb.c:671   ERROR kpb_copy(): nothing to copy sink->free 0 source->avail 5120
[392002472.291667] (        8.020833) c0 kpb          8.42           src/audio/kpb.c:671   ERROR kpb_copy(): nothing to copy sink->free 0 source->avail 5120
[392042351.979167] (        8.229167) c0 kpb          8.42           src/audio/kpb.c:671   ERROR kpb_copy(): nothing to copy sink->free 0 source->avail 5120
[392082350.781250] (        8.177083) c0 kpb          8.42           src/audio/kpb.c:671   ERROR kpb_copy(): nothing to copy sink->free 0 source->avail 5120
[392122288.750000] (        8.229167) c0 DMA                    src/drivers/dw/dma.c:1063  ERROR dw_dma_get_data_size(): xrun detected
[392122299.427083] (       10.677083) c0 dai          8.46           src/audio/dai.c:648   ERROR dai_report_xrun(): overrun due to no data available
[392122309.791667] (       10.364583) c0 dai          8.46           src/audio/dai.c:657   ERROR comp_overrun(): sink->free = 0, copy_bytes = 0
[392122406.822917] (       11.666667) c0 PIPE                   src/audio/pipeline.c:820   ERROR pipeline_copy(): ret = -61, start->comp.id = 46, dir = 0
[392122419.322917] (       12.500000) c0 PIPE         8.47      src/audio/pipeline.c:992   ERROR pipeline_task(): xrun recover failed! pipeline will be stopped!
[411587069.843750] (        7.500000) c0 dai          8.46     src/audio/component.c:153   ERROR comp_set_state(): wrong state = 1, COMP_TRIGGER_STOP
[411587082.395833] (       12.552083) c0 PIPE                   src/audio/pipeline.c:692   ERROR pipeline_trigger(): ret = -22, host->comp.id = 40, cmd = 0
[411587095.104167] (       12.708333) c0 IPC                       src/ipc/handler.c:468   ERROR ipc: comp 40 trigger 0x50000 failed -22

Attached log with PDM overrun traces commented out because of the over flooding.
[buf-size-67200-log.txt](https://github.com/thesofproject/sof/files/4432227/buf-size-67200-log.txt)

Somehow my topology is still different than the one you showed above.
[image]

keyonjie, 2020-04-05:
@yongzhi1 please use buffer-size>=67200 otherwise the fw will reject. Yes, your tplg looks odd, the one I showed above is built after applied your tplg pr, let me attach my .tplg file.

keyonjie, 2020-04-05:
Hi @yongzhi1 your detect pipeline is not started, looks the DAPM widget event not happens, looks your kernel is missing a lot of SOF commits upstreamed after 5.4, so I suggest to backport those commits to 5.4 kernel first.

yongzhi1, 2020-04-05:
Hi, @keyonjie, the kernel branch we are using is the same as TGL/volteer, should not we expect same behavior?

keyonjie, 2020-04-07:
Hi @yongzhi1 the DAPM widget power state doesn't work on your KWD pipelines, there is similar issue reported for TGL smart amplifier here: https://github.com/thesofproject/linux/issues/1987

Need further debug to root cause it.

keyonjie, 2020-04-07:
Hi @yongzhi1, can you help try if https://github.com/thesofproject/linux/pull/1993 can fix the issue?

yongzhi1, 2020-04-07:
Hi, @keyonjie, confirmed your fix, think this issue can be closed.