thesofproject/sof #2569: [BUG][CML-H]“error：can't find BE for DAI ALH0x202.OUT” display after booting up successfully.

sinahuang, 2020-03-18:
**Describe the bug**
“error：can't find BE for DAI ALH0x202.OUT” display after booting up successfully. 

**To Reproduce**
1. Boot up system.
2. Check the dmesg.

**Reproduction Rate**
5/5

**Expected behavior**
Device can boot up successfully, no error.

**Impact**
“error：can't find BE for DAI ALH0x202.OUT” display after booting up successfully.

**Test recipe**

Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev  commit:45f6424
FW: https://github.com/thesofproject/sof/commits/master commit:  a133d5
Tplg: Same with FW branch,  sof-cml-rt711-rt1308-rt715.tplg
Platform:CML-H Laptop with codec ALC711, 2 x RT1308, RT715 in SDW mode.

[dmesg18.log](https://github.com/thesofproject/sof/files/4347072/dmesg18.log)

sinahuang, 2020-03-18:
Reverted commit :https://github.com/thesofproject/sof/commit/5e8de9fd52e67d96a57445b279b2b7947505bf6e, issue can't reproduce on CML-H.

plbossart, 2020-03-18:
Yeah, we need to be more careful with topology changes, we need the matching machine driver to be ready as well, or refer to a PR.

bardliao, 2020-03-19:
It should be "fixed" if we remove the [SOF_SDW_NO_AGGREGATED ](https://github.com/thesofproject/linux/blob/topic/sof-dev/sound/soc/intel/boards/sdw_rt711_rt1308_rt715.c#L247)flag. But we still need #2539 #2540 or alternative PR to make synchronized playback work.

tlauda, 2020-03-19:
@bardliao Are you going to change something in topology or should we move this bug to linux repo?

bardliao, 2020-03-19:
> @bardliao Are you going to change something in topology or should we move this bug to linux repo?

Can we ignore this issue and #2570 until #2539 and #2540 are merged? I will send a kernel PR to remove SOF_SDW_NO_AGGREGATED after #2539 and #2540 are merged.

bardliao, 2020-03-24:
This is due to mismatch between machine driver and topology. It will be fixed by thesofproject/linux#1928

sinahuang, 2020-03-24:
Issue can't be reproduced with sof-dev (commit:9b5b6c2a3)+pr1928 and master(commit:d12ad78) +pr2569+pr2540+pr2541.

closed 2020-03-26 by tlauda