thesofproject/sof #2570: [BUG][CML]" error: ipc error for 0x60010000 size 20" display when aplay via speaker0,2.

sinahuang, 2020-03-18:
**Describe the bug**
Speaker0,2 can't work normally, dmesg show " error: ipc error for 0x60010000 size 20".

**To Reproduce**
1.Boot up system.
2.aplay -Dhw:0,2 -fs16_le -c2 -r 48000 -vv -i wav/48California_Dreaming.wav

**Reproduction Rate**
5/5

**Expected behavior**
Aplay0,2  should work normally.

**Impact**
" error: ipc error for 0x60010000 size 20" display when aplay via speaker0,2.
```
[   43.225747] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   43.225770] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[   43.225776] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[   43.225778] sof-audio-pci 0000:00:1f.3: periods:4
[   43.225800] sof-audio-pci 0000:00:1f.3: stream_tag 1
[   43.225812] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   43.226331] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[   43.226343] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 1
[   43.226348] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[   43.226353]  SDW1-speakers: ASoC: hw_params FE failed -22
[   43.226364] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[   43.226379] intel-master sdw-master-1: sdw_deprepare_stream: failed -22
[   43.226396] sof-audio-pci 0000:00:1f.3: pcm: free stream 2 dir 0
[   43.226404] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[   43.226415] intel-master sdw-master-1: sdw_deprepare_stream: failed -22
[   43.226886] sof-audio-pci 0000:00:1f.3: pcm: close stream 2 dir 0
[   43.636422] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[   43.636449] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   44.136257] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[   44.136292] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   46.543261] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x201
```
```
ubuntu@cd-CML-H-rt711-rt1308-rt715:~$ aplay -Dhw:0,2 -r48000 -c2 -fs16_le wav/48K_Let_It_Go.wav -vv -i
Playing WAVE 'wav/48K_Let_It_Go.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 85000
PERIOD_SIZE: 4080
PERIOD_BYTES: 16320
PERIODS: (4 5)
BUFFER_TIME: 341000
BUFFER_SIZE: 16368
BUFFER_BYTES: 65472
TICK_TIME: 0
```

**Test recipe**

Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev  commit:45f6424
FW: https://github.com/thesofproject/sof/commits/master commit:  a133d5
Tplg: Same with FW branch,  sof-cml-rt711-rt1308-rt715.tplg
Platform:CML-H Laptop with codec ALC711, 2 x RT1308, RT715 in SDW mode.

[dmesg20.log](https://github.com/thesofproject/sof/files/4347372/dmesg20.log)

YvonneYang2, 2020-03-18:
Reverted https://github.com/thesofproject/sof/commit/5e8de9fd52e67d96a57445b279b2b7947505bf6e , issue is gone on CML-U

sinahuang, 2020-03-18:
Reverted commit :https://github.com/thesofproject/sof/commit/5e8de9fd52e67d96a57445b279b2b7947505bf6e , issue can't reproduce on CML-H.

bardliao, 2020-03-24:
The issue is ``error: can't find BE for DAI ALH0x202.OUT`` This is due to mismatch between machine driver and topology. It will be fixed by https://github.com/thesofproject/linux/pull/1928

sinahuang, 2020-03-24:
Issue can't be reproduced with sof-dev (commit:9b5b6c2a3)+pr1928 and master(commit:d12ad78) +pr2569+pr2540+pr2541.

PR #2603 (2020-03-25): Topology: sof-icl-rt711-rt1308-rt715-hdmi: Use demux for platforms with multiple speakers
We only need demux for two speaker dais case.

Fixes: 5e8de9f "topology: sof-icl-rt711-rt1308-rt715-hdmi: Merge two pipeline with demux"

Fixes #2570
closed 2020-03-25 by plbossart