thesofproject/sof #2572: [FEATURE] Is it possible to run sof and along with the Jack connection kit?

buscon, 2020-03-18:
Hi,

Is it possible to run sof and along with the Jack connection kit?
I tried to do that but I get:
Wed Mar 18 10:02:08 2020: creating alsa driver ... hw:sofsklhdacard|hw:sofsklhdacard|1024|3|48000|0|0|nomon|swmeter|-|32bit
Wed Mar 18 10:02:08 2020: configuring for 48000Hz, period = 1024 frames (21.3 ms), buffer = 3 periods
Wed Mar 18 10:02:08 2020: ALSA: final selected sample format for capture: 32bit integer little-endian
Wed Mar 18 10:02:08 2020: ERROR: ALSA: cannot set period size to 1024 frames for capture
Wed Mar 18 10:02:08 2020: ERROR: ALSA: cannot configure capture channel
Wed Mar 18 10:02:08 2020: Released audio card Audio0
Wed Mar 18 10:02:08 2020: ERROR: Cannot initialize driver
Wed Mar 18 10:02:08 2020: ERROR: JackServer::Open failed with -1
Wed Mar 18 10:02:08 2020: ERROR: Failed to open server

lgirdwood, 2020-03-18:
@buscon there should be nothing preventing this except maybe the topology doesn't support the jack config (I think you are also probably the first person to try Jack). Any chance you can share your dmesg, plus any other logs (sof-logger will dump FW logs). 
It does look like the topology you are using does not support 32bit via the PCM (but pipeline may do internally within DSP), can you try 24 or 16bit PCM ?

buscon, 2020-03-18:
Thank you for your answer. 

Dmesg does log something else, maybe it tells you something? Here it is:
[11276.589313] snd_hda_codec_realtek ehdaudio0D0: didn't find PCM for DAI Digital Codec DAI
[11276.589314] snd_hda_codec_realtek ehdaudio0D0: ASoC: can't open codec Digital Codec DAI: -22
[11276.589315]  Digital Playback and Capture: ASoC: BE open failed -22
[11276.589316]  HDA Digital: ASoC: failed to start some BEs -22

Where do I find the sof-logger?

I tried setting Jack to 16bit but I get the same error.

plbossart, 2020-03-18:
@buscon can you first try with plain vanilla aplay and use hw:0,0? I don't think I've seen anyone use the Digital (SPDIF) output as well. and please attach the full dmesg log and point us to the link where the alsa-info.sh results are stored, thank you

lgirdwood, 2020-03-18:
@buscon sof-logger is part of the sof repo, IIRC scripts/build-tools.sh should build it.

buscon, 2020-03-18:
> @buscon can you first try with plain vanilla aplay and use hw:0,0? I don't think I've seen anyone use the Digital (SPDIF) output as well. and please attach the full dmesg log and point us to the link where the alsa-info.sh results are stored, thank you

I did not mention that, but pulseaudio works without problems, so aplay with hw:0,0 works too.,
The alsa-info.sh results are here:
http://alsa-project.org/db/?f=178d7ffbb4b7854b1189531af241bf8019fb8cd7

Dmesg:
[dmesg.log](https://github.com/thesofproject/sof/files/4350309/dmesg.log)

buscon, 2020-03-18:
> @buscon sof-logger is part of the sof repo, IIRC scripts/build-tools.sh should build it.

ok, so far I've been using the already compiled version, so I do not have that for now.

Sciss, 2021-11-22:
Hi there. I think I run into the issue listed here. The sound card is listed by aplay as

```
card 1: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
```
It works with Pulse Audio, and some ALSA applications like aplay, but jackd has lots of trouble with it. Finding a bug [listed for Ubuntu](https://bugs.launchpad.net/ubuntu/+source/jackd-defaults/+bug/1872244), I managed to make them work together by setting the block size to 2016, and then forcing the jack client, in my case SuperCollider, on a block size that is an integer division of 2016, so 32. Also I have to select force-16-bit, otherwise sound is distorted.

Why is such an irregular block size that is not a power of two needed, and why it doesn't work with jack 32-bit? I think this is a bug not a feature request. Thanks

buscon, 2021-11-22:
@Sciss thanks for your suggestion and temporary solution!

I have one question though: what do you mean with block size 2016? Do you mean the buffer size? the possible value is actually 2048.

lgirdwood, 2021-11-22:
@Sciss fwiw, the minimum (kconfig default) block size used by FW is 1ms at 48kHz, so 48 * frame size. Have you been able to try this block size with your buffer config ?

plbossart, 2021-11-22:
IIRC we allowed the topologies to work with periods that were not aligned with 1ms, precisely for Jack support and power-of-two requirements. if this was the case, we might have a regression indeed.

@kv2019i does this rings a bell?

Sciss, 2021-11-22:
>  if this was the case, we might have a regression indeed.

I should mention I run on Debian 11; perhaps it was fixed in a newer version?

```
$ sudo apt list --installed | grep sof
...
firmware-sof-signed/stable,now 1.7-1 all [installed]
```

Sciss, 2021-11-22:
> fwiw, the minimum (kconfig default) block size used by FW is 1ms at 48kHz, so 48 * frame size. Have you been able to try this block size with your buffer config ?

Ah, I see. Well, this means I can run, for example with jack buffer size 768, and keep my jack client SuperCollider at its own default of 64 samples (which must be an integer division of the jack buffer size). So that's not too bad, just weird in the sense that programs like QJackCtl suggest a power of two.

kv2019i, 2025-12-17:
We run SOF with jackd in CI every day, so this surely is possible nowadays. Closing. Please open bugs if there are issues left. Thanks!

closed 2025-12-17 by kv2019i