thesofproject/sof #2585: [BUG][BDW]“DSP panic” when test "Check-playback: Stress - 100x"

Liviali155, 2020-03-19:
Describe the bug
“DSP panic” when test "Check-playback: Stress - 100x",error occured when aplay via (0,0),after error occured,all the pipeline can not work,reboot can recovery

used device: sh-bdw-wsb-rt286-03
```
ubuntu@sh-bdw-wsb-rt286-03:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofbroadwellrt2 [sof-broadwell-rt286], device 0: Low Latency (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofbroadwellrt2 [sof-broadwell-rt286], device 1: Media Playback 1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

To Reproduce
1.sudo reboot
2.cd sof-test
3.cd test-case
4. export TPLG=/lib/firmware/intel/sof-tplg/sof-bdw-rt286.tplg
5. ./check-playback.sh -d 1 -l 100 -r 1(https://github.com/thesofproject/sof-test/blob/master/test-case/check-playback.sh)

Reproduction Rate
2/2
1.failed at 55th of 100
2.failed at 43th of 100

Expected behavior
There is no DSP panic when aplay/arecord

Actual result
DSP panic,all the pipeline can not work

[image]

<details>
 <summary>dmesg_error.log</summary>

````
kernel: [ 3047.127793] sof-audio-acpi INT3438:00: ipc tx: 0x60030000: GLB_STREAM_MSG: PCM_FREE
kernel: [ 3047.127878] sof-audio-acpi INT3438:00: ipc tx succeeded: 0x60030000: GLB_STREAM_MSG: PCM_FREE
kernel: [ 3047.128065] sof-audio-acpi INT3438:00: pcm: free stream 0 dir 0
kernel: [ 3047.128206] sof-audio-acpi INT3438:00: pcm: close stream 0 dir 0
kernel: [ 3050.353860] sof-audio-acpi INT3438:00: error : DSP panic!
kernel: [ 3050.353875] sof-audio-acpi INT3438:00: panic: dsp_oops_offset 649216 offset 649216
kernel: [ 3050.353953] sof-audio-acpi INT3438:00: error: can't enter idle
kernel: [ 3050.353959] sof-audio-acpi INT3438:00: error: trace point 00000800
kernel: [ 3050.353965] sof-audio-acpi INT3438:00: error: panic at src/lib/agent.c:56
kernel: [ 3050.353971] sof-audio-acpi INT3438:00: error: DSP Firmware Oops
kernel: [ 3050.353978] sof-audio-acpi INT3438:00: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060125 SAR     0x00000000
kernel: [ 3050.353984] sof-audio-acpi INT3438:00: EPC1     0x00000000 EPC2     0x00000000 EPC3     0x00017e25 EPC4    0x00000000
kernel: [ 3050.353989] sof-audio-acpi INT3438:00: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
kernel: [ 3050.354036] sof-audio-acpi INT3438:00: EPS2     0x00000000 EPS3     0x00060520 EPS4     0x00000000 EPS5    0x00000000
kernel: [ 3050.354045] sof-audio-acpi INT3438:00: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00002440
kernel: [ 3050.354051] sof-audio-acpi INT3438:00: stack dump from 0x0040a600
kernel: [ 3050.354071] sof-audio-acpi INT3438:00: 0x0040a600: 00409710 00000000 0040a620 00000028
kernel: [ 3050.354092] sof-audio-acpi INT3438:00: 0x0040a604: 00000000 00000000 00000038 00014b4c
kernel: [ 3050.354116] sof-audio-acpi INT3438:00: 0x0040a608: 3e9de000 6b3f1450 c541e828 ffff9da2
kernel: [ 3050.354133] sof-audio-acpi INT3438:00: 0x0040a60c: c541e828 ffff9da2 00000000 00000000
kernel: [ 3050.354145] sof-audio-acpi INT3438:00: 0x0040a610: c4882d80 ffff9da2 b9ec0960 ffffffff
````
</details>

Platform: BDW  with onboard codec RT286 in WSB mode
Firmware: f1fa3be9 https://github.com/thesofproject/sof/commits/master
Kernel: e4ba562e https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topolog/ sof-bdw-rt286.tplg same as firmware

[dmesg.txt](https://github.com/thesofproject/sof/files/4353940/dmesg.txt)
[etrace.txt](https://github.com/thesofproject/sof/files/4353941/etrace.txt)
[slogger.txt](https://github.com/thesofproject/sof/files/4353942/slogger.txt)

Liviali155, 2020-03-19:
same issue as https://github.com/thesofproject/sof/issues/2532

tlauda, 2020-03-19:
@Liviali155 The patch is already merged. Could you bisect?

Liviali155, 2020-03-20:
Changed firmware to 20200214(commit:4d17d0de-XCC),issue can still be reproduced
Changed firmware to 20200103(commit:0194175d-GCC),no DSP panic,but there is "error: ipc timed out for 0x60010000 size 108" when test "Check-playback: Stress - 100x",log as below:

<details>
 <summary>sof_errortrace.log</summary>

````
      CORE  LEVEL      COMP_ID                TIMESTAMP            DELTA                FILE_NAME	CONTENT
    0      1           SA         39207763832.760422              nan      src/lib/agent.c:56  	validate(), ll drift detected, delta = 639998
    0      2          IPC          65636320.416667              nan    src/ipc/handler.c:916 	ipc: pipe 2 comp 16 -> new (type 2)
    0      2          DAI          65636381.510417        61.093750      src/audio/dai.c:145 	dai_new()
    0      2          DAI          65636580.416667       198.906250        src/lib/dai.c:70  	dai_get(), d = 0x4022a0, sref = 3
    0      2          DMA          65636622.083333        41.666668        src/lib/dma.c:128 	dma_get() ID 1 sref = 6 busy channels 1
    0      2          IPC          65637528.854167       906.770813    src/ipc/handler.c:974 	ipc: pipe 2 -> new
    0      2         PIPE          65637619.375000        90.520836 src/audio/pipeline.c:50  	pipeline_new()
    0      2          IPC          65638293.333333       673.958313    src/ipc/handler.c:646 	ipc: pm core mask 0x1 -> enable
    0      2          IPC          65639441.145833      1147.812500    src/ipc/handler.c:913 	ipc: hdr 0x30010000 tx (76) > rx (28)
    0      2          IPC          65639474.270833        33.125000    src/ipc/handler.c:916 	ipc: pipe 3 comp 19 -> new (type 1)
    0      2         HOST          65639545.625000        71.354164     src/audio/host.c:357 	host_new()
    0      2          DMA          65639782.239583       236.614578        src/lib/dma.c:128 	dma_get() ID 1 sref = 7 busy channels 1
    0      2          IPC          65640854.427083      1072.187500    src/ipc/handler.c:913 	ipc: hdr 0x30010000 tx (84) > rx (28)
    0      2          IPC          65640887.552083        33.125000    src/ipc/handler.c:916 	ipc: pipe 3 comp 20 -> new (type 5)
    0      2       VOLUME          65640959.322917        71.770836 udio/volume/volume.c:154 	volume_new()
    0      2       VOLUME 3.20     65641225.781250       266.458344 udio/volume/volume.c:235 	vol->initial_ramp = 250, vol->ramp = 0, vol->min_value = 0, vol->max_value = 65536
    0      2          IPC          65642191.562500       965.781250    src/ipc/handler.c:913 	ipc: hdr 0x30010000 tx (76) > rx (28)
    0      2          IPC          65642224.687500        33.125000    src/ipc/handler.c:916 	ipc: pipe 3 comp 21 -> new (type 8)
    0      2          SRC          65642296.927083        72.239586  src/audio/src/src.c:474 	src_new()
    0      2          IPC          65643285.677083       988.750000    src/ipc/handler.c:946 	ipc: pipe 3 buffer 22 -> new (0x1800 bytes)
    0      2       BUFFER          65643346.770833        61.093750   src/audio/buffer.c:66  	buffer_new()
    0      2       BUFFER          65643381.510417        34.739582   src/audio/buffer.c:25  	buffer_alloc()
    0      2          IPC          65662046.302083     18664.791016    src/ipc/handler.c:946 	ipc: pipe 3 buffer 23 -> new (0x1800 bytes)
    0      2       BUFFER          65662108.697917        62.395832   src/audio/buffer.c:66  	buffer_new()
    0      2       BUFFER          65662143.437500        34.739582   src/audio/buffer.c:25  	buffer_alloc()
    0      2          IPC          65665648.593750      3505.156250    src/ipc/handler.c:946 	ipc: pipe 3 buffer 24 -> new (0x1800 bytes)
    0      2       BUFFER          65665712.291667        63.697918   src/audio/buffer.c:66  	buffer_new()
    0      2       BUFFER          65665747.031250        34.739582   src/audio/buffer.c:25  	buffer_alloc()
    0      2          IPC          65669145.104167      3398.072998    src/ipc/handler.c:974 	ipc: pipe 3 -> new
    0      2         PIPE          65669242.864583        97.760414 src/audio/pipeline.c:50  	pipeline_new()
    0      2          IPC          65670013.750000       770.885437    src/ipc/handler.c:646 	ipc: pm core mask 0x1 -> enable
    0      2          IPC          65671949.583333      1935.833374    src/ipc/handler.c:1013	ipc: comp sink 5, source 1  -> connect
    0      2         PIPE          65671996.145833        46.562500 src/audio/pipeline.c:78  	pipeline: connect comp 1 and buffer 5
    0      2          IPC          65672727.708333       731.562500    src/ipc/handler.c:1013	ipc: comp sink 2, source 5  -> connect
    0      2         PIPE          65672775.260417        47.552082 src/audio/pipeline.c:78  	pipeline: connect comp 2 and buffer 5
    0      2          IPC          65673964.583333      1189.322876    src/ipc/handler.c:1013	ipc: comp sink 6, source 2  -> connect
    0      2         PIPE          65674013.697917        49.114582 src/audio/pipeline.c:78  	pipeline: connect comp 2 and buffer 6
    0      2          IPC          65674760.208333       746.510437    src/ipc/handler.c:1013	ipc: comp sink 4, source 6  -> connect
    0      2         PIPE          65674811.562500        51.354168 src/audio/pipeline.c:78  	pipeline: connect comp 4 and buffer 6 
````
</details>

<details>
 <summary>dmesg_error.log</summary>

````
[ 2916.028335] sof-audio-acpi INT3438:00: pcm: hw params stream 0 dir 0
[... cut ...]
[ 2916.528674] sof-audio-acpi INT3438:00: pcm: free stream 0 dir 0
[ 2916.528693] sof-audio-acpi INT3438:00: error: trace IO error
[ 2916.529547] sof-audio-acpi INT3438:00: pcm: close stream 0 dir 0
[ 2916.530909] aplay (2456) used greatest stack depth: 12344 bytes left
````
</details>

[dmesg0320.log](https://github.com/thesofproject/sof/files/4358864/dmesg0320.log)
[sof_errortrace.log](https://github.com/thesofproject/sof/files/4358865/sof_errortrace.log)
[sof_trace.log](https://github.com/thesofproject/sof/files/4358866/sof_trace.log)

tlauda, 2020-03-23:
@Liviali155 Have this ever worked?

lgirdwood, 2020-03-23:
@Liviali155 can you try the following diff
```diff
diff --git a/src/lib/agent.c b/src/lib/agent.c
index 6bdac2bf..9166b22f 100644
--- a/src/lib/agent.c
+++ b/src/lib/agent.c
@@ -47,8 +47,8 @@ static enum task_state validate(void *data)
        perf_cnt_stamp(TRACE_CLASS_SA, &sa->pcd, true);
 
        /* panic timeout */
-       if (delta > sa->panic_timeout)
-               panic(SOF_IPC_PANIC_IDLE);
+//     if (delta > sa->panic_timeout)
+//             panic(SOF_IPC_PANIC_IDLE);
 
        /* warning timeout */
        if (delta > sa->warn_timeout)
```

I'm suspicious we have a problem waiting on DMA channels flushing. This should at least give us more insights.

Liviali155, 2020-03-23:
> @Liviali155 Has this ever worked?

on latest sof/dev+matser still has this issue

lgirdwood, 2020-03-23:
@Liviali155 can I see the logs for the above diff. This will stop the idle panic, but it may provide more detail in dmesg, trace or error trace.

Liviali155, 2020-03-23:
> @Liviali155 can I see the logs for the above diff. This will stop the idle panic, but it may provide more detail in dmesg, trace or error trace.

How should I try

lgirdwood, 2020-03-23:
Open src/lib/agent.c (in a text editor) and comment out the lines in red above to look like the lines in green.

Liviali155, 2020-03-24:
> Open src/lib/agent.c (in a text editor) and comment out the lines in red above to look like the lines in green.

On latest sof/dev(9b5b6c2a)+master(d12ad78d), seems the Reproduction Rate is very low,
 I  run "./check-playback.sh -d 1 -l 100 -r 1" 6times.only reproduced once "error: ipc timed out for 0x60010000 size 108"

After I comment out the lines in red  line then build the firmware to run below command agagin,issue can not be reproduced
1."./check-playback.sh -d 1 -l 100 -r 1"  6 times
2../check-playback.sh -d 1 -l 1000 -r 1"
3.multiple-pipeline-playback.sh -l 50
4."./check-playback.sh -d 1 -l 100 -r 1"  2 times

tlauda, 2020-03-24:
@Liviali155 @lgirdwood I think it just randomly reproduced for my PR: https://sof-ci.01.org/sofpr/PR2593/build5499/devicetest/.

tlauda, 2020-03-24:
@lgirdwood Unfortunately I don't think the agent will work correctly for platforms, which are not scheduled based on timer. Its purpose is to detect drifts in timer interrupts, but with pipelines scheduled asynchronously on DMAs we can easily starve the agent. Taking into consideration that BYT/CHT and HSW/BDW platforms are not supposed to handle more complicated processing, maybe we should disable the agent on them. We already have option in kconfig: `CONFIG_HAVE_AGENT`.

lgirdwood, 2020-03-24:
@tlauda ack, we should have the kconfig option and disable for older platfioms.

Liviali155, 2020-03-25:
Tried with PR #2617,run 2000 times,error not seen
./check-playback.sh  -d 1 -l 2000 -r 1

PR #2617 (2020-03-25): byt: bdw: disable agent by default
Disables system agent for BYT/CHT and HSW/BDW platforms.
System agent is supposed to track drift between timer ticks,
but it's only valid for the platforms with timer based scheduling.
Platforms with DMA based scheduling have asynchronous interrupts,
which can potentially starve the agent and cause system panic.

Fixes #2585.
closed 2020-03-25 by tlauda