thesofproject/sof #2623: [BUG] Failed to use SOF on HP zbook  15 G6

elorriaux, 2020-03-24:
**Describe the bug**
dmesg reports:
[    4.122285] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    4.122812] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    4.123269] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    4.123798] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    4.124252] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    4.124851] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    4.125485] sof-audio-pci 0000:00:1f.3: error: init i915 and HDMI codec failed
[    4.126042] sof-audio-pci 0000:00:1f.3: error: failed to probe DSP -19
[    4.126496] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -19

**Impact**
Cannot use built-in mic on this laptop

**Environment**
1) sof (firmware/topology) and linux (kernel driver).
    * Kernel: 5.5.11 (gentoo-sources patchset)
    * SOF: 1.4.2
2) Name of the topology file
    * Topology: sof-cnl-v1.4.2.ri
3) Name of the platform(s) on which the bug is observed.
    * Platform: x86_64

**Screenshots or console output**
lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation 8th Gen Core Processor Host Bridge/DRAM Registers [8086:3ec4] (rev 07)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) [8086:1901] (rev 07)
00:04.0 Signal processing controller [1180]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem [8086:1903] (rev 07)
00:12.0 Signal processing controller [1180]: Intel Corporation Cannon Lake PCH Thermal Controller [8086:a379] (rev 10)
00:14.0 USB controller [0c03]: Intel Corporation Cannon Lake PCH USB 3.1 xHCI Host Controller [8086:a36d] (rev 10)
00:14.2 RAM memory [0500]: Intel Corporation Cannon Lake PCH Shared SRAM [8086:a36f] (rev 10)
00:15.0 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH Serial IO I2C Controller #0 [8086:a368] (rev 10)
00:15.1 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH Serial IO I2C Controller #1 [8086:a369] (rev 10)
00:16.0 Communication controller [0780]: Intel Corporation Cannon Lake PCH HECI Controller [8086:a360] (rev 10)
00:17.0 SATA controller [0106]: Intel Corporation Cannon Lake Mobile PCH SATA AHCI Controller [8086:a353] (rev 10)
00:1b.0 PCI bridge [0604]: Intel Corporation Cannon Lake PCH PCI Express Root Port #17 [8086:a340] (rev f0)
00:1c.0 PCI bridge [0604]: Intel Corporation Cannon Lake PCH PCI Express Root Port #3 [8086:a33a] (rev f0)
00:1c.4 PCI bridge [0604]: Intel Corporation Cannon Lake PCH PCI Express Root Port #5 [8086:a33c] (rev f0)
00:1d.0 PCI bridge [0604]: Intel Corporation Cannon Lake PCH PCI Express Root Port #9 [8086:a330] (rev f0)
00:1d.5 PCI bridge [0604]: Intel Corporation Cannon Lake PCH PCI Express Root Port #14 [8086:a335] (rev f0)
00:1f.0 ISA bridge [0601]: Intel Corporation Device [8086:a30e] (rev 10)
00:1f.3 Multimedia audio controller [0401]: Intel Corporation Cannon Lake PCH cAVS [8086:a348] (rev 10)
00:1f.4 SMBus [0c05]: Intel Corporation Cannon Lake PCH SMBus Controller [8086:a323] (rev 10)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH SPI Controller [8086:a324] (rev 10)
00:1f.6 Ethernet controller [0200]: Intel Corporation Ethernet Connection (7) I219-LM [8086:15bb] (rev 10)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation TU117GLM [Quadro T1000 Mobile] [10de:1fb9] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:10fa] (rev a1)
02:00.0 Non-Volatile memory controller [0108]: ADATA Technology Co., Ltd. XPG SX8200 Pro PCIe Gen3x4 M.2 2280 Solid State Drive [1cc1:8201] (rev 03)
03:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS525A PCI Express Card Reader [10ec:525a] (rev 01)
05:00.0 PCI bridge [0604]: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] [8086:15ea] (rev 06)
06:00.0 PCI bridge [0604]: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] [8086:15ea] (rev 06)
06:01.0 PCI bridge [0604]: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] [8086:15ea] (rev 06)
06:02.0 PCI bridge [0604]: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] [8086:15ea] (rev 06)
06:04.0 PCI bridge [0604]: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] [8086:15ea] (rev 06)
07:00.0 System peripheral [0880]: Intel Corporation JHL7540 Thunderbolt 3 NHI [Titan Ridge 4C 2018] [8086:15eb] (rev 06)
3b:00.0 USB controller [0c03]: Intel Corporation JHL7540 Thunderbolt 3 USB Controller [Titan Ridge 4C 2018] [8086:15ec] (rev 06)
70:00.0 Non-Volatile memory controller [0108]: Sandisk Corp WD Black 2018/PC SN720 NVMe SSD [15b7:5002]
71:00.0 Network controller [0280]: Intel Corporation Wi-Fi 6 AX200 [8086:2723] (rev 1a)

plbossart, 2020-03-25:
@elorriaux that looks like a known issue where an external NVIDIA GPU is used, the log is similar to https://github.com/thesofproject/linux/issues/1658

@kv2019i I think you fixed this with https://github.com/thesofproject/linux/pull/1731, but can it be back-ported to 5.5-stable?

elorriaux, 2020-03-25:
@plbossart thanks Pierre for the quick answer. Actually you're wright my IT disabled the intel graphics card, I did not pay attention to that. If I reactivate it I can go through that error but my X server does not work anymore, and my other distro does not boot anymore. So I will probably have to 

I was also wrong regarding the firmware, the next error is that sof-cfl.ri is missing.

kv2019i, 2020-03-25:
@plbossart Ack, this is fixed by upstream commit "ASoC: SOF: Intel: hda: allow operation without i915 gfx" . This is currently not queued even for 5.6, so only in 5.7 queue. Mark wasn't happy recently with the volume of changes to 5.5, so not sure if this is accepted. There is going to be a fair amount of patches, so maybe this is still worth it.

kv2019i, 2020-03-25:
@plbossart @elorriaux Hmm, backport to 5.5.y is not really straightforward. A minimum set requires many more patches than the one mentioned. I backported three patches, but then realized that the upstream fix depends on the refactored machine selection code (in 5.5.y, bus probe is done after machine driver selection, so the upstream logic to ignore lack of i915 won't work). Backport to 5.5.y within limits of stable kernel rules (all patches need to be in upstream, <100 lines), does not seem feasible.

plbossart, 2020-03-25:
@kv2019i I think we need to start thinking about a 5.5 SOF branch, it'd still be useful for a number of people who have non-functional devices. You can argue that 5.6 is just around the corner though but some distros will stick with 5.5. Some will even stick with 5.4 LTS.

elorriaux, 2020-04-02:
Hi, I tried to configure bumblebee to keep intel built-in grapĥics card enabled and it solves my initial error, I can go further but it fails to load the firmware (I symlinked sof-cfl.ri tp sof-cfl.ri):

dmesg | grep audio
[    4.130562] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    4.131228] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    4.131839] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    4.132512] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    4.134776] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    4.135454] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    4.136682] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops 0xffffffffa069a3c0)
[    4.147497] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    4.148610] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    4.151704] sof-audio-pci 0000:00:1f.3: Direct firmware load for intel/sof/sof-cfl.ri failed with error -2
[    4.152441] sof-audio-pci 0000:00:1f.3: error: request firmware intel/sof/sof-cfl.ri failed err: -2
[    4.153791] sof-audio-pci 0000:00:1f.3: error: failed to load DSP firmware -2
[    4.153953] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -2

ls /lib/firmware/intel/sof/sof-cfl.ri -alrt
lrwxrwxrwx 1 root root 10 Apr  2 20:33 /lib/firmware/intel/sof/sof-cfl.ri -> sof-cnl.ri

(firmware version 1.4.2)

plbossart, 2020-04-02:
@elorriaux you sure about the symlink? If the file was in the right location it'd be a different error...

elorriaux, 2020-04-02:
@plbossart That's the first thing I checked. You are confirming that I should continue to check in that direction.

I guess that read permission on cof-cnl.ri should be sufficient? 

I simply took sof-cnl-v1.4.2.ri from https://github.com/thesofproject/sof/releases and copied it to /lib/firmware/intel/sof/sof-cnl.ri and then created a sof-cfl.ri symlink. I'll try to copy it directly with the right name without symlink.

plbossart, 2020-04-02:
https://github.com/thesofproject/sof-bin/tree/stable-v1.4.2

elorriaux, 2020-04-02:
Reinstalled all the firmwares with the go.sh script, not exactly the same error msg:

dmesg | grep audio
[    4.114546] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    4.115214] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    4.115821] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    4.116463] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    4.117060] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    4.117753] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    4.118752] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops 0xffffffff90a9a3c0)
[    4.128847] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    4.129424] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    4.131533] sof-audio-pci 0000:00:1f.3: error: codec #0 probe error, ret: -2
[    4.132544] sof-audio-pci 0000:00:1f.3: Direct firmware load for intel/sof/sof-cfl.ri failed with error -2
[    4.133668] sof-audio-pci 0000:00:1f.3: error: request firmware intel/sof/sof-cfl.ri failed err: -2
[    4.134939] sof-audio-pci 0000:00:1f.3: error: failed to load DSP firmware -2
[    4.135094] sof-audio-pci 0000:00:1f.3: error: sof_probe_work failed err: -2

lgirdwood, 2020-04-03:
@elorriaux what was the script output ? You will need to run as root/sudo as it will be installing files under /lib/firmware (or wherever your distro places system firmware)

elorriaux, 2020-04-03:
everything was done with root account.

./go.sh
Installing Intel firmware and topology v1.4.2 to lib/firmware/intel
Done installing Intel firmware and topology v1.4.2

ls -alrt /lib/firmware/intel/sof
total 12
drwxr-xr-x 4 root root 4096 Apr  3 15:43 v1.4.2
lrwxrwxrwx 1 root root   24 Apr  3 15:43 sof-bdw.ri -> v1.4.2/sof-bdw-v1.4.2.ri
lrwxrwxrwx 1 root root   24 Apr  3 15:43 sof-cht.ri -> v1.4.2/sof-cht-v1.4.2.ri
lrwxrwxrwx 1 root root   24 Apr  3 15:43 sof-byt.ri -> v1.4.2/sof-byt-v1.4.2.ri
lrwxrwxrwx 1 root root   37 Apr  3 15:43 sof-apl.ri -> v1.4.2/intel-signed/sof-apl-v1.4.2.ri
lrwxrwxrwx 1 root root   37 Apr  3 15:43 sof-glk.ri -> v1.4.2/intel-signed/sof-apl-v1.4.2.ri
lrwxrwxrwx 1 root root   37 Apr  3 15:43 sof-cnl.ri -> v1.4.2/intel-signed/sof-cnl-v1.4.2.ri
lrwxrwxrwx 1 root root   37 Apr  3 15:43 sof-cfl.ri -> v1.4.2/intel-signed/sof-cnl-v1.4.2.ri
lrwxrwxrwx 1 root root   37 Apr  3 15:43 sof-cml.ri -> v1.4.2/intel-signed/sof-cnl-v1.4.2.ri
lrwxrwxrwx 1 root root   37 Apr  3 15:43 sof-icl.ri -> v1.4.2/intel-signed/sof-icl-v1.4.2.ri

If you pay attention on dmesg  logs, the usage of the script to install firmware causes one additional line compared to the firmware install I did manually:

[ 4.131533] sof-audio-pci 0000:00:1f.3: error: codec #0 probe error, ret: -2

elorriaux, 2020-04-03:
Forgot to add a listing to the directory where the link is pointing at:

ls -alrt /lib/firmware/intel/sof/v1.4.2/intel-signed/
total 844
-rw-r--r-- 1 root root 286720 Apr  3 15:43 sof-cnl-v1.4.2.ri
-rw-r--r-- 1 root root 282624 Apr  3 15:43 sof-apl-v1.4.2.ri
drwxr-xr-x 2 root root   4096 Apr  3 15:43 .
-rw-r--r-- 1 root root 286720 Apr  3 15:43 sof-icl-v1.4.2.ri
drwxr-xr-x 4 root root   4096 Apr  3 15:43 ..

paulstelian97, 2020-04-03:
4 seconds, which means early boot... Do you have an initramfs? It's possible you may need to add the /lib/firmware/intel/sof and /lib/firmware/intel/sof-tplg folders inside the initramfs.

root ignores read/write permissions and so should the kernel module itself.

plbossart, 2020-04-03:
maybe a red-herring but do you compile all SOF stuff as modules? if you use built-ins the request_firmware won't work.

Also the error you have
[ 4.131533] sof-audio-pci 0000:00:1f.3: error: codec #0 probe error, ret: -2

shows a problem with HDMI, so are you sure you are using the intel card?

kv2019i, 2020-04-03:
Ouch, you @elorriaux seem to be hitting many issues. Thanks for trying and reporting back. You could try reloading "modprobe -r snd_sof_pci ; modprobe snd_sof_pci" after the system has booted. If the issue is with firmware files missing from initramfs, that should work around that. 

@plbossart wrote:
> Also the error you have
> [ 4.131533] sof-audio-pci 0000:00:1f.3: error: codec #0 probe error, ret: -2
> 
> shows a problem with HDMI, so are you sure you are using the intel card?

#0 is still the external HDA codec. I'm afraid this is a case of https://github.com/thesofproject/linux/issues/1807 I've been waiting for the reporter of that one to try my patch:
https://github.com/thesofproject/linux/issues/1807#issuecomment-589231066

Or both symptoms are explained by what @plbossart described, e.g. not building SOF as module. That could explain the errors.

elorriaux, 2020-04-03:
@paulstelian97 good remark, I don't use initramfs but I can include the firmware when compiling the kernel

@plbossart No I was not using modules, I'll search in that direction. And yes I was using the intel card, when intel card is disabled in the BIOS, I'm getting the error described at the beginning of the bug:

[ 4.125485] sof-audio-pci 0000:00:1f.3: error: init i915 and HDMI codec failed

as soon as I enable hybrid graphics and add i915.enable_guc=2 to the kernel line I can go through that.

plbossart, 2020-04-03:
@elorriaux To remove any timing or resource dependency, can you first blacklist snd-sof-pci, then after boot do a manual modprobe snd-sof-pci.

plbossart, 2020-04-03:
Another trick @elorriaux is to add the attached file as /etc/modprobe.d/sof-dyndbg.conf, that will generate a longer dmesg trace and help us figure out what's going on. don't copy/paste the results, attach a file or provide a link, thanks!
[ddebug.txt](https://github.com/thesofproject/sof/files/4430080/ddebug.txt)

elorriaux, 2020-04-04:
So, using modules instead of built in kernel solved the question, no need to blacklist the module. The firmware seems to be successfully loaded, there remains one error, I don't know if it's critical.

[    6.110301] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    6.110426] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    6.110436] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    6.110509] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    6.110512] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    6.110630] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    6.110708] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops 0xffffffff84c9ce80)
[    6.119405] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    6.119406] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    6.131233] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    6.131235] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    6.261736] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
[    6.261737] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[    6.261991] sof-audio-pci 0000:00:1f.3: firmware boot complete

At the end I have no sound and no mic, I've upgraded pulseaudio to version 13.99.1, alsa 1.2.2. From now I don't know in which direction I should search.

@plbossart: here is the dmesg log with dynamic debug (had to add CONFIG_DYNAMIC_DEBUG) in my kernel. 
[dmesg_dyndbg.log](https://github.com/thesofproject/sof/files/4431524/dmesg_dyndbg.log)

Other strange thing, I used to build snd-hda-intel in the kernel to get sound, now I'm trying to load it as a module but I can't get sound after the module is loaded, and the dmesg states that snd-hda-intel is trying to use SOF for DMIC...

[ 2027.208939] snd_hda_intel 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[ 2027.209181] snd_hda_intel 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[ 2027.209291] snd_hda_intel 0000:01:00.1: enabling device (0000 -> 0002)
[ 2027.209354] snd_hda_intel 0000:01:00.1: Disabling MSI

plbossart, 2020-04-04:
@elorriaux that's progress you now boot successfully and detect HDAudio cards ([ 6.119405] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
)

Can you try and do aplay -l and arecord -l, you should now see playback and capture devices. and should be able to play/capture using the -Dhw:0,x devices (replace x by what you see).

Let's take pulseaudio separately.

plbossart, 2020-04-04:
ah and also don't use snd-hda-intel, it cannot be used for capture. You need SOF to deal with the digital mics.

elorriaux, 2020-04-05:
@plbossart for snd-hda-intel I was just trying to get sound while sof is not operational.

When loading snd-sof-pci, I don't have any interface:

~ # aplay -l
aplay: device_list:274: no soundcards found...
~ # arecord -l
arecord: device_list:274: no soundcards found...

plbossart, 2020-04-05:
@elorriaux I don't know what to tell you except: please retry and check that you did remove blacklists or did a manual modprobe snd-sof-pci

kv2019i, 2020-04-06:
@elorriaux wrote:
> ~ # aplay -l
> aplay: device_list:274: no soundcards found...
> ~ # arecord -l
> arecord: device_list:274: no soundcards found...

Did you run these as root (seems so but just checking)? On most distros, only members of "audio" group can access the ALSA device nodes so "aplay -l" as a normal user will not report any soundcards. But as root it should work.

The log in https://github.com/thesofproject/sof/issues/2623#issuecomment-609033760 seems good, no errors anymore, firmware is loaded and probe works for the all codecs.

When you load snd-sof-pci, could you still send the output of "alsa-info.sh". There's still something we are missing.

elorriaux, 2020-04-06:
@plbossart I did so many attempts ;) I've even tried to upgrade to a 5.6.2 kernel without more success. Loading snd-sof-pci when booting or after booting does not change anything. My blacklist is empty (my system does not load snd-hda-intel automatically)

@kv2019i yes I ran scripts as root and my users ares in audio group

alsa-info: http://alsa-project.org/db/?f=b47e6e3b699a29d2769da51f8cd0acfa65807468

Is there a live distro with recent enough kernel and which is known to work for such a hardware?

kv2019i, 2020-04-06:
@elorriaux wrote:
> Other strange thing, I used to build snd-hda-intel in the kernel to get sound, now I'm trying to load it as a module but I can't get sound after the module is loaded, and the dmesg states that snd-hda-intel is trying to use SOF for DMIC...

With newer kernel's there is a mechanism to decide automatically which driver to choose. You can force a specific choice by putting to e.g. /etc/modprobe.d/blacklist-dsp.conf

```
#Force the DSP driver for Intel DSP (0=auto, 1=legacy, 2=SST, 3=SOF
options snd\_intel\_dspcfg dsp\_driver=1
```

kv2019i, 2020-04-06:
@elorriaux wrote:
> @kv2019i yes I ran scripts as root and my users ares in audio group
> 
> alsa-info: http://alsa-project.org/db/?f=b47e6e3b699a29d2769da51f8cd0acfa65807468

Ack, this confirms the driver load did not complete. You have no PCM devices:

```
!!ALSA Device nodes
!!-----------------

crw-rw---- 1 root audio 116,  1 Apr  5 15:35 /dev/snd/seq
crw-rw---- 1 root audio 116, 33 Apr  5 15:35 /dev/snd/timer 
```

In your dynamic-dbg log, it would seem init of snd_soc_skl_hda_dsp machine driver does not happen at all. It also seems the module is not loaded based on your alsa-info output.

A few things to try:
- can you confirm CONFIG_SND_SOC_INTEL_SKL_HDA_DSP_GENERIC_MACH is enabled in your kernel config
- you can try loading the driver manually: sudo modprobe snd_soc_skl_hda_dsp
- double-check this module is not blacklisted in any file in /etc/modprobe.d

> Is there a live distro with recent enough kernel and which is known to work for such a hardware?

I'm not aware of other HP zbook15g6 users that have tried SOF, so there are no verified to work cases. You could try Fedora Rawhide Live, but it's a bit of a stretch as there is nothing wrong in your current config that would explain the problems.

elorriaux, 2020-04-06:
@kv2019i : you are actually pointing at something I've investigated yesterday after looking at that thread [https://github.com/thesofproject/sof/issues/2275](url) 

So yes snd-soc-skl_hda_dsp  is enabled and compiled as module. My blacklist.conf is empty, and if I don't load modules by myself, nothing is loaded for audio devices (snd-hda-intel). Here is a dmesg log, where I modprobe snd-sof-pci after X login, and then modprobe snd-soc-skl_hda_dsp.

[dmesg_modprobe.log](https://github.com/thesofproject/sof/files/4440214/dmesg_modprobe.log)

And alsa-info: http://alsa-project.org/db/?f=45578fbc786fd50fc89363ac494a38c27be525ac

Since installing gentoo is just like starting from a white page, I would not be surprised that I forgot something... I forgot to mention that I've also tried to generate a kernel with the genkernel tool which results in a generic kernel conf as the one which is used in gentoo live dvd.

I'll try to have a look to Fedora live... Since I really don't see where to look.

elorriaux, 2020-04-06:
Log with dynamic debug, just in case... 
[dmesg_modprobe_dyndbg.log](https://github.com/thesofproject/sof/files/4440352/dmesg_modprobe_dyndbg.log)

plbossart, 2020-04-06:
@elorriaux can you check that you have the machine driver compiled in?
CONFIG_SND_SOC_INTEL_SKL_HDA_DSP_GENERIC_MACH=m

grep HDA sof-dev-defconfig
CONFIG_SND_HDA=m
CONFIG_SND_HDA_INTEL=m
CONFIG_SND_HDA_HWDEP=y
CONFIG_SND_HDA_RECONFIG=y
CONFIG_SND_HDA_INPUT_BEEP=y
CONFIG_SND_HDA_INPUT_BEEP_MODE=1
CONFIG_SND_HDA_PATCH_LOADER=y
CONFIG_SND_HDA_CODEC_REALTEK=m
CONFIG_SND_HDA_CODEC_ANALOG=m
CONFIG_SND_HDA_CODEC_SIGMATEL=m
CONFIG_SND_HDA_CODEC_VIA=m
CONFIG_SND_HDA_CODEC_HDMI=m
CONFIG_SND_HDA_CODEC_CIRRUS=m
CONFIG_SND_HDA_CODEC_CONEXANT=m
CONFIG_SND_HDA_CODEC_CA0110=m
CONFIG_SND_HDA_CODEC_CA0132=m
# CONFIG_SND_HDA_CODEC_CA0132_DSP is not set
CONFIG_SND_HDA_CODEC_CMEDIA=m
CONFIG_SND_HDA_CODEC_SI3054=m
CONFIG_SND_HDA_GENERIC=m
CONFIG_SND_HDA_POWER_SAVE_DEFAULT=0
CONFIG_SND_HDA_CORE=m
CONFIG_SND_HDA_COMPONENT=y
CONFIG_SND_HDA_I915=y
CONFIG_SND_HDA_EXT_CORE=m
CONFIG_SND_HDA_PREALLOC_SIZE=0
CONFIG_SND_SOC_INTEL_SKL_HDA_DSP_GENERIC_MACH=m
CONFIG_SND_SOC_SOF_HDA_COMMON=m
CONFIG_SND_SOC_SOF_HDA_LINK=y
CONFIG_SND_SOC_SOF_HDA_AUDIO_CODEC=y
CONFIG_SND_SOC_SOF_HDA_PROBES=y
CONFIG_SND_SOC_SOF_HDA_LINK_BASELINE=m
CONFIG_SND_SOC_SOF_HDA=m
CONFIG_SND_SOC_HDAC_HDMI=m
CONFIG_SND_SOC_HDAC_HDA=m

elorriaux, 2020-04-06:
@plbossart 

```
grep HDA .config
CONFIG_SND_HDA=m
CONFIG_SND_HDA_INTEL=m
CONFIG_SND_HDA_HWDEP=y
# CONFIG_SND_HDA_RECONFIG is not set
# CONFIG_SND_HDA_INPUT_BEEP is not set
# CONFIG_SND_HDA_PATCH_LOADER is not set
CONFIG_SND_HDA_CODEC_REALTEK=m
CONFIG_SND_HDA_CODEC_ANALOG=m
CONFIG_SND_HDA_CODEC_SIGMATEL=m
CONFIG_SND_HDA_CODEC_VIA=m
CONFIG_SND_HDA_CODEC_HDMI=m
CONFIG_SND_HDA_CODEC_CIRRUS=m
CONFIG_SND_HDA_CODEC_CONEXANT=m
CONFIG_SND_HDA_CODEC_CA0110=m
CONFIG_SND_HDA_CODEC_CA0132=m
# CONFIG_SND_HDA_CODEC_CA0132_DSP is not set
CONFIG_SND_HDA_CODEC_CMEDIA=m
CONFIG_SND_HDA_CODEC_SI3054=m
CONFIG_SND_HDA_GENERIC=m
CONFIG_SND_HDA_POWER_SAVE_DEFAULT=0
CONFIG_SND_HDA_CORE=m
CONFIG_SND_HDA_DSP_LOADER=y
CONFIG_SND_HDA_COMPONENT=y
CONFIG_SND_HDA_I915=y
CONFIG_SND_HDA_EXT_CORE=m
CONFIG_SND_HDA_PREALLOC_SIZE=64
CONFIG_SND_SOC_INTEL_SKYLAKE_HDAUDIO_CODEC=y
CONFIG_SND_SOC_INTEL_SKL_HDA_DSP_GENERIC_MACH=m
CONFIG_SND_SOC_SOF_HDA_COMMON=m
CONFIG_SND_SOC_SOF_HDA_LINK=y
CONFIG_SND_SOC_SOF_HDA_AUDIO_CODEC=y
CONFIG_SND_SOC_SOF_HDA_ALWAYS_ENABLE_DMI_L1=y
CONFIG_SND_SOC_SOF_HDA_COMMON_HDMI_CODEC=y
CONFIG_SND_SOC_SOF_HDA_LINK_BASELINE=m
CONFIG_SND_SOC_SOF_HDA=m
CONFIG_SND_SOC_HDAC_HDMI=m
CONFIG_SND_SOC_HDAC_HDA=m
# CONFIG_SENSORS_HDAPS is not set
```

plbossart, 2020-04-06:
don't do this: CONFIG_SND_SOC_SOF_HDA_ALWAYS_ENABLE_DMI_L1=y

plbossart, 2020-04-06:
Also add this to a file in modprobe.d:

options snd_soc_skl_hda_dsp dyndbg=+p

that would enable dynamic debug and hopefully trigger some logs in this driver. If you you may have to add traces in there to figure out what's going one.

plbossart, 2020-04-06:
ok, so there's a trace that shows you are going through the machine driver probe.
[   43.131740] skl_hda_dsp_generic skl_hda_dsp_generic: skl_hda_audio_probe: entry

I have 2 hypotheses here
a) I remember seeing cases where you cannot see the card unless you log in with the UI.
b) something goes wrong in the card registration, but we don't trace it.  

Maybe you can apply the diff below to see what's going on?

````diff
diff --git a/sound/soc/intel/boards/skl_hda_dsp_generic.c b/sound/soc/intel/boards/skl_hda_dsp_generic.c
index 3be764299ab0..aaca55885d27 100644
--- a/sound/soc/intel/boards/skl_hda_dsp_generic.c
+++ b/sound/soc/intel/boards/skl_hda_dsp_generic.c
@@ -206,7 +206,9 @@ static int skl_hda_audio_probe(struct platform_device *pdev)
                hda_soc_card.components = hda_soc_components;
        }
 
-       return devm_snd_soc_register_card(&pdev->dev, &hda_soc_card);
+       ret = devm_snd_soc_register_card(&pdev->dev, &hda_soc_card);
+       dev_info(&pdev->dev, "Completed probe with status %d\n", ret);
+       return ret;
 }
 
 static struct platform_driver skl_hda_audio = {
````

ranj063, 2020-04-06:
@plbossart I have a feeling this might be the case of missing mach_params->platform. The following diff will confirm it too:
```
diff --git a/sound/soc/intel/boards/skl_hda_dsp_generic.c b/sound/soc/intel/boards/skl_hda_dsp_generic.c
index 3be764299ab0..e3a8e5694cba 100644
--- a/sound/soc/intel/boards/skl_hda_dsp_generic.c
+++ b/sound/soc/intel/boards/skl_hda_dsp_generic.c
@@ -195,6 +195,7 @@ static int skl_hda_audio_probe(struct platform_device *pdev)
        ctx->pcm_count = hda_soc_card.num_links;
        ctx->dai_index = 1; /* hdmi codec dai name starts from index 1 */
        ctx->platform_name = mach->mach_params.platform;
+       dev_dbg(&pdev->dev, "platform name %s\n", ctx->platform_name);
        ctx->common_hdmi_codec_drv = mach->mach_params.common_hdmi_codec_drv;
 
        hda_soc_card.dev = &pdev->dev;
```

elorriaux, 2020-04-06:
@plbossart here it is

[dmesg_modprobe_dyndbg_hda_dsp_return.log](https://github.com/thesofproject/sof/files/4440849/dmesg_modprobe_dyndbg_hda_dsp_return.log)

plbossart, 2020-04-06:
@elorriaux something's not right then, -517 is EPROBE_DEFER. So you are trying to use a resource that's not available. It'd guess it's an HDMI issue.

@kv2019i can we remove HDMI support from this card?