thesofproject/sof #2628: [BUG][UP2][PCM512X]Headset arecord can't capture voice

dengyangchao, 2020-03-25:
**Describe the bug**
PCM512x arecord can't capture voice.
```
ubuntu@cd-apl-up2-pcm512x-07:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofpcm512x [sof-pcm512x], device 0: Port5 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 5: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofpcm512x [sof-pcm512x], device 6: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

**To Reproduce**
1."Sudo reboot" with headset connected.
2.cd sof-test
3.cd test-case
4. export TPLG=/lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg
5. ./check-capture.sh -d 100 -l 1 -r 1(https://github.com/thesofproject/sof-test/blob/master/test-case/check-capture.sh)

**Reproduction Rate**
100%

**Expected behavior**
Headset can capture voice and the output normally

**Impact**
can't capture the voice

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev 33f864de
    * SOF: master  b8b96a20
2) Name of the topology file
    * Topology: sof-apl-pcm512x.tplg
3) Name of the platform(s) on which the bug is observed.
    * Platform: APL-UP2-PCM512X

[dmesg.txt](https://github.com/thesofproject/sof/files/4379899/dmesg.txt)

tlauda, 2020-03-25:
@dengyangchao There are no FW logs. Are you sure it's not kernel/alsamixer/UCM problem?

dengyangchao, 2020-03-25:
> @dengyangchao There are no FW logs. Are you sure it's not kernel/alsamixer/UCM problem?

@tlauda i have updated a sof log, i'm not sure it is a sof issue or kernel issue, so i open a bug here to track, if it is a kernel issue ,i'll transfer to kernel

ClarexZhou, 2020-03-26:
The PCM512x we used not support Port5 capture. Close.

closed 2020-03-26 by ClarexZhou