thesofproject/sof #2631: [FEATURE] Scripts with arguments should use getopt

jajanusz, 2020-03-25:
**Is your feature request related to a problem? Please describe.**
Manual parsing of arguments makes code harder to read can cause issues like one that is reported here: https://github.com/thesofproject/sof/pull/2625

**Describe the solution you'd like**
Use getopt where applicable

crazoes, 2020-03-26:
@jajanusz @dbaluta I am working on this one.

jajanusz, 2020-03-26:
@crazoes Thanks, I've updated description to use `getopts` instead of `getopt`, I think it's better cos it's in posix standard while `getopt` is not. Are you ok with that?

jajanusz, 2020-03-26:
Also there is one script that uses it already: https://github.com/thesofproject/sof/blob/9d9620ae05ca351d53f71f6d0a3a7587cc650e23/scripts/scan/clang-scan-build-xtensa.sh#L37

closed 2025-12-17 by kv2019i