thesofproject/sof #2633: [BUG][CML-H]" error: ipc error for 0x60010000 size 20" display when aplay via speaker0,2.

sinahuang, 2020-03-26:
**Describe the bug**
Speaker0,2 can't work normally, dmesg show " error: ipc error for 0x60010000 size 20".

**To Reproduce**
1.Boot up system.
2.aplay -Dhw:0,2 -fs16_le -c2 -r 48000 -vv -i wav/48California_Dreaming.wav

**Reproduction Rate**
5/5

**Expected behavior**
Aplay0,2  should work normally.

**Impact**
" error: ipc error for 0x60010000 size 20" display when aplay via speaker0,2.
```
[  161.012879] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  161.012898] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  161.012901] sof-audio-pci 0000:00:1f.3: period_bytes:0x3fc0
[  161.012901] sof-audio-pci 0000:00:1f.3: periods:4
[  161.012918] sof-audio-pci 0000:00:1f.3: stream_tag 1
[  161.012922] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  161.013418] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  161.013420] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 1
[  161.013422] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[  161.013423]  SDW1-speakers: ASoC: hw_params FE failed -22
[  161.013425] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[  161.013427] intel-master sdw-master-1: sdw_deprepare_stream: failed -22
[  161.013427] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[  161.013428] intel-master sdw-master-2: sdw_deprepare_stream: failed -22
[  161.013431] sof-audio-pci 0000:00:1f.3: pcm: free stream 2 dir 0
[  161.013439] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[  161.013440] intel-master sdw-master-1: sdw_deprepare_stream: failed -22
[  161.013441] sdw_deprepare_stream: subdevice #0-Playback: inconsistent state state 1
[  161.013442] intel-master sdw-master-2: sdw_deprepare_stream: failed -22
[  161.013589] sof-audio-pci 0000:00:1f.3: pcm: close stream 2 dir 0
[  161.483755] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  161.483790] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  161.983580] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  161.983628] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG

```
```
ubuntu@cd-CML-H-rt711-rt1308-rt715:~$ aplay -Dhw:0,2 -r48000 -c2 -fs16_le wav/48K_Let_It_Go.wav -vv -i
Playing WAVE 'wav/48K_Let_It_Go.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 85000
PERIOD_SIZE: 4080
PERIOD_BYTES: 16320
PERIODS: (4 5)
BUFFER_TIME: 341000
BUFFER_SIZE: 16368
BUFFER_BYTES: 65472
TICK_TIME: 0
```

**Test recipe**

Kernel:https://github.com/thesofproject/linux/commits/topic/sof-dev  commit:a27277
FW: https://github.com/thesofproject/sof/commits/master commit:  ae53aeb
Tplg: Same with FW branch,  sof-cml-rt711-rt1308-rt715.tplg
Platform:CML-H Laptop with codec ALC711, 2 x RT1308, RT715 in SDW mode.

[dmesg26.log](https://github.com/thesofproject/linux/files/4384877/dmesg26.log)

sinahuang, 2020-03-26:
Test with sof-dev ( commit:a27277) and master (commit:  ae53aeb)+PR#2540 +pr#2539, issue can be fixed.

mengdonglin, 2020-03-31:
PR #2540 is closed and replaced by https://github.com/thesofproject/sof/pull/2594. PR #2539 is closed and to be replaced by new PR by @slawblauciak

slawblauciak, 2020-04-06:
Fix awaiting merge #2666

sinahuang, 2020-04-09:
Issue can't be reproduced with latest FW master (commit:939c14 ) and kernel side + PR https://github.com/thesofproject/linux/pull/1982 and https://github.com/thesofproject/linux/pull/1943 after speaker sync enabled.

tlauda, 2020-04-21:
Kernel side PR merged.

closed 2020-04-21 by tlauda