thesofproject/sof #2635: [BUG] Failed to detect Audio Card Samsung Book Flex 2020.

ArtOfSettling, 2020-03-26:
Default install of Ubuntu 18.04 LTS, required driver sof-icl.ri was not installed by default, after manually installing latest version (1.4.2), Dummy Audio is still detected and system fails to find Audio Card.

Expected behavior
Audio Card should be detected and functional after installing SOF driver.

Impact
No Audio / Mic

Environment
Kernel : 5.3.0-42-generic

```
dmesg output : 

[    4.256683] integrity: Loaded X.509 cert 'Microsoft Windows Production PCA 2011: a92902398e16c49778cd90f99e4f9ae17c55af53'
[    4.256695] integrity: Loaded X.509 cert 'Microsoft Corporation UEFI CA 2011: 13adbf4309bd82709c8cd54f316ed522988a1bd4'
[   10.616371] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[   10.616378] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[   10.616513] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[   10.878484] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[   12.168187] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[   12.168189] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[   12.178976] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[   12.178977] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   12.252372] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
[   12.252375] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:8:0
[   12.252375] sof-audio-pci 0000:00:1f.3: warn: FW ABI is more recent than kernel
[   12.252669] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   12.254957] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:7:0 Kernel ABI 3:8:0
[   12.254962] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[   12.255823] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[   12.255825] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[   12.256676] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[   12.256677] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[   12.257493] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[   12.257495] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[   12.258344] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[   12.258346] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[   12.258347] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[   12.258348] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[   12.258349] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[   12.258351] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[   12.259205] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[   12.259206] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[   12.260083] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[   12.260084] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[   12.260970] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[   12.268217] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[   12.331790] Modules linked in: snd_soc_skl_hda_dsp(+) snd_soc_hdac_hdmi snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio fjes(-) nls_iso8859_1 intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass snd_soc_dmic mei_hdcp sof_pci_dev snd_sof_intel_hda_common snd_sof_intel_hda snd_sof_intel_byt snd_sof_intel_ipc snd_sof snd_sof_xtensa_dsp snd_soc_hdac_hda snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi crct10dif_pclmul snd_soc_core crc32_pclmul snd_compress ac97_bus snd_pcm_dmaengine ghash_clmulni_intel snd_hda_intel snd_intel_nhlt snd_hda_codec snd_hda_core snd_hwdep snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi aesni_intel aes_x86_64 crypto_simd snd_seq cryptd glue_helper nouveau(+) intel_cstate intel_rapl_perf iwlmvm mac80211 snd_seq_device input_leds snd_timer i915 libarc4 serio_raw wmi_bmof idma64 virt_dma snd mxm_wmi btusb ttm btrtl btbcm btintel soundcore drm_kms_helper bluetooth uvcvideo

```

ArtOfSettling, 2020-03-26:
Ok, so I managed to get somewhere with this myself. It seems like I was somehow missing the required firmware. The file 'sof-hda-generic-2ch.tplg' was missing from my system. I downloaded it manually and got  bit farther. My system now detects some cards and Audio Devices show up correctly on my system. However there is no audio.

My dmesg...

```
[    5.709672] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    5.806823] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    5.806838] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    5.809083] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    5.809134] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    5.809321] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    5.989487] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    8.000882] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    8.000885] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    8.012842] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    8.012844] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    8.102162] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:1:0-fcf6c
[    8.102164] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[    8.102411] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    8.104701] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:11:0
[    8.104705] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    8.105478] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    8.105479] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    8.106253] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    8.106255] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    8.107012] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    8.107014] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    8.107808] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    8.107810] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    8.107812] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    8.107813] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    8.107815] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    8.107816] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    8.108592] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    8.108594] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    8.109394] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    8.109396] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    8.110206] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    8.117784] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    8.183961] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[    8.219541] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
[    8.219581] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input20
[    8.219614] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input21
[    8.219644] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input22
[   39.097357] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   39.181898] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   60.178325] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   60.268823] sof-audio-pci 0000:00:1f.3: firmware boot complete
```

ArtOfSettling, 2020-03-26:
Some additional logs whilst playing back audio.

```
[   48.789521] sof-audio-pci 0000:00:1f.3: error: reply expected 12 got 120 bytes
[   48.789849] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x80010000 size 12
[   48.789851] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[   49.307811] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60050000 size 12
[   49.307987] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   49.313862] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   49.314146] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000080 rirb 00
[   49.314147] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   49.314419] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   49.819970] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[   49.820239] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   49.825650] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   49.826198] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[   49.826199] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   49.826218] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   50.327793] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[   50.327955] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   50.333852] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   50.334134] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[   50.334135] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   50.334408] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   50.334410] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[   53.335849] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x40010000 size 76
[   53.335867] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   53.336191] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   53.336208] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[   53.336210] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[   53.336220] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[   53.336224] sof-audio-pci 0000:00:1f.3: ctx_save ipc error -110, proceeding with suspend
[   56.821190] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   72.698453] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[   72.698729] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   72.704873] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   72.705419] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x80000000 intlctl 0x40000001 rirb 00
[   72.705422] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x80000000 adspis 0x00000001
[   72.705697] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[   73.206160] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x60030000 size 12
[   73.206377] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   73.213310] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   73.213593] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x80000000 intlctl 0x40000001 rirb 00
[   73.213594] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x80000000 adspis 0x00000001
[   73.213867] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[   73.721789] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[   73.721865] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[   73.728890] sof-audio-pci 0000:00:1f.3: error: unexpected fault 0x00000000 trace 0x00004000
[   73.729437] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x80000000 intlctl 0x40000001 rirb 00
[   73.729438] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x80000000 adspis 0x00000001
[   73.729720] sof-audio-pci 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
[   73.729723] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture

```

lgirdwood, 2020-03-26:
@ArtOfSettling it looks liek the FW you have is older, can you try the latest stable v1.4.2
https://github.com/thesofproject/sof-bin
It's in the stable-v1.4.2 branch.

ArtOfSettling, 2020-03-26:
Yes, no worries.

Boot dmesg after the uograde : 

```[    6.450431] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    6.450434] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    6.525996] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
[    6.525998] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[    6.526314] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    6.674096] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:11:0
[    6.674101] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    6.675266] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    6.675268] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    6.676083] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    6.676084] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    6.676850] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    6.676851] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    6.677777] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    6.677779] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    6.677781] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    6.677782] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    6.677784] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    6.677789] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    6.678635] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    6.678636] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    6.679497] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    6.679499] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    6.680390] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    6.688282] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    6.755219] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[    6.755259] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
[    6.755293] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input20
[    6.755343] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input21
[    6.755380] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input22
[    9.073622] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    9.164056] sof-audio-pci 0000:00:1f.3: firmware boot complete
```

When playing audio the following repeats

```
[    9.073622] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    9.164056] sof-audio-pci 0000:00:1f.3: firmware boot complete
[   94.521358] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   94.612276] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  131.141005] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  131.230923] sof-audio-pci 0000:00:1f.3: firmware boot complete
[  286.988064] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[  287.076174] sof-audio-pci 0000:00:1f.3: firmware boot complete

```

lgirdwood, 2020-03-26:
@juimonen is @ArtOfSettling missing UCM/Pualseaudio patches ?

ArtOfSettling, 2020-03-26:
Definitely not, actually, I have not patched either. Do you have the patches to hand (or know where I can find them)?

-------- Original Message --------

plbossart, 2020-03-26:
For Ubuntu 18.04 LTS, see see https://github.com/thesofproject/linux/issues/1877#issuecomment-596600017

ArtOfSettling, 2020-03-28:
Ran this with and without blacklisting.

dmesg shows

```
[    5.772938] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if info 0x040100
[    5.847441] sof-audio-pci 0000:00:1f.3: Digital mics found on Skylake+ platform, using SOF driver
[    5.847453] sof-audio-pci 0000:00:1f.3: enabling device (0000 -> 0002)
[    5.852318] sof-audio-pci 0000:00:1f.3: warning: No matching ASoC machine driver found
[    5.852567] sof-audio-pci 0000:00:1f.3: DSP detected with PCI class/subclass/prog-if 0x040100
[    5.852740] sof-audio-pci 0000:00:1f.3: use msi interrupt mode
[    5.885915] sof-audio-pci 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    5.894640] sof-audio-pci 0000:00:1f.3: hda codecs found, mask 5
[    5.894642] sof-audio-pci 0000:00:1f.3: using HDA machine driver skl_hda_dsp_generic now
[    5.915263] sof-audio-pci 0000:00:1f.3: unexpected ipc interrupt raised!
[    5.915265] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[    5.991372] sof-audio-pci 0000:00:1f.3: Firmware info: version 1:4:2-66a5a
[    5.991373] sof-audio-pci 0000:00:1f.3: Firmware: ABI 3:11:0 Kernel ABI 3:11:0
[    5.991634] sof-audio-pci 0000:00:1f.3: firmware boot complete
[    6.058900] sof-audio-pci 0000:00:1f.3: Topology: ABI 3:11:0 Kernel ABI 3:11:0
[    6.058903] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp3 Tx not handled
[    6.059722] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec0_in not handled
[    6.059724] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp2 Tx not handled
[    6.061210] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec1_in not handled
[    6.061213] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name iDisp1 Tx not handled
[    6.062049] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec0_out not handled
[    6.062051] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Analog CPU Playback not handled
[    6.062861] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec1_out not handled
[    6.062863] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Digital CPU Playback not handled
[    6.062864] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name codec2_in not handled
[    6.062867] sof-audio-pci 0000:00:1f.3: warning: widget type 7 name Alt Analog CPU Playback not handled
[    6.062868] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name codec2_out not handled
[    6.062869] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Analog CPU Capture not handled
[    6.063672] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp1_out not handled
[    6.063674] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Digital CPU Capture not handled
[    6.064662] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp2_out not handled
[    6.064664] sof-audio-pci 0000:00:1f.3: warning: widget type 0 name Alt Analog CPU Capture not handled
[    6.065541] sof-audio-pci 0000:00:1f.3: warning: widget type 1 name iDisp3_out not handled
[    6.073101] sof-audio-pci 0000:00:1f.3: ASoC: Parent card not yet available, widget card binding deferred
[    6.139657] input: sof-hda-dsp Mic as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input18
[    6.175792] input: sof-hda-dsp Headphone as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input19
[    6.175826] input: sof-hda-dsp HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input20
[    6.175857] input: sof-hda-dsp HDMI/DP,pcm=4 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input21
[    6.175887] input: sof-hda-dsp HDMI/DP,pcm=5 as /devices/pci0000:00/0000:00:1f.3/skl_hda_dsp_generic/sound/card0/input22
[   37.004403] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[   37.090006] sof-audio-pci 0000:00:1f.3: firmware boot complete
```

Seems like it actually detects and does some work with the microphone, still no sound unfortunately

plbossart, 2020-03-28:
@ArtOfSettling can you share the results of "arecord -l" 
Or better yet just the link to the results of alsa-info (or alsa-info.sh, can't recall how it's called)

ArtOfSettling, 2020-03-28:
alsa-info as requested :)

http://alsa-project.org/db/?f=cf6d6a903ddb5068115d0e1306fc0f5bec10ce32

plbossart, 2020-03-28:
@ArtOfSettling so you do have a DMIC device, so can you just try and record manually?

See https://github.com/thesofproject/linux/issues/1877#issuecomment-595978695

ArtOfSettling, 2020-03-28:
I tried both arecord and aplay (for input / output) but neither played or recorded anything. :(

ArtOfSettling, 2020-03-28:
I'm not sure if ts is usefu information but I managed to get audio paying trough my bluetooth headphones if I navigated to the settings pane and manually changed the profile (The profile at the bottom of the sound settings pane). It was already set to High Fidelity Playback but I had to change it to HSP/HFP and back to High Fidelity Playback, then audio came over when using bluetooth headphones.

[image]

ArtOfSettling, 2020-03-28:
I've actually noticed that I can hear audio with incredibly low volume and tons of distortion if using wired headphones. It's barely perceptible. alsamixer shows everything turned up to the max.

plbossart, 2020-03-28:
> I tried both arecord and aplay (for input / output) but neither played or recorded anything. :(

can you 
a) dmesg -C (to clear console)
b) arecord -Dhw:0,6 -c2 -r48000 -f S32_LE -d 10 mytest.wav
c) provide both terminal output and dmesg log

Thanks! And no the BT info is not useful, different subsystem.

show the terminal output when you did record?

ArtOfSettling, 2020-03-28:
Here ya go.

```~/Downloads$ sudo dmesg -C
~/Downloads$ arecord -Dhw:0,6 -c2 -r48000 -f S32_LE -d 10 mytest.wav
Recording WAVE 'mytest.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo
~/Downloads$ dmesg
[ 2875.742705] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
[ 2875.831821] sof-audio-pci 0000:00:1f.3: firmware boot complete
~/Downloads$ 
```

plbossart, 2020-03-29:
> Indeed, Here you go.
> 
> ```
> terry@terry-950QCG:~/Downloads$ sudo dmesg -C
> terry@terry-950QCG:~/Downloads$ arecord -Dhw:0,6 -c2 -r48000 -f S32_LE -d 10 mytest.wav
> Recording WAVE 'mytest.wav' : Signed 32 bit Little Endian, Rate 48000 Hz, Stereo
> terry@terry-950QCG:~/Downloads$ dmesg
> [ 2875.742705] sof-audio-pci 0000:00:1f.3: error: no reply expected, received 0x0
> [ 2875.831821] sof-audio-pci 0000:00:1f.3: firmware boot complete
> ```

Have you tried to play or visualize mytest.wav, e.g. with audacity?
Also we'd need dynamic debug enabled. I don't know if there's a shortcut to do it, we typically enable this by default in our development tree.

galmoc, 2020-03-30:
hi all, thank you for this great thread and your efforts!

i have a very similar issue with my Acer Swift SF 314-57. maybe it's the same chipset as in the Samsung Book Flex, didn't check.

I updated the firmware, 
Downloaded the last _'sof-icl.ri'_ file and put it in '/lib/firmware/intel/sof',
Modified 'alsa-base.conf' to add blacklists

 No luck so far. only my Bluetooth headset allows me to hear sound on my computer

here's my system info 
`[code]
System:    Host: xxx-Swift-SF314-57 Kernel: 5.3.0-42-generic x86_64 bits: 64 compiler: gcc 
           v: 7.4.0 Desktop: Xfce 4.14.1 tk: Gtk 3.22.30 wm: xfwm4 dm: LightDM 
           Distro: Linux Mint 19.3 Tricia base: Ubuntu 18.04 bionic 
Machine:   Type: Laptop System: Acer product: Swift SF314-57 v: V1.09 serial: <filter> Chassis: 
           type: 10 serial: <filter> 
           Mobo: IL model: Floris_IL v: V1.09 serial: <filter> UEFI: Insyde v: 1.09 
           date: 09/24/2019 
Battery:   ID-1: BAT0 charge: 18.5 Wh condition: 50.6/48.0 Wh (105%) volts: 11.6/11.2 
           model: LGC KT0030G020 AP18C8K serial: <filter> status: Charging 
           Device-1: hidpp_battery_0 model: Logitech Wireless Mouse M315/M235 serial: <filter> 
           charge: 55% status: Discharging 
CPU:       Topology: Quad Core model: Intel Core i7-1065G7 bits: 64 type: MT MCP arch: Ice Lake 
           rev: 5 L2 cache: 8192 KiB 
           flags: lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx bogomips: 23961 
           Speed: 1699 MHz min/max: 400/3900 MHz Core speeds (MHz): 1: 1854 2: 2920 3: 2714 
           4: 3789 5: 3405 6: 3257 7: 2492 8: 3583 
Graphics:  Device-1: Intel vendor: Acer Incorporated ALI driver: i915 v: kernel bus ID: 00:02.0 
           chip ID: 8086:8a52 
           Display: x11 server: X.Org 1.20.5 driver: modesetting unloaded: fbdev,vesa 
           resolution: 1920x1080~60Hz 
           OpenGL: renderer: Mesa DRI Intel Iris Plus Graphics (Ice Lake 8x8 GT2) 
           v: 4.5 Mesa 19.2.8 compat-v: 3.0 direct render: Yes 
Audio:     Device-1: Intel vendor: Acer Incorporated ALI driver: sof-audio-pci bus ID: 00:1f.3 
           chip ID: 8086:34c8 
           Sound Server: ALSA v: k5.3.0-42-generic 
Network:   Device-1: Intel driver: iwlwifi v: kernel port: 3000 bus ID: 00:14.3 chip ID: 8086:34f0 
           IF: wlp0s20f3 state: up mac: <filter> 
Drives:    Local Storage: total: 476.94 GiB used: 10.84 GiB (2.3%) 
           ID-1: /dev/nvme0n1 vendor: Kingston model: RBUSNS8154P3512GJ1 size: 476.94 GiB 
           speed: 15.8 Gb/s lanes: 2 serial: <filter> 
Partition: ID-1: / size: 467.96 GiB used: 10.83 GiB (2.3%) fs: ext4 dev: /dev/nvme0n1p2 
USB:       Hub: 1-0:1 info: Full speed (or root) Hub ports: 1 rev: 2.0 chip ID: 1d6b:0002 
           Hub: 2-0:1 info: Full speed (or root) Hub ports: 4 rev: 3.1 chip ID: 1d6b:0003 
           Hub: 3-0:1 info: Full speed (or root) Hub ports: 12 rev: 2.0 chip ID: 1d6b:0002 
           Device-1: 3-4:2 info: Logitech Unifying Receiver type: Keyboard,Mouse,HID 
           driver: logitech-djreceiver,usbhid rev: 2.0 chip ID: 046d:c52b 
           Device-2: 3-5:3 info: LighTuning type: <vendor specific> driver: N/A rev: 1.1 
           chip ID: 1c7a:0570 
           Device-3: 3-6:4 info: Chicony type: Video driver: uvcvideo rev: 2.0 chip ID: 04f2:b5c5 
           Device-4: 3-10:5 info: Intel type: Bluetooth driver: btusb rev: 2.0 chip ID: 8087:0026 
           Hub: 4-0:1 info: Full speed (or root) Hub ports: 6 rev: 3.1 chip ID: 1d6b:0003 
Sensors:   System Temperatures: cpu: 62.0 C mobo: N/A 
           Fan Speeds (RPM): N/A 
Repos:     No active apt repos in: /etc/apt/sources.list 
           Active apt repos in: /etc/apt/sources.list.d/brave-browser-release.list 
           1: deb [arch=amd64] https: //brave-browser-apt-release.s3.brave.com/ stable main
           Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list 
           1: deb http: //mirror6.layerjet.com/linuxmint/packages tricia main upstream import backport
           2: deb http: //ubuntu.univ-reims.fr/ubuntu bionic main restricted universe multiverse
           3: deb http: //ubuntu.univ-reims.fr/ubuntu bionic-updates main restricted universe multiverse
           4: deb http: //ubuntu.univ-reims.fr/ubuntu bionic-backports main restricted universe multiverse
           5: deb http: //security.ubuntu.com/ubuntu/ bionic-security main restricted universe multiverse
           6: deb http: //archive.canonical.com/ubuntu/ bionic partner
           Active apt repos in: /etc/apt/sources.list.d/spotify.list 
           1: deb http: //repository.spotify.com stable non-free
           Active apt repos in: /etc/apt/sources.list.d/tista-adapta-bionic.list 
           1: deb http: //ppa.launchpad.net/tista/adapta/ubuntu bionic main
           Active apt repos in: /etc/apt/sources.list.d/wireguard-wireguard-bionic.list 
           1: deb http: //ppa.launchpad.net/wireguard/wireguard/ubuntu bionic main
Info:      Processes: 257 Uptime: 29m Memory: 7.53 GiB used: 2.46 GiB (32.7%) Init: systemd v: 237 
           runlevel: 5 Compilers: gcc: 7.5.0 alt: 7 Client: Unknown python3.6 client inxi: 3.0.32 
[/code]
`
the dmesg log 
``
and here's my alsa-info report http://alsa-project.org/db/?f=6c2b508de44433e82a56cb0a6832367b3565649d

If you can provide any help, that'd be much appreciated!
thanks
best

plbossart, 2020-03-30:
@galmoc Ubuntu kernel 5.3.0-42 is known to be broken. please try and update, newer versions look fine.

ArtOfSettling, 2020-03-31:
@galmoc I'm not sure we have the exact same issues, but I have also updated my kernel from 5.3.x. I'll create a gist with everything i've done when I manage to get it all working.

@plbossart I ended up listening to the file on another machine and there is playback, sure enough. URL, if you're interested in hearing me clap : https://keybase.pub/terrydrever/mytest.wav

So, we have successful recording on the device now, the last remaining problem is the incredibly low volume and distorted audio. Do you have any ideas one what I can do to resolve that? If you can provide me a way to enable dynamic debug, I can do it for you.

plbossart, 2020-03-31:
@ArtOfSettling Capture seems fine, no? what volume issue are you referring to? 

[image]

ArtOfSettling, 2020-03-31:
@plbossart capture looks perfect, but when I play it back on my machine, there's no audio coming from the speakers (or headphones).

paulstelian97, 2020-03-31:
@plbossart I think he's saying that he can't listen his recording on the affected machine, there is a volume issue on the playback side.

@ArtOfSettling If you can kill pulseaudio and try playing with aplay instead (as root would be simplest) and let us know if it's audible that way it would be nice, to be able to distinguish between kernel/firmware (which would also affect aplay) vs UCM (which only affects PA and what, say, your browser or audio player is doing when trying to play audio)

plbossart, 2020-03-31:
@ArtOfSettling did you install the UCM files as recommended:
https://github.com/thesofproject/alsa-ucm-conf/tree/sof_ucm1/ucm/sof-skl_hda_card

ArtOfSettling, 2020-03-31:
@plbossart , @paulstelian97 is correct, I can't listen to the recording on my affected machine, the volume is incredibly low and distorted when using wired headphones, when using the built in speakers, I can't hear anything at all.

@paulstelian97 I killed pulseaudio and tried aplay, but still nothing perceptible from my speakers, same problem from my headphones.

@plbossart I have indeed installed the UCM files.

paulstelian97, 2020-04-01:
So capture works but playback doesn't, at either kernel or firmware level.

When trying aplay, can you also fiddle around in alsamixer to try to unmute or give an increased volume? (try the easiest fixes _first_ to see if there is merely a configuration issue instead of an actual bug)

ArtOfSettling, 2020-04-01:
So, I tested every combination I thought was meaningful for my machine with alsamixer, it was basically fiddling with every settings on/off up/down and no luck with the audio. I'm attaching a screengrab of my alsamixer, see if anything is obviously borked. I tried with and without pulseaudio.

[image]

ArtOfSettling, 2020-04-01:
If it's useful at all, the hardware specs for this device are listed as.

AKG Stereo Speakers ( Max 5 W x 2 ) with Smart Amp

ArtOfSettling, 2020-04-01:
Another interesting thing is that I can see audio is being played with pavucontrol : (This is with pulseaudio and pavucontrol), but no sound  comes out of the speakers.

[image]

ArtOfSettling, 2020-04-01:
Some additional potentially useful information : 

UCM files :
[image]

sof firmware files : 
[image]

Is it possible my setup is selecting an incorrect topology for playback?

paulstelian97, 2020-04-01:
> If it's useful at all, the hardware specs for this device are listed as.
> 
> AKG Stereo Speakers ( Max 5 W x 2 ) with Smart Amp

Hm I wonder if the Smart Amp has something to do with this, I have seen that there are a few open pull requests so I'm not certain about the support.

> Is it possible my setup is selecting an incorrect topology for playback, or do you think this is a legitimate bug?

Topology should be selected correctly by the Linux driver based on ACPI and SMBIOS information so, unless there is a peculiar bug in this situation, it should be correct.

plbossart, 2020-04-01:
@ArtOfSettling yes at this point I would try and blacklist snd-sof-pci and 'unblacklist' snd-hda-intel.
If you can't get any playback working with the legacy HDaudio platform, then it's a codec issue. Maybe you have a missing quirk or something.

The SOF driver is only needed for the DMIC, and you've shown this work. So let's try the playback part and triangulate what the issue might be.

kv2019i, 2020-04-01:
@ArtOfSettling @plbossart Yup, it starts to look like ALC298 codec driver is also missing some functionality, see e.g. this:
https://bugs.launchpad.net/ubuntu/+source/alsa-driver/+bug/1851518

ArtOfSettling, 2020-04-01:
@paulstelian97 @plbossart Yes! You guys are completely correct, if I blacklist snd-sof-pci and get snd-hda-intel running, I have exactly the same problem.

In that case the steps i've run so far (incase they're useful for anyone else with mic issues).

from a base of ubuntu 18.04 LTS, 
1. update kernel to 5.5.13, 
2. Install linux-firmware (somehow it was missing from my system).
3. Add the UCM files from https://github.com/thesofproject/alsa-ucm-conf/tree/sof_ucm1/ucm/sof-skl_hda_card.

I guess I can close this bug here now?

ArtOfSettling, 2020-04-02:
I followed the thread from @kv2019i (https://bugs.launchpad.net/ubuntu/+source/alsa-driver/+bug/1851518) but was unable to get anything working. Attached is a screenshot of everything I tried after switching away from sof.

[image]

I tried 4 different models, but each gave me the same result, either Dummy Audio or low volume distorted sound. Should I create a bug over at bugs.launchpad.net/ubuntu or is this a useful discussion to have here?

ArtOfSettling, 2020-04-02:
Please ignore the above, I had not correctly unblacklisted sof.

I managed to get at least headphones working.

kv2019i, 2020-05-26:
@ArtOfSettling wrote:
> I managed to get headphones working with model=mono-speakers. Nothing from the internal speakers yet.
...
> Should I create a bug over at bugs.launchpad.net/ubuntu or is this a useful discussion to have here?

A bit late follow-up, but that is good to hear. So the remaining problem you have is with the codec driver and that is better dealt as a bug to ALSA kernel (https://www.alsa-project.org/wiki/Bug_Tracking) or via distro (like you mentioned). It seems https://bugzilla.kernel.org/show_bug.cgi?id=207423 is covering this problem. Thanks for reporting the issue!

closed 2020-05-26 by kv2019i

linhhoaile, 2020-12-04:
Have you found the solution to this yet? i'm having the same problem with my Flex.

kv2019i, 2020-12-04:
@linhhoaile It seems multiple people have got sound work on a Flex in https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1851518 -- this requires patches to the codec driver