thesofproject/sof #2638: [BUG][GLK]"error: ipc error for 0x60010000 size 20" when arecord via dmic(0,99) using  S32_LE/ S24_LE format

Liviali155, 2020-03-26:
**Describe the bug**
"error: ipc error for 0x60010000 size 20" when arecord via dmic(0,99) using  S32_LE/ S24_LE format
If using S16_LE has no this issue,after issue occured ,all the other pipeline can work normally
```
ubuntu@sh-glk-bob-da7219-01:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofglkda7219max [sof-glkda7219max], device 1: Headset (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofglkda7219max [sof-glkda7219max], device 99: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

**To Reproduce**
1.sudo reboot
2. arecord -Dhw:0,99 -r 48000 -c 4 -f S32_LE  test.wav -vv -i

**Reproduction Rate**
10/10

**Expected behavior**
Dmic arecord can work normally when using S32_LE/ S24_LE format

**Impact**
"Unable to install hw params",dmesg show "error: ipc error for 0x60010000 size 20"

[image]

**Environment**
    * Kernel: {a2727779} /sof/dev
    * SOF: {ae53aebc}   master
2) Name of the topology file
    * Topology: {sof-glk-da7219.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {GLK Chrome with onboard codec DA7219 in I2S mode}

<details>
 <summary>dmesg_error.log</summary>

````
[   58.157809] sof-audio-pci 0000:00:0e.0: generating page table for 00000000e6521154 size 0xffc0 pages 16
[   58.157826] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
[   58.157863] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
[   58.157866] sof-audio-pci 0000:00:0e.0: period_bytes:0x3fc0
[   58.157867] sof-audio-pci 0000:00:0e.0: periods:4
[   58.157875] sof-audio-pci 0000:00:0e.0: stream_tag 2
[   58.157879] sof-audio-pci 0000:00:0e.0: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[   58.158165] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x60010000 size 20
[   58.158170] sof-audio-pci 0000:00:0e.0: error: hw params ipc failed for stream 2
[   58.158172] sof-audio-pci 0000:00:0e.0: ASoC: 0000:00:0e.0 hw params failed: -12
[   58.158175]  DMIC: ASoC: hw_params FE failed -12
[   58.158179] sof-audio-pci 0000:00:0e.0: pcm: free stream 99 dir 1
[   58.159112] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   58.159118] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   58.170350] sof-audio-pci 0000:00:0e.0: pcm: close stream 99 dir 1
[   58.659063] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   58.659070] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   59.159017] sof-audio-pci 0000:00:0e.0: ipc rx: 0x90020000: GLB_TRACE_MSG
[   59.159022] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x90020000: GLB_TRACE_MSG
[   60.570007] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x20140000 successful
[   60.574673] sof-audio-pci 0000:00:0e.0: ipc tx: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   60.574727] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x40010000: GLB_PM_MSG: CTX_SAVE
[   60.574804] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1010303 successful
[   60.574807] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x1000303 successful
[   60.574809] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 0 : core_mask 3
[   60.574826] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000010 at  00000044
[   60.574829] sof-audio-pci 0000:00:0e.0: Turning i915 HDAC power 0
[   60.588212] sof-audio-pci 0000:00:0e.0: Current DSP power state: D3
[   64.850072] sof-audio-pci 0000:00:0e.0: Turning i915 HDAC power 1
[   64.875494] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000044
[   64.878487] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000040 at  00000048
[   64.878490] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000000 at  00000048
[   64.881727] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000048
[   64.881730] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000040 at  00000048
[   64.883004] sof-audio-pci 0000:00:0e.0: Current DSP power state: D0I0
[   64.883007] sof-audio-pci 0000:00:0e.0: loading firmware
[   64.883011] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000002 at  00000048
[   64.883013] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000000 at  00000048
[   64.883018] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000044
[   64.883021] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000004 at  00000044
[   64.883022] sof-audio-pci 0000:00:0e.0: booting DSP firmware
[   64.883096] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
[   64.883108] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
````
</details>

No sof_errortrace

[dmesg0326.log](https://github.com/thesofproject/sof/files/4386051/dmesg0326.log)
[sof_trace0326.log](https://github.com/thesofproject/sof/files/4386052/sof_trace0326.log)

Liviali155, 2020-03-26:
Changed the kernel and master to 20191201 also has this problem

Liviali155, 2020-03-26:
On CML Chrome with onboard codec RT5682 and RT1011 in I2S mode has no this issue

lgirdwood, 2020-03-26:
@Liviali155 please check that your topology supports this format.

Liviali155, 2020-03-27:
> @Liviali155 please check that your topology supports this format.

I used below command to check ,support S32_LE/ S24_LE format:
```
ubuntu@sh-glk-bob-da7219-1237:~$ arecord -Dhw:0,99 --dump-hw-params /dev/zero
Recording WAVE '/dev/zero' : Unsigned 8 bit, Rate 8000 Hz, Mono
HW Params of device "hw:0,99":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [64 128]
CHANNELS: 4
RATE: 48000
PERIOD_TIME: [250 42500]
PERIOD_SIZE: [12 2040]
PERIOD_BYTES: [192 16320]
PERIODS: [2 16]
BUFFER_TIME: [500 170500]
BUFFER_SIZE: [24 8184]
BUFFER_BYTES: [192 65472]
TICK_TIME: ALL
--------------------
arecord: set_params:1299: Sample format non available
Available formats:
- S16_LE
- S24_LE
- S32_LE

```

lgirdwood, 2020-03-27:
@bkokoszx this is another one where we need the 2/4 channel DMIC option.

sinahuang, 2020-04-13:
Error also can be seen with FW: https://github.com/thesofproject/sof/releases/tag/v1.5-rc1 commit：4b4681eb +https://github.com/thesofproject/linux/commits/topic/sof-dev branch commit：5f7d4e2

bkokoszx, 2020-04-15:
@Liviali155 @sinahuang 
Can you check whether PR #2767 fixes this issue?

sinahuang, 2020-04-15:
Issue can't be reproduced with PR https://github.com/thesofproject/sof/pull/2767, so issue can be fixed by this PR.

plbossart, 2020-04-15:
Well that's rather a kernel driver bug if we accept S32_LE for the front-end but it's not supported by the topology? @ranj063 what do you think?

PR #2767 (2020-04-16): apl: change dmic format to s32_le in sof-glk-da7219.tplg
In order to allow record s16_le/s24_le/s32_le format this
commit changes DAI component's and DMIC's format to s32_le
on PCM99.
In case recording s16_le/s24_le DAI component makes pcm
conversion from s16_le/s24_le to s32_le.

fixes #2638
closed 2020-04-16 by lgirdwood