thesofproject/sof #2643: [BUG] Segmentation Fault when running testbench

sebcarlucci, 2020-03-26:
**Describe the bug**
Running the testbench causes a seg fault. I was able to isolate the issue at 
_src/include/sof/audio/pipeline.h_ in the function pipeline_posn_offset_get().

I tried removing commits e34a64d16a582d7399f630a0e7110755a39fca4f and b20191b362a6d14a44eead247ae677edd90ff6dd and this issue was resolved.

**To Reproduce**
Run the testbench

**Reproduction Rate**
100%

**Environment**
sof master branch (e73344a73658cb0b16f5381954c4f41f5ff42ca6)

sebcarlucci, 2020-03-26:
fyi @singalsu @tlauda

singalsu, 2020-03-26:
Please try this PR #2630

sebcarlucci, 2020-03-26:
issue fixed by PR #2630

singalsu, 2020-03-27:
It's not yet merged. Let's reopen this.

paulstelian97, 2020-03-27:
@singalsu It is now.

singalsu, 2020-03-27:
Yep it is. I'll close this!

There's another issue in testbench build it seems that is related to gcc version 9.2.1. I can create a PR for it.

closed 2020-03-27 by singalsu