thesofproject/sof #2645: [BUG][TGL][KWD] KWD only works with buffer_size >= 67200,

keqiaozhang, 2020-03-27:
**Describe the bug**
KWD only works with buffer size >=67200, or it will be failed with "ipc error for 0x60010000 size 20"
So what the range of buffer size for KWD on TGL?
 
**To Reproduce**
$ arecord -Dhw:0,100 -M -N -c 2 -f S32_LE -r 16000 --buffer-size=67100 kwd.wav -vvv

**Reproduction Rate**
10/10

**Expected behavior**
KWD can works with different buffer size

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: sof-dev/ a27277
    * SOF: master/9b1b61
2) Name of the topology file
    * Topology: sof-tgl-max98373-rt5682.tplg  sof-master/9b1b61 + PR2405
3) Name of the platform(s) on which the bug is observed.
    * Platform: TGL chromebook

**Screenshots or console output**
```
arecord -Dhw:0,100 -M -N -c 2 -f S32_LE -r 16000 --buffer-size=67100 kwd.wav -vvv
Recording WAVE 'kwd.wav' : Signed 32 bit Little Endian, Rate 16000 Hz, Stereo
arecord: set_params:1363: Unable to install hw params:
ACCESS:  MMAP_INTERLEAVED
FORMAT:  S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: 32
FRAME_BITS: 64
CHANNELS: 2
RATE: 16000
PERIOD_TIME: 1048500
PERIOD_SIZE: 16776
PERIOD_BYTES: 134208
PERIODS: 4
BUFFER_TIME: 4194000
BUFFER_SIZE: 67104
BUFFER_BYTES: 536832
TICK_TIME: 0
```
**dmesg**
```
[  367.664949] sof-audio-pci 0000:00:1f.3: pcm: open stream 100 dir 1
[  367.664950] sof-audio-pci 0000:00:1f.3: period min 192 max 256000 bytes
[  367.664951] sof-audio-pci 0000:00:1f.3: period count 2 max 160
[  367.664952] sof-audio-pci 0000:00:1f.3: buffer max 1280000 bytes
[  367.665154] sof-audio-pci 0000:00:1f.3: pcm: hw params stream 100 dir 1
[  367.665156] sof-audio-pci 0000:00:1f.3: generating page table for 0000000026d1e849 size 0x83100 pages 132
[  367.665161] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  367.665180] sof-audio-pci 0000:00:1f.3: FW Poll Status: reg=0x40000 successful
[  367.665182] sof-audio-pci 0000:00:1f.3: period_bytes:0x20c40
[  367.665183] sof-audio-pci 0000:00:1f.3: periods:4
[  367.665193] sof-audio-pci 0000:00:1f.3: stream_tag 2
[  367.665198] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  367.666345] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  367.666364] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  367.666367] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[  367.666369] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[  367.666371]  DMIC16k: ASoC: hw_params FE failed -22
[  367.666374] sof-audio-pci 0000:00:1f.3: pcm: free stream 100 dir 1
[  367.666375] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  367.666473] sof-audio-pci 0000:00:1f.3: pcm: close stream 100 dir 1
[  368.165817] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  368.165856] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
[  368.665749] sof-audio-pci 0000:00:1f.3: ipc rx: 0x90020000: GLB_TRACE_MSG
[  368.665780] sof-audio-pci 0000:00:1f.3: ipc rx done: 0x90020000: GLB_TRACE_MSG
```

[dmesg-buffer-size.txt](https://github.com/thesofproject/sof/files/4392451/dmesg-buffer-size.txt)

lgirdwood, 2020-03-27:
@keyonjie @mrajwa I guess this is intentional, we should have this in topology though so that ALSA will reject this directly.

mrajwa, 2020-03-27:
@lgirdwood yes this behaviour is expected. We reject buffer sizes which won't allows for secure draining, especially in d0ix. By having this in the topology I guess you min to set MIN_BUFFER_SIZE accordingly to our calculated min buffer size so ALSA will reject user space request right away right?
If so I agree and I will add this support to the topology.

lgirdwood, 2020-03-27:
@mrajwa yep, that's what I mean. Thanks.

mrajwa, 2020-04-01:
@lgirdwood  I have added patch for it in this PR https://github.com/thesofproject/sof/pull/2688

slawblauciak, 2020-04-07:
@keqiaozhang @mrajwa Are we good to close this issue?

keqiaozhang, 2020-04-08:
Yes, this issue can be closed.

closed 2020-04-08 by keqiaozhang