thesofproject/sof #2646: [BUG][GLK]"DSP panic"  when do  module load/unload test

Liviali155, 2020-03-27:
**Describe the bug**
"DSP panic" when do  module load/unload test
After error occured,all the pipeline can work normally

**To Reproduce**
1.sudo reboot
2. cd sof-test
3.cd test-case
4.export TPLG=/lib/firmware/intel/sof-glk-da7219.tplg
5../check-kmod-load-unload.sh(https://github.com/thesofproject/sof-test/blob/master/test-case/check-kmod-load-unload.sh)

**Reproduction Rate**
10/10

**Expected behavior**
There is no DSP panic when do module load/unload

**Impact**
"DSP panic" showed in kernel log

[image]

**Environment**
    * Kernel: {a2727779} /sof/dev
    * SOF: {9b1b6158}   master
2) Name of the topology file
    * Topology: {sof-glk-da7219.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {GLK Chrome with onboard codec DA7219 in I2S mode}

<details>
 <summary>dmesg_error.log</summary>

````
[   92.559577] sof-audio-pci 0000:00:0e.0: periods:1
[   92.560193] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x3030303 successful
[   92.560221] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x3030302 successful
[   92.560223] sof-audio-pci 0000:00:0e.0: unstall/run core: core_mask = 1
[   92.560226] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 1 : core_mask 1
[   92.562330] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000000 successful
[   92.562347] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x3010202 successful
[   92.591822] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x5000001 successful
[   92.608163] sof-audio-pci 0000:00:0e.0: error : DSP panic!
[   92.608170] sof-audio-pci 0000:00:0e.0: unknown ROM status value 00000000
[   92.608172] sof-audio-pci 0000:00:0e.0: error: status = 0x0dead000 panic = 0x00000220
[   92.608192] sof-audio-pci 0000:00:0e.0: unknown ROM status value 00000000
[   92.608815] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x5 successful
[   92.609313] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x140000 successful
[   92.609316] sof-audio-pci 0000:00:0e.0: Firmware download successful, booting...
[   92.612338] sof-audio-pci 0000:00:0e.0: ipc rx: 0x70000000: FW_READY
[   92.612341] sof-audio-pci 0000:00:0e.0: ipc: DSP is ready 0x70000000 offset 0x81000
[   92.612358] sof-audio-pci 0000:00:0e.0: Firmware info: version 1:4:1-9b1b6
[   92.612360] sof-audio-pci 0000:00:0e.0: Firmware: ABI 3:14:0 Kernel ABI 3:13:0
[   92.612361] sof-audio-pci 0000:00:0e.0: warn: FW ABI is more recent than kernel
[   92.612362] sof-audio-pci 0000:00:0e.0: Firmware debug build 1 on Mar 27 2020-05:17:14 - options:
                GDB: disabled
                lock debug: disabled
                lock vdebug: disabled
[   92.612390] sof-audio-pci 0000:00:0e.0: found ext header type 1 size 0xb8
[   92.612403] sof-audio-pci 0000:00:0e.0: found ext header type 2 size 0x50
[   92.612405] sof-audio-pci 0000:00:0e.0: Firmware info: used compiler XCC 12:0:8 RG-2017.8-linux used optimization flags O2
[   92.612420] sof-audio-pci 0000:00:0e.0: found ext header type 3 size 0x1c
[   92.612422] sof-audio-pci 0000:00:0e.0: warning: unknown ext header type 3 size 0x1c
[   92.612426] sof-audio-pci 0000:00:0e.0: found ext header type 4 size 0x10
[   92.612427] sof-audio-pci 0000:00:0e.0: warning: unknown ext header type 4 size 0x10
[   92.612457] sof-audio-pci 0000:00:0e.0:  mailbox upstream 0x81000 - size 0x1000
[   92.612459] sof-audio-pci 0000:00:0e.0:  mailbox downstream 0xa0000 - size 0x2000
[   92.612460] sof-audio-pci 0000:00:0e.0:  stream region 0xc1000 - size 0x1000
[   92.612464] sof-audio-pci 0000:00:0e.0: ipc rx done: 0x70000000: FW_READY
[   92.612469] sof-audio-pci 0000:00:0e.0: firmware boot complete
[   92.612475] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000048
````
</details>

No sof_errortrace
[sof_trace0317.log](https://github.com/thesofproject/sof/files/4392459/sof_trace0317.log)

[dmesg0327.log](https://github.com/thesofproject/sof/files/4392458/dmesg0327.log)

Liviali155, 2020-03-27:
Changed the Firmware to 20191201(feb9e2b) also has this problem

Liviali155, 2020-03-27:
On CML Chrome with onboard codec RT5682 and RT1011 in I2S mode
TGL-U RVP with onboard codec ALC711 in HDA mode
has no this issue

lgirdwood, 2020-03-27:
I see this twice in the logs **before** the FW boots. Could this be left over from a previous FW crash or @lbetlej is it a ROM error ?

xiulipan, 2020-04-01:
@lgirdwood the pattern is quite wired, I think there is a DSP panic happen before the FW load success. log from a PR CI test: https://sof-ci.01.org/sofpr/PR2680/build5647/devicetest/GLK_BOB_DA7219/check-kmod-load-unload/dmesg.txt

`0x0dead000  => SOF_IPC_PANIC_MEM `
```
sh-glk-bob-da7219-01 kernel: [  612.917441] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000002 at  00000048
sh-glk-bob-da7219-01 kernel: [  612.917456] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000000 at  00000048
sh-glk-bob-da7219-01 kernel: [  612.917460] sof-audio-pci 0000:00:0e.0: Debug PCIR: 00000000 at  00000044
sh-glk-bob-da7219-01 kernel: [  612.917463] sof-audio-pci 0000:00:0e.0: Debug PCIW: 00000004 at  00000044
sh-glk-bob-da7219-01 kernel: [  612.917464] sof-audio-pci 0000:00:0e.0: booting DSP firmware
sh-glk-bob-da7219-01 kernel: [  612.917530] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
sh-glk-bob-da7219-01 kernel: [  612.917544] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000 successful
sh-glk-bob-da7219-01 kernel: [  612.917548] sof-audio-pci 0000:00:0e.0: period_bytes:0x0
sh-glk-bob-da7219-01 kernel: [  612.917549] sof-audio-pci 0000:00:0e.0: periods:1
sh-glk-bob-da7219-01 kernel: [  612.918139] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x3030303 successful
sh-glk-bob-da7219-01 kernel: [  612.918191] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x3030302 successful
sh-glk-bob-da7219-01 kernel: [  612.918194] sof-audio-pci 0000:00:0e.0: unstall/run core: core_mask = 1
sh-glk-bob-da7219-01 kernel: [  612.918210] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 1 : core_mask 1
sh-glk-bob-da7219-01 kernel: [  612.920168] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000000 successful
sh-glk-bob-da7219-01 kernel: [  612.920172] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x3010202 successful
sh-glk-bob-da7219-01 kernel: [  612.951182] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x5000001 successful
sh-glk-bob-da7219-01 kernel: [  612.967873] sof-audio-pci 0000:00:0e.0: error : DSP panic!
sh-glk-bob-da7219-01 kernel: [  612.967881] sof-audio-pci 0000:00:0e.0: unknown ROM status value 00000000
sh-glk-bob-da7219-01 kernel: [  612.967884] sof-audio-pci 0000:00:0e.0: error: status = 0x0dead000 panic = 0x00000220
sh-glk-bob-da7219-01 kernel: [  612.967886] sof-audio-pci 0000:00:0e.0: unknown ROM status value 00000000
sh-glk-bob-da7219-01 kernel: [  612.968155] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x5 successful
sh-glk-bob-da7219-01 kernel: [  612.968158] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x140000 successful
sh-glk-bob-da7219-01 kernel: [  612.968160] sof-audio-pci 0000:00:0e.0: Firmware download successful, booting...
sh-glk-bob-da7219-01 kernel: [  612.971951] sof-audio-pci 0000:00:0e.0: ipc rx: 0x70000000: FW_READY
sh-glk-bob-da7219-01 kernel: [  612.971954] sof-audio-pci 0000:00:0e.0: ipc: DSP is ready 0x70000000 offset 0x81000
sh-glk-bob-da7219-01 kernel: [  612.971983] sof-audio-pci 0000:00:0e.0: Firmware info: version 1:4:1-90a74
sh-glk-bob-da7219-01 kernel: [  612.971985] sof-audio-pci 0000:00:0e.0: Firmware: ABI 3:16:0 Kernel ABI 3:15:0
sh-glk-bob-da7219-01 kernel: [  612.971986] sof-audio-pci 0000:00:0e.0: warn: FW ABI is more recent than kernel
sh-glk-bob-da7219-01 kernel: [  612.971988] sof-audio-pci 0000:00:0e.0: Firmware debug build 1 on Apr  1 2020-14:57:14 - options:
```

wenqingfu, 2020-04-01:
from CI results in a timeline, this is a regression started happening in the last two days.

lgirdwood, 2020-04-01:
@xiulipan panic does a ```while(1)``` so should not get any further.
@Liviali155 can you bisect given it's only started to happen in last 2 days.

Liviali155, 2020-04-02:
> @xiulipan panic does a `while(1)` so should not get any further.
> @Liviali155 can you bisect given it's only started to happen in last 2 days.

I check the test result of CI,it should not be a regression,the result of 20200320 has DSP panic too,because the issue https://gitlab.devtools.intel.com/sof-ci/sofcitest/issues/407,the error was not found
```
Inserting snd_sof_acpi
Inserting snd_sof_pci
Inserting snd_usb_audio
2020-03-19 21:40:51 UTC [REMOTE_INFO] checking for general errors after kmod insert with sof-kernel-log-check tool
Caught error: Mar 20 05:40:50 sh-glk-bob-da7219-03 kernel: [  682.562298] sof-audio-pci 0000:00:0e.0: hda_sdw_acpi_scan failed
Mar 20 05:40:50 sh-glk-bob-da7219-03 kernel: [  682.562302] sof-audio-pci 0000:00:0e.0: skipping SoundWire, ACPI scan error
Mar 20 05:40:50 sh-glk-bob-da7219-03 kernel: [  682.615243] sof-audio-pci 0000:00:0e.0: error : DSP panic!
kernel log has error(s)
2020-03-19 21:40:51 UTC [REMOTE_INFO] checking for fw_boot success
2020-03-19 21:40:51 UTC [REMOTE_INFO] ==== completed boot firmware: 2 of 2 ====
2020-03-19 21:40:52 UTC [REMOTE_INFO] Test Result: PASS!
```
<details>
 <summary>dmesg_error.log</summary>

````
kernel: [  678.717400] sof-audio-pci 0000:00:0e.0: DSP core(s) enabled? 1 : core_mask 1
kernel: [  678.719687] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x40000000 successful
kernel: [  678.719704] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x3010202 successful
kernel: [  678.725638] usbcore: registered new interface driver snd-usb-audio
kernel: [  678.748028] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x5000001 successful
kernel: [  678.764479] sof-audio-pci 0000:00:0e.0: error : DSP panic!
kernel: [  678.764485] sof-audio-pci 0000:00:0e.0: unknown ROM status value 00000000
kernel: [  678.764488] sof-audio-pci 0000:00:0e.0: error: status = 0x0dead000 panic = 0x00000220
kernel: [  678.764490] sof-audio-pci 0000:00:0e.0: unknown ROM status value 00000000
kernel: [  678.765015] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x5 successful
kernel: [  678.765018] sof-audio-pci 0000:00:0e.0: FW Poll Status: reg=0x140000 successful
kernel: [  678.765021] sof-audio-pci 0000:00:0e.0: Firmware download successful, booting...
kernel: [  678.768487] sof-audio-pci 0000:00:0e.0: ipc rx: 0x70000000: FW_READY
kernel: [  678.768489] sof-audio-pci 0000:00:0e.0: ipc: DSP is ready 0x70000000 offset 0x81000
kernel: [  678.768506] sof-audio-pci 0000:00:0e.0: Firmware info: version 1:4:1-c6398
kernel: [  678.768508] sof-audio-pci 0000:00:0e.0: Firmware: ABI 3:14:0 Kernel ABI 3:13:0
kernel: [  678.768509] sof-audio-pci 0000:00:0e.0: warn: FW ABI is more recent than kernel
kernel: [  678.768511] sof-audio-pci 0000:00:0e.0: Firmware debug build 1 on Mar 20 2020-05:16:27 - options:
kernel: [  678.768511]  GDB: disabled
kernel: [  678.768511]  lock debug: disabled
kernel: [  678.768511]  lock vdebug: disabled
kernel: [  678.768538] sof-audio-pci 0000:00:0e.0: found ext header type 1 size 0xb8
kernel: [  678.768553] sof-audio-pci 0000:00:0e.0: found ext header type 2 size 0x50
kernel: [  678.768555] sof-audio-pci 0000:00:0e.0: Firmware info: used compiler XCC 12:0:8 RG-2017.8-linux  used optimization flags O2
kernel: [  678.768569] sof-audio-pci 0000:00:0e.0: found ext header type 3 size 0x1c
kernel: [  678.768571] sof-audio-pci 0000:00:0e.0: warning: unknown ext header type 3 size 0x1c
kernel: [  678.768574] sof-audio-pci 0000:00:0e.0: found ext header type 4 size 0x10.
````
</details>

[check-kmod-load-unload.sh.txt](https://github.com/thesofproject/sof/files/4418609/check-kmod-load-unload.sh.txt)
[cmd.txt](https://github.com/thesofproject/sof/files/4418610/cmd.txt)
[dmesg.txt](https://github.com/thesofproject/sof/files/4418611/dmesg.txt)

Liviali155, 2020-04-09:
Tried with PR#1992,issue still can be reproduced

lgirdwood, 2020-04-09:
@Liviali155 can you update which PR you tested, above PR is a bug.
Please also put a space between "PR" and "#nnnn" as GH will then automatically insert a link for you.

Liviali155, 2020-04-09:
> @Liviali155 can you update which PR you tested, above PR is a bug.
> Please also put a space between "PR" and "#nnnn" as GH will then automatically insert a link for you.

OK,this PR https://github.com/thesofproject/linux/pull/1992

Liviali155, 2020-04-10:
Tried with PR #2751,issue still can be reproduced

tlauda, 2020-04-10:
@Liviali155 But PR #2751 has nothing to do with this bug. Why would you test with it?

Liviali155, 2020-04-10:
> @Liviali155 But PR #2751 has nothing to do with this bug. Why would you test with it?

I just see the message：
@kv2019i kv2019i mentioned this issue 16 hours ago
schedule: dox: complete api documentation #2751

wenqingfu, 2020-04-10:
It can be traced as back as 11th Mar 2020 , the earliest date old retention policy allowed. I'm sure it can be traced back even earlier if there are more database record, so unfortunately the actual date here isn't much helpful on bi-sect.  Just a FYI. 

“Luckily" this "DSP panic" issue is 100% reproducible when unload the kernel module, hopefully the root cause can be relatively easy.

wenqingfu, 2020-04-10:
> > @Liviali155 But PR #2751 has nothing to do with this bug. Why would you test with it?
> 
> I just see the message：
> @kv2019i kv2019i mentioned this issue 16 hours ago
> schedule: dox: complete api documentation #2751

thanks for the diligent follow up, @Liviali155 !  Because this is currently a CI blocker, so every PR submitter who responsibly check the CI on-device testing result would be puzzled, and they may eventually figure out that this issue is tracking it and quote as reply in their PR review, GH would automatically create a link in this thread pointing back to that conversation. You need to read the context to determine if they are talking about a fix or just mention this issue.

ranj063, 2020-04-10:
@lgirdwood @mrajwa The reason for this DSP panic is that after the module is reloaded, the lp_sram_init times out here:
https://github.com/thesofproject/sof/blob/b12a6626c5ade0a2d102d5a8a4f0147eefea813a/src/arch/xtensa/boot_loader.c#L307

Does it have to do with maybe an improper LP SRAM power gating reset possibly?

ranj063, 2020-04-10:
Also this check here makes no sense.
https://github.com/thesofproject/sof/blob/b12a6626c5ade0a2d102d5a8a4f0147eefea813a/src/arch/xtensa/boot_loader.c#L330

lp_sram_init can never return anything < 0. Thats why even though the DSP panics, we go on with FW load anyway

Liviali155, 2020-04-11:
> thanks for the diligent follow up, @Liviali155 ! Because this is currently a CI blocker, so every PR submitter who responsibly check the CI on-device testing result would be puzzled, and they may [...]

ok,I got it

mengdonglin, 2020-04-12:
Update the priority to P1 as it is causing 100% failure in Linux PR test.

lgirdwood, 2020-04-12:
@tlauda @lbetlej any comment on the LPSRAM init failing ?

sinahuang, 2020-04-13:
Error also can be seen with  FW: https://github.com/thesofproject/sof/releases/tag/v1.5-rc1 commit：4b4681eb +https://github.com/thesofproject/linux/commits/topic/sof-dev branch commit：5f7d4e2

plbossart, 2020-04-13:
maybe a red herring but could we try to revert this kernel patch "ASoC: SOF: Intel: hda-ctrl: add reset cycle before parsing capabilities"
It's the only thing I can think of that changes the boot flow. It's required for SoundWire platforms but maybe it has adverse effects on others. We could never get clear information from hardware folks on this.

ranj063, 2020-04-13:
> maybe a red herring but could we try to revert this kernel patch "ASoC: SOF: Intel: hda-ctrl: add reset cycle before parsing capabilities"
> It's the only thing I can think of that changes the boot [...]

@plbossart I Just tried with this revert and still see the same error. I think this panic pre-dates this patch and it was caught much later because the CI test script missed this.

plbossart, 2020-04-13:
> @plbossart I Just tried with this revert and still see the same error. I think this panic pre-dates this patch and it was caught much later because the CI test script missed this.

ok, thanks for testing.

mrajwa, 2020-04-13:
> @lgirdwood @mrajwa The reason for this DSP panic is that after the module is reloaded, the lp_sram_init times out here: [...]

This piece of code you quoted checks power status of configured LPSRAM banks. If it times out there it means they were not powered on time. Have you tried to increase the counter there or add some other delay to check if it eventually powers on o not? This implementation was based on already existing hpsram_init which basically does the same thing.
BTW, this code is there for a year now, haven't we tested it before or it was working and suddenly stopped?

keqiaozhang, 2020-04-16:
Bisect result shows that this regression is caused by(one year ago):
```
commit c17de6a9793e9554eff46e8659fd91be2ea3a3f3
Author: Marcin Rajwa <marcin.rajwa@linux.intel.com>
Date:   Mon Apr 15 12:46:48 2019 +0200

memory: enable LPSRAM on APL

```
Commit info: https://github.com/thesofproject/sof/commit/c17de6a9793e9554eff46e8659fd91be2ea3a3f3

Kernel version for testing(GLK-V1.2 release kernel version): https://github.com/thesofproject/linux/commit/18a233f3f676a98dde00947535d99ab1a54da340

lgirdwood, 2020-04-16:
@slawblauciak @mrajwa I guess the fix is to increase the delay ? If you can do this today I can merge for v1.5

wenqingfu, 2020-04-16:
> BTW, this code is there for a year now, haven't we tested it before or it was working and suddenly stopped?

There was a false positive test case bug that we only recently corrected.

ranj063, 2020-04-16:
> @slawblauciak @mrajwa I guess the fix is to increase the delay ? If you can do this today I can merge for v1.5

@lgirdwood I increased the delay to 512 in lp_sram_init() and it didnt help.

keqiaozhang, 2020-04-17:
Bi-sect result points to:
commit c17de6a9793e9554eff46e8659fd91be2ea3a3f3
Author: Marcin Rajwa <marcin.rajwa@linux.intel.com>
Date:   Mon Apr 15 12:46:48 2019 +0200

memory: enable LPSRAM on APL

Kernel version for testing:https://github.com/thesofproject/linux/commit/2291b8c29a77e822f514d96433efc9a4cf00d1e8

@mrajwa , please help to confirm if it's reasonable.
As we debugged yesterday, this issue cannot be reproduced after disabling LPSRAM check status.
Do you have any new findings after that?

mrajwa, 2020-04-17:
Yes, debugging in ongoing...in touch...

mrajwa, 2020-04-21:
@keqiaozhang, I know what is happening and how to fix it. However the rootcause is unusual bahaviour of the HW. So before I go into implemntation of the solution, I would like you please you to tell on how many GLK platforms this issue is observed? I want to rule out bad silicon part.

ranj063, 2020-04-21:
> @keqiaozhang, I know what is happening and how to fix it. However the rootcause is unusual bahaviour of the HW. So before I go into implemntation of the solution, I would like you to tell on how many GLK platforms this issue is observed? I want to rule out bad silicon part.

@mrajwa iwe have observed this on our CI GLK systems and I have seen this on my yorp as well. So i doubt its the issue with one or two boards.

fredoh9, 2020-04-21:
@mrajwa I found same problem in my GLK too. This was consistent problem for a while.

keqiaozhang, 2020-04-22:
@mrajwa , according to my observation, there is no GLK models that cannot reproduce this issue.

wenqingfu, 2020-04-22:
> @keqiaozhang, I know what is happening and how to fix it. However the rootcause is unusual bahaviour of the HW. So before I go into implemntation of the solution, I would like you to tell on how many GLK platforms this issue is observed? I want to rule out bad silicon part.

when can we have a patch? @mrajwa  thanks!

mrajwa, 2020-04-23:
@wenqingfu, I have just pushed patches
Kernel -> https://github.com/thesofproject/linux/pull/2044
FW -> https://github.com/thesofproject/sof/pull/2837

sinahuang, 2020-04-24:
Check this issue with sof-dev (commit:8dc25) +PR https://github.com/thesofproject/linux/pull/2044 and master (commit: 761bbac)+PR https://github.com/thesofproject/sof/pull/2837, error can't be reproduced.

sinahuang, 2020-05-13:
Error can't be seen after PR 2044 merged, so closed it.

closed 2020-05-13 by sinahuang