thesofproject/sof #2650: [FEATURE]Improve probe point add API

ranj063, 2020-03-27:
**Is your feature request related to a problem? Please describe.**
When multiple probe points are added with the same command, only probe points with valid input description should be added and the rest disregarded.

**Describe the solution you'd like**
Currently, the API returns an error upon hitting the first invalid probe point description. Instead, we should process all valid inputs and reject the invalid ones or reject the entire set if even one of the probe point desc is invalid

**Additional context**
Add any other context or screenshots about the feature request here.

ranj063, 2020-03-27:
@abonislawski FYI

kv2019i, 2022-06-16:
FYI @jsarha @ujfalusi