thesofproject/sof #2661: [Enhancement] sof-apl-dmic clean up

fredoh9, 2020-03-29:
**Describe the bug**
There are many sof-apl-dmic*.m4 in topology. Most of codes are duplicated and can be shared/reused. Maybe some files can be deleted simply. And review entire DMIC macro for further enhancement. 

1. sof-apl-dmic-a2ch-b2ch.m4
2. sof-apl-dmic-a2ch-b4ch.m4
3. sof-apl-dmic-a2ch.m4
4. sof-apl-dmic-a4ch-b2ch.m4
5. sof-apl-dmic-a96k-b16k.m4
6. sof-apl-dmic-b2ch.m4
7. sof-apl-dmic.m4

**To Reproduce**
n/a

**Reproduction Rate**
n/a

**Expected behavior**
n/a

**Impact**
Keep necessary m4 files in topology

**Environment**
n/a

**Screenshots or console output**
n/a

plbossart, 2020-03-30:
to the best of my knowledge, these files are for test purposes and are not used as-is by the driver - I am not aware of any case where we look for those files in ACPI tables (see sound/soc/intel/common)

So maybe we can move them to a test directory, but I am not sure there's value in optimizing them. Macros, etc, quickly make files complicated for tests.

@singalsu this is for you :-)

singalsu, 2020-03-30:
Yep I created this to test various non-symmetrical configurations for the DMIC IP. They really belong to test topologies like the SSP variations.There's a lot more supported variation than this set that I at time of creation considered as minimum to test.

I need to figure out how to script the possible parameters combinations and add that to tplg-build.sh. The topology file names will change to follow sort how the SSP configurations are differentiated.

closed 2020-04-03 by lgirdwood