thesofproject/sof #2672: [BUG][CML]ASRC]“error: ipc error for 0x60010000 size 20” when aplay/arecord with headset(0,0)

Liviali155, 2020-03-31:
**Describe the bug**
“error: ipc error for 0x60010000 size 20” when aplay/arecord with headset(0,0)

```
ubuntu@cml-hda-mantis-1:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
ubuntu@cml-hda-mantis-1:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 6: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 7: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
**To Reproduce**
1.download sof-hda-asrc-2ch.tplg rename to sof-hda-generic-2ch.tplg
2.sudo reboot
3.aplay -Dhw:0,0 -r48000 -c2 -fs16_le  wav/48California_Dreaming.wav -vv -i or
  arecord -Dhw:0,0 -r48000 -c2 -fs16_le s30.wav

**Reproduction Rate**
5/5

**Expected behavior**
The pipeline can work normally

**Impact**
Tips”Unable to install hw params”,dmesg show “error: ipc error for 0x60010000 size 20”

**Environment**
1) Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
    * Kernel: {38ac5b69} /sof/dev
    * SOF: {aa6c0f2a} master
2) Name of the topology file
    * Topology: {sof-hda-asrc-2ch.tplg rename to sof-hda-generic-2ch.tplg}
3) Name of the platform(s) on which the bug is observed.
    * Platform: {CML laptop with onboard codec ALC3204 in HDA mode}

<details>
 <summary>sof_errortrace.log</summary>

````
  TIMESTAMP              DELTA C# COMPONENT          LOCATION                       CONTENT
[155584189.791667] (155584192.000000) c0 dai          2.17           src/audio/dai.c:449   ERROR dai_params(): invalid period_bytes.
[155584211.302083] (       21.510416) c0 PIPE                   src/audio/pipeline.c:443   ERROR pipeline_params(): ret = -22, host->comp.id = 11
[155584231.875000] (       20.572916) c0 IPC                       src/ipc/handler.c:302   ERROR ipc: pipe 2 comp 11 params failed -22
[155650778.437500] (    66546.562500) c0 dai          2.17           src/audio/dai.c:449   ERROR dai_params(): invalid period_bytes.
[155650799.739583] (       21.302084) c0 PIPE                   src/audio/pipeline.c:443   ERROR pipeline_params(): ret = -22, host->comp.id = 11
[... 118 lines cut ...]
[  123.799464] sof-audio-pci 0000:00:1f.3: period_bytes:0x1080
[  123.799465] sof-audio-pci 0000:00:1f.3: periods:15
[  123.799474] sof-audio-pci 0000:00:1f.3: stream_tag 3
[  123.799478] sof-audio-pci 0000:00:1f.3: ipc tx: 0x60010000: GLB_STREAM_MSG: PCM_PARAMS
[  123.799992] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[  123.799997] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 3
[  123.799998] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -22
[  123.799999]  HDA Analog: ASoC: hw_params FE failed -22
[  123.800005] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
[  123.800193] sof-audio-pci 0000:00:1f.3: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
[  123.800196] sof-audio-pci 0000:00:1f.3: pcm: free stream 0 dir 1
[  123.800198] sof-audio-pci 0000:00:1f.3: hda_link_hw_free: link_dev is not assigned
[  123.800251] sof-audio-pci 0000:00:1f.3: pcm: close stream 0 dir 1
[  123.852401] sof-audio-pci 0000:00:1f.3: pcm: open stream 0 dir 1
[  123.852408] sof-audio-pci 0000:00:1f.3: period min 192 max 16384 bytes
[  123.852411] sof-audio-pci 0000:00:1f.3: period count 2 max 16
[  123.852414] sof-audio-pci 0000:00:1f.3: buffer max 65536 bytes
[  123.852703] sof-audio-pci 0000:00:1f.3: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
.
````
</details>

[dmesg0331.log](https://github.com/thesofproject/sof/files/4408453/dmesg0331.log)
[sof_errortrace0331.log](https://github.com/thesofproject/sof/files/4408454/sof_errortrace0331.log)
[sof_trace0331.log](https://github.com/thesofproject/sof/files/4408455/sof_trace0331.log)

singalsu, 2020-04-01:
@mengdonglin Yep, this is known. Playback is rejected due to incorrect pipeline paramters (zero sample rate). There's no more crash , it was fixed by my PR, but the same fix as was done for SoundWire by @bardliao needs to be done for HDA.

Liviali155, 2020-04-13:
This issue also can be reproduced on sof v1.5-rc1 after download sof-hda-asrc-2ch.tplg rename to sof-hda-generic-2ch.tplg
Kernel:4b4681eb Sof-dev 
fw:5f7d4e2 https://github.com/thesofproject/sof/releases/tag/v1.5-rc1
UCM2(https://github.com/thesofproject/alsa-ucm-conf/tree/master/ucm2/sof-hda-dsp)

mengdonglin, 2020-04-14:
Tplg PR https://github.com/thesofproject/sof/pull/2750 and kernel PR https://github.com/thesofproject/linux/pull/2007 submitted.

Still need firmware update to config rate and channels, please see  https://github.com/thesofproject/sof/pull/2750#issue-401287535 @singalsu @slawblauciak

slawblauciak, 2020-04-14:
FW change https://github.com/thesofproject/sof/pull/2765

Liviali155, 2020-04-16:
Tried with PR #2750&PR #2007& PR #2765 
 aplay/arecord (0,0) can work,but dmesg still has error message “error: ipc error for 0x60010000 size 20” 

<details>
 <summary>dmesg_error.log</summary>

````
[    2.843718] leds platform::micmute: Setting an LED's brightness failed (-19)
[    3.312184] random: crng init done
[    3.770392] leds platform::micmute: Setting an LED's brightness failed (-19)
[    3.770405] leds platform::micmute: Setting an LED's brightness failed (-19)
[    4.237954] iwlwifi 0000:00:14.3: BIOS contains WGDS but no WRDS
[    4.391044] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.391053] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.391057] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.391061]  HDA Analog: ASoC: hw_params FE failed -12
[    4.391976] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.391983] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.391987] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.391991]  HDA Analog: ASoC: hw_params FE failed -12
[    4.392871] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.392878] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.392882] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.392886]  HDA Analog: ASoC: hw_params FE failed -12
[    4.393740] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.393747] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.393752] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.393756]  HDA Analog: ASoC: hw_params FE failed -12
[    4.394595] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.394603] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.394607] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.394611]  HDA Analog: ASoC: hw_params FE failed -12
[    4.396120] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.396126] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.396130] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.396133]  HDA Analog: ASoC: hw_params FE failed -12
[    4.397025] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.397032] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.397036] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.397040]  HDA Analog: ASoC: hw_params FE failed -12
[    4.397877] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.397884] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.397889] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.397892]  HDA Analog: ASoC: hw_params FE failed -12
[    4.398741] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.398749] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.398753] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.398757]  HDA Analog: ASoC: hw_params FE failed -12
[    4.399564] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.399572] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.399577] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -12
[    4.399581]  HDA Analog: ASoC: hw_params FE failed -12
[    4.401712] sof-audio-pci 0000:00:1f.3: error: ipc error for 0x60010000 size 20
[    4.401719] sof-audio-pci 0000:00:1f.3: error: hw params ipc failed for stream 2
[    4.401724] sof-audio-pci 0000:00:1f.3: ASoC: 0000:00:1f.3 hw params failed: -1         
````
</details>

[dmesg0416.log](https://github.com/thesofproject/sof/files/4487318/dmesg0416.log)
[sof-logger0416.log](https://github.com/thesofproject/sof/files/4487320/sof-logger0416.log)

slawblauciak, 2020-04-16:
Can you please provide the sof error trace?

Liviali155, 2020-04-16:
> Can you please provide the sof error trace?

There is no sof error trace,and the error dmesg message occured after bootup

Liviali155, 2020-04-17:
Because aplay/arecord  can work after applied  PR #2750&PR #2007& PR #2765,just a error message in dmesg occured,so closed this issue,have creat another issue  https://github.com/thesofproject/sof/issues/2783 to trace Warning message issue

closed 2020-04-17 by Liviali155