thesofproject/sof #2674: [BUG][CML][EQ]"no FIR or IIF filter is available in this pipeline, skip" when do EQ test

Liviali155, 2020-03-31:
**Describe the bug**
" no FIR or IIF filter is available in this pipeline, skip" when do EQ test
The pipeline aplay/arecord normally
```
ubuntu@cml-hda-mantis-1:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 3: HDMI1 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 4: HDMI2 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 5: HDMI3 (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
ubuntu@cml-hda-mantis-1:~$ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sofhdadsp [sof-hda-dsp], device 0: HDA Analog (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 1: HDA Digital (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 6: DMIC (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: sofhdadsp [sof-hda-dsp], device 7: DMIC16kHz (*) []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
**To Reproduce**
1.chmod +x sof-ctl 
2.sudo mv sof-ctl /usr/bin/
3.get sof-hda-generic-eq-2ch.tplg,then renamed to sof-hda-generic-2ch.tplg
4.reboot
5.cd sof-test
6.cd test-case
7.export TPLG=/lib/firmware/intel/sof-hda-generic-2ch.tplg
8../check-audio-equalizer.sh(https://github.com/thesofproject/sof-test/blob/master/test-case/check-audio-equalizer.sh)

**Reproduction Rate**
10/10

**Expected behavior**
EQ test pass

**Impact**
EQ test failed
```
ubuntu@cml-hda-mantis-1:~/sof-test/test-case$ ./check-audio-equalizer.sh 
2020-03-31 09:12:45 UTC [INFO] ./check-audio-equalizer.sh will use topology /lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg to run the test case
2020-03-31 09:12:45 UTC [INFO] Pipeline list to ignore is specified, will ignore 'pcm=HDA Digital,Media Playback,DMIC16k' in test case
2020-03-31 09:12:45 UTC [INFO] Run command: 'sof-tplgreader.py /lib/firmware/intel/sof-tplg/sof-hda-generic-2ch.tplg -f type:playback -b pcm:'HDA Digital,Media Playback,DMIC16k' -s 0 -e' to get pipeline parameters
2020-03-31 09:12:45 UTC [INFO] 0 - total pipeline= 4
2020-03-31 09:12:45 UTC [INFO] eq_support= EQIIR2.0
2020-03-31 09:12:45 UTC [INFO] no FIR or IIF filter is available in this pipeline, skip
2020-03-31 09:12:45 UTC [INFO] 1 - total pipeline= 4
2020-03-31 09:12:45 UTC [INFO] None of FIR/IIF filter is available in this pipeline, skip
2020-03-31 09:12:45 UTC [INFO] 2 - total pipeline= 4
2020-03-31 09:12:45 UTC [INFO] None of FIR/IIF filter is available in this pipeline, skip
2020-03-31 09:12:45 UTC [INFO] 3 - total pipeline= 4
2020-03-31 09:12:45 UTC [INFO] None of FIR/IIF filter is available in this pipeline, skip
2020-03-31 09:12:45 UTC [INFO] Test Result: FAIL!
```

**Environment**
Branch name and commit hash of the 2 repositories: sof (firmware/topology) and linux (kernel driver).
     Kernel: {38ac5b69} /sof/dev
     SOF: {aa6c0f2} master
Name of the topology file
     Topology: {sof-hda-asrc-2ch.tplg rename to sof-hda-generic-2ch.tplg}
Name of the platform(s) on which the bug is observed.
     Platform: {CML laptop with onboard codec ALC3204 in HDA mode}

There is no dmesg error message and sof error trace
[dmesg_eq.log](https://github.com/thesofproject/sof/files/4408652/dmesg_eq.log)
[sof_traceeq.log](https://github.com/thesofproject/sof/files/4408653/sof_traceeq.log)

Liviali155, 2020-04-03:
On glk eq test also failed.all the pipeline can work normally
test step:
To Reproduce
1.chmod +x sof-ctl
2.sudo mv sof-ctl /usr/bin/
3.get sof-glk-eq-da7219.tplg,then renamed to sof-glk-da7219.tplg
4.reboot
5.cd sof-test
6.cd test-case
7.export TPLG=/lib/firmware/intel/sof-glk-da7219.tplg
8../check-audio-equalizer.sh(https://github.com/thesofproject/sof-test/blob/master/test-case/check-audio-equalizer.sh)

Test result:
```
ubuntu@sh-glk-bob-da7219-1237:~/sof-test/test-case$ ./check-audio-equalizer.sh 
2020-04-03 14:04:31 UTC [INFO] ./check-audio-equalizer.sh will use topology /lib/firmware/intel/sof-tplg/sof-glk-da7219.tplg to run the test case
2020-04-03 14:04:31 UTC [INFO] Pipeline list to ignore is specified, will ignore 'pcm=HDA Digital,Media Playback,DMIC16k' in test case
2020-04-03 14:04:31 UTC [INFO] Run command: 'sof-tplgreader.py /lib/firmware/intel/sof-tplg/sof-glk-da7219.tplg -f type:playback -b pcm:'HDA Digital,Media Playback,DMIC16k' -s 0 -e' to get pipeline parameters
2020-04-03 14:04:31 UTC [INFO] 0 - total pipeline= 5
2020-04-03 14:04:31 UTC [INFO] None of FIR/IIF filter is available in this pipeline, skip
2020-04-03 14:04:31 UTC [INFO] 1 - total pipeline= 5
2020-04-03 14:04:31 UTC [INFO] None of FIR/IIF filter is available in this pipeline, skip
2020-04-03 14:04:31 UTC [INFO] 2 - total pipeline= 5
2020-04-03 14:04:31 UTC [INFO] None of FIR/IIF filter is available in this pipeline, skip
2020-04-03 14:04:31 UTC [INFO] 3 - total pipeline= 5
2020-04-03 14:04:31 UTC [INFO] None of FIR/IIF filter is available in this pipeline, skip
2020-04-03 14:04:31 UTC [INFO] 4 - total pipeline= 5
2020-04-03 14:04:31 UTC [INFO] None of FIR/IIF filter is available in this pipeline, skip
2020-04-03 14:04:31 UTC [INFO] Test Result: FAIL!
```
[dmesg0403.log](https://github.com/thesofproject/sof/files/4427871/dmesg0403.log)
[sof-trace0403.log](https://github.com/thesofproject/sof/files/4427875/sof-trace0403.log)

fredoh9, 2020-04-07:
I tested eq test in WHL with sof-hda-generic-eq-4ch.tplg. Worked fine.

And I tested same eq test in CML Mantis with sof-hda-generic-eq-2ch.tplg. It worked fine as well. Please double check you used eq topology for testing
```
ubuntu@sh-cml-mantis-hda-03:~/sof-test/test-case$ ./check-audio-equalizer.sh
...
2020-04-07 07:50:23 UTC [INFO] EQ test done: failed_cnt is 0
2020-04-07 07:50:23 UTC [INFO] 1 - total pipeline= 4
2020-04-07 07:50:23 UTC [INFO] None of FIR/IIF filter is available in this pipeline, skip
2020-04-07 07:50:23 UTC [INFO] 2 - total pipeline= 4
2020-04-07 07:50:23 UTC [INFO] None of FIR/IIF filter is available in this pipeline, skip
2020-04-07 07:50:23 UTC [INFO] 3 - total pipeline= 4
2020-04-07 07:50:23 UTC [INFO] None of FIR/IIF filter is available in this pipeline, skip
2020-04-07 07:50:23 UTC [ERROR] Failed to generate topology graph
2020-04-07 07:50:23 UTC [INFO] Test Result: PASS!
```

However for GLK is real error. GLK eq topology support only IIR filter in DMIC capture. Let me provide custom test script for now.

fredoh9, 2020-04-07:
When I submit equalizer test, I assume EQ in playback pipeline. That was not good assumption. And there is a TODO to update tplgtool to populate eq pipeline only. I will check with @aiChaoSONG for tplg tool support.

By that time, can you use this test script for glk eq test? I tested with sof-glk-eq-da7219.tplg in GLK BOBBA.
[check-audio-equalizer-glk-dmic.sh.txt](https://github.com/thesofproject/sof/files/4442352/check-audio-equalizer-glk-dmic.sh.txt)

fredoh9, 2020-04-07:
For your CML Mantis error, I see you have only IIR.
`
2020-03-31 09:12:45 UTC [INFO] eq_support= EQIIR2.0
`

It should have IIR and FIR. I'm not sure what went wrong.
`
2020-04-06 23:29:34 UTC [INFO] eq_support= EQIIR1.0 EQFIR1.0
`

aiChaoSONG, 2020-04-07:
@fredoh9 The feature is now available,  In the test script, use `func_pipeline_export $tplg "eq:any"` to get all pipelines with EQ. then iterate all pipelines you get as other test scripts do.

fredoh9, 2020-04-07:
@aiChaoSONG thanks. I will take a look when I'm available for this task.

Liviali155, 2020-04-08:
I retested on CML Mantis with sof-hda-generic-eq-2ch.tplg,result is  pass
sof/dev/(6ae32d68)+master(17f92963)
[image]

lgirdwood, 2020-04-08:
@Liviali155 can this be closed ? or are there other tests pending ?

aiChaoSONG, 2020-04-08:
This is actually an sof-test issue, may transfer it to sof-test

fredoh9, 2020-04-08:
I agreed the warning is due to missing -r option. this is not blocking issue for the test.
I will close this and submit a issue to sof-test.

closed 2020-04-08 by fredoh9