thesofproject/sof #2678: [RFC] clarify firmware errors and warnnings

aiChaoSONG, 2020-04-01:
On CI, we have a test case to check firmware trace log, and grep "error" from the log. But sometimes, we see the key word "error" in the trace log, but the error is not actually a really error. 
For example: `ERROR validate(), ll drift detectec, data = 25549`  and other block alloc errors.

We are not sure we should fail the test case or not. We can make a white list to ignore those "errors", but before that, we want to have a discussion with FW folks.
Maybe you guys can clarify errors and warnings, or give some white list for us?

aiChaoSONG, 2020-04-01:
@lgirdwood @tlauda @ktrzcinx @mmaka1 @bkokoszx @dbaluta @jajanusz @slawblauciak @wwittbrx

paulstelian97, 2020-04-01:
The ll drift detected one should be transient, maybe a reason of concern but could happen when the load is high (multiple pipelines stress test). I think this one is fine to ignore for the most part (when the drift is too high you get a panic though)

Allocation errors may be considered errors, however there was a rejected proposal to mark certain allocations as not giving out an error (via usage of a flag) so that only the important errors in allocation can be considered as errors. I don't think this one should be ignored but instead check on a case by case basis.

mmaka1, 2020-04-01:
No error should be whitelisted imo. Unexpected events that do not reduce any functionality as the system is able to recover should not be logged as errors. The latter is sth to stop, investigate and fix.
There is a PR (see link above) that addresses the agent message.

marc-hb, 2020-04-01:
> No error should be whitelisted imo. 

Thanks, I agree non-errors should not lie and pretend they are errors, this is confusing not just for CI but for everyone :-) I can think of one exception though: when _testing error handling itself_ (the area that consistently has the least test coverage). In that particular case the tests themselves are ideally smart enough to trap the logs?

closed 2022-01-24 by aiChaoSONG

marc-hb, 2022-01-24:
Tests check for neither errors not warnings... anything goes. https://github.com/thesofproject/sof-test/pull/799