thesofproject/sof #2679: [BUG][DSM] Channel 3,4 are empty when testing with S16_LE format

keqiaozhang, 2020-04-01:
**Describe the bug**
Tested the DSM on TGL, S24_4LE and S32_LE are working well. but when testing with S16_LE format, channel 3 and 4 are empty in the recorded wav. From waveform, it looks like the data of channel 3 and 4 are mixed in channel 1 and 2. 

[image]

**To Reproduce**
aplay -Dhw:0,0 -f S16_LE -r 48000 -c2 sine_s16_le_48k.wav --dump-hw-params & arecord -Dhw:0,0 -f S16_LE -r 48000 -c 4 -d 10 DSM.wav -vv

**Reproduction Rate**
5/5

**Expected behavior**
channel 3,4 should also have data

**Impact**
no data in Channel 3 and 4, and channel 1 and 2 data are also wrong. 

**Environment**
1) Kernel: thesofproject/linux topic/smart-amp
2) Sof: thesofproject/sof tgl-004-drop-stabe 
3) Tplg: sof-tgl-smart-amplifier.m4 (rename it to sof-tgl-nocodec.tplg
4) Platform:TGL

[trace-dsm.txt](https://github.com/thesofproject/sof/files/4413109/trace-dsm.txt)
[dmesg-dsm.txt](https://github.com/thesofproject/sof/files/4413110/dmesg-dsm.txt)

slawblauciak, 2020-04-01:
Low priority for now, this format is not supported currently.

paulstelian97, 2020-04-01:
Probably the DMA engine is still configured for 4 bytes per elementary transfer instead of 2? Who's the owner of the DMA engine on this platform?

lgirdwood, 2020-04-01:
Isn't demux/mux used here ? see #2687

emilchudzik, 2020-04-03:
Issue fixed on:
Sof: bkokoszx/topic/smart-amp-revert-mux (859110d79)
Tplg: sof-tgl-smart-amplifier.m4 (rename it to sof-tgl-nocodec.tplg)

Checked on:
Kernel: thesofproject/linux topic/smart-amp topic/smart-amp (10b767431cbb) 
Platform:TGL

keqiaozhang, 2020-04-27:
This issue is fixed in tgl-005-drop-stable branch. We tested it on Volteer-I2S and this issue is not reproducible.
Test recipe:
Driver: fce7f934f + https://github.com/thesofproject/linux/pull/1971 + https://git.kernel.org/pub/scm/linux/kernel/git/broonie/sound.git/commit/?h=for-next&id=914f674bec6efe42f9d6b036850a618fd1698290
Firmware:
SOF branch:  tgl-005-drop-stable Commit: 277a9717
TPLG: sof-tgl-max98373-rt5682.tplg

closed 2020-04-27 by keqiaozhang