thesofproject/sof #2682: [FEATURE] Formalize tracking if all content from product branches is upstreamed

lbetlej, 2020-04-01:
**Is your feature request related to a problem? Please describe.**
There is a need to establish procedure to verify that fixes/features are not lost during upstream process between product release (from release branch) and master branch.

**Describe the solution you'd like**
For every commit on a product branch there is created a separated issue. The issue is assigned to author of a commit on a production branch. The original author is responsible for verification that the commit is properly reflected (as there may be code changes during upstream process) on a master branch. The issue may be closed after verification is done. The issues may be submitted with a help of a dedicated tool (preferred) or manually.  

The assumption is that the procedure is mandatory for Intel product branches and is executed after product release from a branch. 

**Additional context**
[image]

Depicted above is an example list of issues generated with help of dedicated tool. Note that commit hash is mandatory in a title as is used by tool to make sure that there are no duplicating issues generated accidentally if the tool is invoked multiple times. The hash is followed by the first line of commit message.

[image]

Currently the tool adds references to commit hash and a source branch in a description, assigns owner and adds label.

I'm looking for your feedback especially regarding "output" on GH : if more/less data is need to be included in issue body, if issue title is better to be trimmed to make sure fits in a single line, if "Commit" word in a title is perceived as a redundant.

paulstelian97, 2020-04-01:
I think this belongs to the linux repo? For SOF this repo _is_ the upstream.

lbetlej, 2020-04-01:
@paulstelian97 By upstream I meant the master branch in SOF firmware repo. There is a bunch of additional branches in the repo that are used for SOF releases targeting dedicated customers/features. There changes from the branches are later upstreamed (merged to master) following standard PR review process.

paulstelian97, 2020-04-01:
Ah, fair enough, this wasn't clear to me.

I still think that for those of us who receive notifications for all of the PRs/issues or whenever I'm trying to look for an old PR this may generate a lot of noise.

lgirdwood, 2020-04-01:
@wenqingfu any inputs ?