thesofproject/sof #2684: [BUG] IPC timeout setting DAI config

lgirdwood, 2020-04-01:
**Describe the bug**
IPC timeout when setting DAI config for SSP5 reported by CI several times now.

**To Reproduce**
Run the following test
```
cmd: TPLG=sof-apl-pcm512x.tplg ~/sof-test/test-case/check-runtime-pm-status.sh -d 10
====================
framework load: ssh ubuntu@sh-apl-up2-pcm512x-02.sh.intel.com  'nohup bash -c "TPLG=sof-apl-pcm512x.tplg ~/sof-test/test-case/check-runtime-pm-status.sh -d 10 2>&1 | nc -N 10.239.156.45 41223" >/dev/null &'
====================
framework run: 'netcat -l -k 41223'
====================
2020-03-30 23:49:16 UTC [REMOTE_INFO] /home/ubuntu/sof-test/test-case/check-runtime-pm-status.sh will use topology /lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg to run the test case
2020-03-30 23:49:16 UTC [REMOTE_INFO] Pipeline list to ignore is specified, will ignore 'pcm=HDA Digital,Media Playback,DMIC16k' in test case
2020-03-30 23:49:16 UTC [REMOTE_INFO] Run command: 'sof-tplgreader.py /lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg -b pcm:'HDA Digital,Media Playback,DMIC16k' -s 0 -e' to get pipeline parameters
2020-03-30 23:49:16 UTC [REMOTE_INFO] Iteration 1 of 3 for Port5
2020-03-30 23:49:16 UTC [REMOTE_COMMAND] aplay -D hw:0,0 -r 48000 -c 2 -f S16_LE /dev/zero -q
2020-03-30 23:49:19 UTC [REMOTE_INFO] runtime status: active
2020-03-30 23:49:19 UTC [REMOTE_COMMAND] kill process: kill -9 7904
2020-03-30 23:49:19 UTC [REMOTE_INFO] aplay killed
2020-03-30 23:49:19 UTC [REMOTE_INFO] wait dsp power status to become suspended
2020-03-30 23:49:21 UTC [REMOTE_INFO] dsp suspended in 3s
2020-03-30 23:49:21 UTC [REMOTE_INFO] runtime status: suspended
2020-03-30 23:49:21 UTC [REMOTE_INFO] Iteration 2 of 3 for Port5
2020-03-30 23:49:21 UTC [REMOTE_COMMAND] aplay -D hw:0,0 -r 48000 -c 2 -f S16_LE /dev/zero -q
2020-03-30 23:49:24 UTC [REMOTE_INFO] runtime status: active
2020-03-30 23:49:24 UTC [REMOTE_COMMAND] kill process: kill -9 7978
2020-03-30 23:49:24 UTC [REMOTE_INFO] aplay killed
2020-03-30 23:49:24 UTC [REMOTE_INFO] wait dsp power status to become suspended
2020-03-30 23:49:26 UTC [REMOTE_INFO] dsp suspended in 3s
2020-03-30 23:49:27 UTC [REMOTE_INFO] runtime status: suspended
2020-03-30 23:49:27 UTC [REMOTE_INFO] Iteration 3 of 3 for Port5
2020-03-30 23:49:27 UTC [REMOTE_COMMAND] aplay -D hw:0,0 -r 48000 -c 2 -f S16_LE /dev/zero -q
2020-03-30 23:49:29 UTC [REMOTE_INFO] runtime status: active
2020-03-30 23:49:29 UTC [REMOTE_COMMAND] kill process: kill -9 8052
2020-03-30 23:49:29 UTC [REMOTE_INFO] aplay killed
2020-03-30 23:49:29 UTC [REMOTE_INFO] wait dsp power status to become suspended
2020-03-30 23:49:32 UTC [REMOTE_INFO] dsp suspended in 3s
2020-03-30 23:49:32 UTC [REMOTE_INFO] runtime status: suspended
2020-03-30 23:49:32 UTC [REMOTE_INFO] Iteration 1 of 3 for Port5
2020-03-30 23:49:32 UTC [REMOTE_COMMAND] arecord -D hw:0,0 -r 48000 -c 2 -f S16_LE /dev/null -q
arecord: set_params:1363: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 48000
PERIOD_TIME: 85000
PERIOD_SIZE: 4080
PERIOD_BYTES: 16320
PERIODS: (4 5)
BUFFER_TIME: 341000
BUFFER_SIZE: 16368
BUFFER_BYTES: 65472
TICK_TIME: 0
/home/ubuntu/sof-test/test-case/check-runtime-pm-status.sh: line 96: kill: (8145) - No such process
2020-03-30 23:49:35 UTC [REMOTE_ERROR] arecord process for pcm Port5 is not alive
2020-03-30 23:49:37 UTC [REMOTE_INFO] Test Result: FAIL!
```

**Reproduction Rate**
Not sure, probably 50% since I've seen it twice on CI.

**Expected behavior**
No FW crash

**Impact**
Showstopper

**Environment**
CI config
```
tested at 2020-03-31 00:00:23.900650 UTC against:

Linux: 38ac5b69 ASoC: SOF: remove unneeded variables
SOF: aa6c0f2ad topology: cmake: silence super chatty alsatplg unless VERBOSE=1
```

**Screenshots or console output**
```
kernel: [  445.755763] sof-audio-pci 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
kernel: [  445.755815] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
kernel: [  445.755824] sof-audio-pci 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
kernel: [  445.755875] sof-audio-pci 0000:00:0e.0: ipc tx succeeded: 0x80010000: GLB_DAI_MSG: CONFIG
kernel: [  445.755884] sof-audio-pci 0000:00:0e.0: ipc tx: 0x80010000: GLB_DAI_MSG: CONFIG
kernel: [  445.755971] sof-audio-pci 0000:00:0e.0: error: ipc error for 0x80010000 size 12
kernel: [  445.755980] sof-audio-pci 0000:00:0e.0: error: failed to set dai config for SSP5-Codec
kernel: [  445.755985] sof-audio-pci 0000:00:0e.0: error: failed to restore pipeline after resume -22
```
Please also include the relevant sections from the firmware log and kernel log in the report (and attach the full logs for complete reference). Kernel log is taken from *dmesg* and firmware log from *sof-logger*. See https://thesofproject.github.io/latest/developer_guides/debugability/logger/index.html

dwrobel, 2020-04-01:
I'm observing the similar repeating on the  Lenovo Yoga C740-14IML.

dmesg excerpt:
```
[73920.496183] sof-audio-pci 0000:00:1f.3: error: ipc timed out for 0x80010000 size 120
[73920.496204] sof-audio-pci 0000:00:1f.3: status: fw entered - code 00000005
[73920.496462] sof-audio-pci 0000:00:1f.3: error: can't enter idle
[73920.496466] sof-audio-pci 0000:00:1f.3: error: trace point 00004000
[73920.496469] sof-audio-pci 0000:00:1f.3: error: panic at src/lib/agent.c:50
[73920.496473] sof-audio-pci 0000:00:1f.3: error: DSP Firmware Oops
[73920.496478] sof-audio-pci 0000:00:1f.3: EXCCAUSE 0x0000003f EXCVADDR 0x00000000 PS       0x00060725 SAR     0x00000000
[73920.496482] sof-audio-pci 0000:00:1f.3: EPC1     0x00000000 EPC2     0xbe00d30e EPC3     0x00000000 EPC4    0x00000000
[73920.496485] sof-audio-pci 0000:00:1f.3: EPC5     0x00000000 EPC6     0x00000000 EPC7     0x00000000 DEPC    0x00000000
[73920.496489] sof-audio-pci 0000:00:1f.3: EPS2     0x00060d20 EPS3     0x00000000 EPS4     0x00000000 EPS5    0x00000000
[73920.496493] sof-audio-pci 0000:00:1f.3: EPS6     0x00000000 EPS7     0x00000000 INTENABL 0x00000000 INTERRU 0x00000222
[73920.496496] sof-audio-pci 0000:00:1f.3: stack dump from 0xbe04f5b0
[73920.496504] sof-audio-pci 0000:00:1f.3: 0xbe04f5b0: be00fc00 be04f5e0 be062180 00000001
[73920.496509] sof-audio-pci 0000:00:1f.3: 0xbe04f5b4: 00000000 00000000 00000032 00000000
[73920.496514] sof-audio-pci 0000:00:1f.3: 0xbe04f5b8: 8b2b4100 f9adccb9 0ac96828 ffff8a4d
[73920.496517] sof-audio-pci 0000:00:1f.3: 0xbe04f5bc: 01cb79a4 ffffb1b3 00000000 ffffffff
[73920.496520] sof-audio-pci 0000:00:1f.3: 0xbe04f5c0: 088e6e68 ffff8a4d 00000078 00000000
[73920.496523] sof-audio-pci 0000:00:1f.3: 0xbe04f5c4: 07bc79a8 ffff8a4d 00000000 00000000
[73920.496527] sof-audio-pci 0000:00:1f.3: 0xbe04f5c8: 00000005 00000000 8b2b4100 f9adccb9
[73920.496530] sof-audio-pci 0000:00:1f.3: 0xbe04f5cc: c0f12c7b ffffffff ffffffff ffffffff
[73920.496552] sof-audio-pci 0000:00:1f.3: error: hda irq intsts 0x00000000 intlctl 0xc0000000 rirb 00
[73920.496555] sof-audio-pci 0000:00:1f.3: error: dsp irq ppsts 0x00000000 adspis 0x00000000
[73920.496565] sof-audio-pci 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
[73920.496569] sof-audio-pci 0000:00:1f.3: error: failed to set dai config for Analog Playback and Capture
[73920.496573] sof-audio-pci 0000:00:1f.3: ASoC: can't set Analog CPU DAI hw params: -110
[73920.496583]  Analog Playback and Capture: ASoC: hw_params BE failed -110
[73920.496585]  HDA Analog: ASoC: hw_params BE failed -110
```

kernel: kernel-5.5.11-200.fc31.x86_64

dwrobel, 2020-04-01:
A question, is there maybe a way to somehow reload/reboot the firmware without rebooting the whole machine?

lgirdwood, 2020-04-01:
@dwrobel thanks for the info, best to unload and reload the SOF driver modules. I think auto recovery is on @plbossart plans for SOF driver, but not sure when .

plbossart, 2020-04-01:
> @dwrobel thanks for the info, best to unload and reload the SOF driver modules. I think auto recovery is on @plbossart plans for SOF driver, but not sure when .

when we have a watchdog coming from the DSP :-)

lgirdwood, 2020-04-01:
@plbossart hehe, I was thinking some reload logic tacked onto IPC timeout handling. i.e close any open streams and reload FW

plbossart, 2020-04-01:
> @plbossart hehe, I was thinking some reload logic tacked onto IPC timeout handling. i.e close any open streams and reload FW

that's a part of the solution, but that requires the driver to send IPCs, which isn't necessarily possible or practical in all cases. E.g. during playback there are no IPCs, or only indirect ones when an underflow is reported by the core.

lgirdwood, 2020-04-02:
@plbossart I think we should be OK here as any stream error (caused by FW crash) would mean no movement of the hw_ptr which will always result in an IPC (where we can trap the crash).

YvonneYang2, 2020-06-17:
Issue still can be reproduced on suspend/resume stress test with latest topic/sof-dev+master. 

reproduced steps:
 1. System boot up.
 2. cd sof-test then run "export TPLG=/lib/firmware/intel/sof-tplg/sof-apl-pcm512x.tplg"
 3. cd test-case then run "./check-suspend-resume.sh -l 200"

Environment
Platform: APL UP2 with I2S PCM512x mode
Firmware: 4f1090d0 https://github.com/thesofproject/sof/commits/master
Kernel: cc26638 https://github.com/thesofproject/linux/commits/topic/sof-dev
Topology: file: tools/topology/sof-apl-pcm512x.tplg same as firmware

[dmesg.log](https://github.com/thesofproject/sof/files/4792118/dmesg.log)
[all_log.log](https://github.com/thesofproject/sof/files/4792120/all_log.log)

Liviali155, 2020-07-27:
On  APL UP2 with I2S PCM512x  with  latest sof-dev (b31892b) +master(1c53b8e9) 
run  "./check-suspend-resume.sh -l 200" 2times
       "check-runtime-pm-status.sh -d 10"  20 times 

Error not seen

Liviali155, 2020-07-28:
Daily regression test also not reproduce this issue recently,closed

closed 2020-07-28 by Liviali155