thesofproject/sof #2689: [RFC] enhance topology validation

fredoh9, 2020-04-01:
**Describe the bug**
Lately I have worked on topology clean up and enhancements. Before every push I need to make sure no side effect nor unwanted changes are accidentally introduced. Simple changes also required careful check. Sometime quotation or extra `,` create different results. hard to find.

Current sof-ci supports checkpatch and build. Defintely we need more.

1. M4 macro validation. I found a feature request, https://github.com/thesofproject/sof/issues/2515. I'm looking at how this might be implemented and integrated.
2. Graph comparison, this is low hanging fruit. We have a tool to generate dot and png. I can compare dot text files and throw an error if diff is found. At least we can keep same graph by this.
3. Ultimately we need diff *.conf files. I did manually every time and go through line by line. Every change will make some difference in conf file. Some are desired, some are not. **I need some input/comment/suggestion for this task**.

This is one example from I fixed with sof-apl-dmic-a4ch-b2ch.conf comparison.
[image]

DMIC is accidentally put(or not removed). Hard to catch this error without comparing conf. May be #1 task might help here if we can validate type of parameter.
```
PIPELINE_PCM_ADD(sof/pipe-volume-capture.m4,
        6, 6, 4, s16le,
        1000, 0, 0, DMIC,
        48000, 48000, 48000)
```

lgirdwood, 2020-04-02:
@fredoh9 some of these need to go into the ALSA compiler, i..e validation of type

fredoh9, 2020-04-21:
There is no immediate and EASY automated solution.
Close for now.

closed 2020-04-21 by fredoh9